cyoung/stratux #217: WingX using AHRS heading

ScraboTower, 2016-01-22:
This may be a known issue but its the first time I have had an chance to play with this combo in a while - latest release of WingX and latest Stratux beta.

WingX does not recommend to use AHRS to drive the heading but I switched it on anyway (  I am a rebel !!) and was rotating the stratux thru 360 degrees - heading on WingX jumped from 133 to 225 degrees with approx 1mm of rotation 

Known issue ?

cyoung, 2016-01-23:
Version?
https://github.com/cyoung/stratux/blob/master/CONTRIBUTING.md

ScraboTower, 2016-01-23:
v0.5b5

AvSquirrel, 2016-01-23:
Try pulling latest commit from master.

The missing headings from 135-225 degrees was noted on #190 and fixed with [this commit](https://github.com/cyoung/linux-mpu9150/commit/af550804eff32bbef740bbaeb62eb0b4094496c7) to linux-mpu9150. Note that this commit also puts the AHRS into what could be called "free gyro" mode, i.e. accelerometer removed from the fused IMU attitude. (This is needed for future AHRS development).

It will probably start up at about 30 degrees nose-down pitch, but you can cage / zero the attitude in WingX

ScraboTower, 2016-01-23:
Thanks - I'll just wait for the next beta... pulling commits are new to me and I dont want to mess up.

skypuppy, 2016-01-23:
avsquirrel, if you'll examine the white papers on this subject, for the
full ahrs stabilized readings, all 3 instruments are critical, including
the accelerometer.  Please re-examine your code with this in mind.  Gyro
alone is good for only about 5-10 milliseconds for accuracy.  I do not
know which algorithm Invensense has posted for the mpu9150 at this
point, but disregarding the accel readings is a recipe for disaster. 
Please examine the Bosch and MIT (?) discussions regarding this issue. 
Each of the 3 instruments is used to counter the deficiencies of the
other two, leaving a "sensor fusion" approach as being required to
determine accuracy over time.

closed 2016-01-23 by ScraboTower