cyoung/stratux #219: WebUI Discrepancies 0.5b5

kjablonski, 2016-01-25:
1. Stratux 0.5b5
2. Config:  2 SDR, RY835AI

Testing out the 0.5b5 build I noticed that there are differences between what i can see using a laptop connected to the Stratux and the link provided by Stratux on my cellphone for the favorite link.  My laptop will show the GPS constellation count and status where the mobile application only shows the number of satellites.
Mobile:

Laptop:
[image]

The GPS / AHRS page shows the place holders for the values but does not update, Mobile is fine.
[image]

bradanlane, 2016-01-25:
looks like the mobile has cached some old[er] files. The best solution to this is to establish a "manifest" for the files so the browser knows when local cached copies are no longer current.

kjablonski, 2016-01-25:
I'll clear my cache and recheck.  I was running 0.5b1

braddward, 2016-01-25:
If you've saved it to your home screen on an Apple device I've found that
Apple stores that cache separately from the Safari cache.  The only way to
clear that sometimes is to delete the icon and reinstall it.   A manifest
file is probably a good idea.

cyoung, 2016-01-26:
Need: cache manifest.

bradanlane, 2016-01-26:
A manifest file have a version number to increment when files change. Is there an easy way to generate this number as part of the build? 

_I'm sure I could hack something wish bash scripting but im feeling lazy and hope there is a better way_

cyoung, 2016-01-26:
Yes, in the Makefile:

```
www:
	echo "# " `git describe --tags --abbrev=0` >>/var/www/stratux.appcache
```

Or something similar. That would tag the file on install with the git commit hash.

cyoung, 2016-01-29:
https://github.com/cyoung/stratux/commit/442e882646467bf6db9d7dd84dc0e8a5ddd527fa

commit 442e8826 (2016-01-29): Cache manifest. Fixes #219.
closed 2016-01-29 by cyoung