cyoung/stratux #221: 1090ES Traffic on FltPlan go 0.5b6

jwp01, 2016-01-27:
Minor problem / observation: 1090ES traffic on Fltplan go now has a dark ring around each target.  This was not the case with 0.5b5.  WingX does not have the rings.
[image]

cyoung, 2016-01-27:
I think this has more to do with a FltPlan Go change - how they're displaying traffic. Just tried it out and the ring doesn't seem proportional to speed or anything else that one might expect.

cyoung, 2016-01-27:
One of the FltPlan Go devs responded to an email about this - they say the black circle is just for visibility. Their choice, not mine.

closed 2016-01-27 by cyoung