cyoung/stratux #223: As last GPS fix gets older, last GPS fix time moves further into future

bkwny, 2016-01-27:
Here's how to reproduce it.  Enable GPS on the WebUI but don't actually plug in any GPS device.  Watch the line-mode console output for a while and observe the sequence of "Last GPS fix:" messages printed by gen_gdl90.go line 1131.  No GPS fix is arriving because there is no GPS device, so the message ought to indicate amounts of time increasingly in the past:  1 minute ago, 2 minutes ago, 5 minutes ago, and so on.  Instead the GPS fix times move increasingly into the future:  1 minute from now, 2 minutes from now, 5 minutes from now, 1 hour from now, and so on.

bkwny, 2016-01-27:
I did some digging on this.  Somebody check me to see whether I got it correct.

From /dustin/go-humanize/times.go  line 58:  
The invocation of humanize.RelTime is supposed to be humanize.RelTime(candidate_moment, reference_moment, "earlier", "later").
This will result in:
a.  If candidate is before reference the suffix will be "earlier".
b.  If candidate is after reference the suffix will be "later".

But in /cyoung/stratux/monotonic.go   line 38:
We can see the invocation of humanize.RelTime is backward.
He wants to humanize the input time t with respect to reference moment m.Time.
And his suffix strings have the same temporal relationship as the example in dustin/times.go.
This means t should be the first argument to humanize.RelTime.
But it's not.  t is the second argument.

I believe the fix is on monotonic.go line 38 to switch the first and second arguments in the invocation of humanize.RelTime.

cyoung, 2016-01-27:
@bkwny - that is correct. AvSquirrel's pull request has fixed this.

PR #224 (2016-01-27): Fix #223. Change ticker to 50 ms intervals for future improvements.
Fixes #223, which was caused by an incorrect variable order in HumanizeTime().

Changes monotonic ticks from 1 second to 50 millisecond intervals (supports some back-end changes on my development branch). Uptime status report in gen_gdl90.go updated.
commit 2f9ce16b (2016-01-27): Fix #223. Change ticker to 50 ms intervals for future improvements.
commit 9ba317ab (2016-01-27): Merge pull request #224 from AvSquirrel/monotonic-fixes

Fix #223. Change ticker to 50 ms intervals for future improvements.
closed 2016-01-27 by cyoung