cyoung/stratux #225: [request] ZeroConf/mdns networking - avahi

egid, 2016-01-29:
It'd be great to be able to visit http://stratux.local or something to reach the webui, rather than http://192.168.10.1. Could we add mdns support? I've done it on other raspberry pi units I've built before.

```sudo apt-get update
sudo apt-get install libnss-mdns
sudo hostname stratux
```

cyoung, 2016-01-29:
We could, but we'd lose the ability to use cellular internet when connected to stratux.

egid, 2016-01-29:
Ah. Damn.

cyoung, 2016-01-30:
Well, unless you can think of a way to have the DHCP clients configured with an internal DNS server and somehow switch between that and cell service DNS - forward packets, or setting a primary/secondary type setup. That's the challenge.

egid, 2016-01-30:
I'll poke around. Maybe I can finally contribute some actual code to the project. :) This sort of thing is vaguely in my wheelhouse.

egid, 2016-01-30:
Discussed on slack, but for the record: I set up `avahi-daemon` on my 0.5b5 build yesterday, changed hostname to `stratux`, and was able to both reach the internet _and_ `http://stratux.local` from my iPhone.

Haven't tested on a cellular iPad as I don't have one. I'll be able to test Android monday morning.

Ergonomicmike, 2016-01-31:
Thanks for putting it on the record here. I got an invite to Slack but I find it hard enough to keep up with everything monitoring this and the reddit forum. I fear that, with three forums now, some things are bound to drop thru the cracks and not be reported/published in the more public/open forums.

cyoung, 2016-02-01:
It should help make it easier to selectively keep up with things - less chat here.

egid, 2016-02-11:
We may want to make this an option - it seems to flat-out not work with Android. I can do a little more investigation around how android devices support mdns, but for now it seems like the short answer is "they don't".

cyoung, 2016-03-08:
Still need better options here.

closed 2016-03-08 by cyoung