cyoung/stratux #227: WingX elevation off.

zn105, 2016-01-29:
I am running the RY835 connected via the GPIO. The Stratux status screen shows an elevation of 350ft and in WingX connected to the Stratux it is sgowing 700ft. WingX is set to use GPS and there is no adjustment factor put in.

cyoung, 2016-01-29:
Probably using pressure alt from ownship? @HiltonG

zn105, 2016-01-29:
Yes that is it. Stratux preasure altitude is 707ft. How do I change that? Can't find a setting in Stratus setup or WingX.

HiltonG, 2016-01-30:
WingX Pro7 is just reporting what is being sent in the ownship message (https://github.com/cyoung/stratux/blob/master/main/gen_gdl90.go#L284).  I would recommend changing this to always report GPS altitude which would be consistent with other ADS-B receivers.  @zn105 you can adjust the altitude in WingX Pro7 on iPad by tapping on the altitude, then adjusting the altimeter in the drop down view - to make it even simpler, just tap on Ground while on the ground and WingX Pro7 will automagically set the altimeter for you.

Hilton

cyoung, 2016-01-30:
Makes sense. Hilton, what do you think is the best way to deal with pressure alt vs GPS alt in ownship? GDL90 says that Ownship should use pressure altitude (that if statement you pasted is because most EFBs don't follow this) and Ownship Geometric Altitude should be the message that sends the GPS alt. If we change this to GPS alt only then we are sending two messages with GPS alt and pressure alt isn't used at all.

AvSquirrel, 2016-01-30:
@cyoung -- don't deviate from GDL90 on the 0x0A ownship message. Traffic is reported as pressure altitudes, and ownship pressure altitude is needed to correctly determine relative elevation. At least two other EFBs are using it correctly as a reference datum for traffic relative altitude. 

@HiltonG -- Stratux sends both pressure altitude and GPS altitude. Pressure altitude part of the 0x0A ownship message. GPS altitude is encoded as an int16 in bytes 2-3 of the 0x0B ownship geometric altitude, reported as height above ellipsoid (HAE) in 5-foot increments. We've deviated slightly from strict GDL90 by reporting MSL height in lieu of HAE, but that is something of a compatibility hack to make certain other EFBs indicate correctly.

A little competitive intelligence for you:

* **WingX**: Interprets pressure altitude from ownship 0x0A message as MSL GPS altitude
* **ForeFlight**: Interprets GPS altitude from ownship 0x0B geo alt message as MSL GPS altitude; interprets pressure altitude from ownship 0x0A message as MSL pressure altitude.
* **FltPlan Go**: Interprets GPS altitude from ownship 0x0B geo alt message as MSL GPS altitude; interprets pressure altitude from ownship 0x0A message as MSL pressure altitude.
* **iFly GPS**: Interprets GPS altitude from ownship 0x0B geo alt message as MSL GPS altitude
* **Avare**: Seems to alternate between ownship 0x0B geo altitude and phone GPS altitude. Starting up without the Stratux connected appears to flash the WGS84 altitude (-100 ft offset at my location) for a second or two before settling in to the MSL altitude.
* **Naviator**:Used to interpret ownship 0x0B geo altitude as WGS84 HAE, in accordance with GDL90 spec; recently changed to MSL for compatibility purposes.

cyoung, 2016-01-30:
@AvSquirrel is FF actually following this as stated now? If so, then we might just remove the GPS alt from 0x0A. Before the 0xFFF "Invalid Altitude" in 0x0A was actually being interpreted as an altitude (because we don't send 0x0B when GPS alt is not available).

AvSquirrel, 2016-01-30:
@cyoung -- It's getting there, but ForeFlight still doesn't interpret the edge cases correctly as of v7.5.2. Testing with fixed values:

0xFFE = 101,350' MSL (correct)
0xFFF = **0' MSL**  (should by invalid)
0x000 = **Unknown / invalid** (should be -1000' MSL)
0x001 =  -975' MSL (correct)

Ownship detection logic for vertical bounds hasn't changed. On the ground, ForeFlight will latch onto 0x0A pressure altitude and use it in the ownship message only if it's within ±1000' of 0x0B GPS altitude. Otherwise, it's ignored, ownship is not "detected", and a "Stratux" traffic target will be displayed at ownship position with the with the 0x0A.

If 0x0B is not sent, then FF uses 0x0A altitude as GPS altitude, detects ownship, and references all traffic against that altitude.

HiltonG, 2016-01-30:
Hi,

My suggestion: Follow the GDL90 spec as closely as possible and don't be influenced by any app's behavior.  Looks like I stand corrected; i.e. pressure altitude versus GPS altitude.  I will check WingX Pro7's code and ensure it adheres to the spec.

Hilton

cyoung, 2016-02-15:
Is it fixed @HiltonG ?

closed 2016-03-08 by cyoung