cyoung/stratux #228: [Request/Idea] Companion App for additional functionality

peepsnet, 2016-01-31:
Would there be any use for a companion app to allow access to functions that are not available in the EFB's that we are connection to. 

After reading this idea [https://github.com/cyoung/stratux/issues/215](url) I thought about having an app to set this function, as well as any other ideas/functions we come up with??

ssokol, 2016-01-31:
A companion app is a good idea for several reasons, but would require a
good deal of effort. It might be necessary for users who want to use
Stratux with EFBs that don't officially support it, but I would really like
to see us work with the existing EFB vendors to add support for Stratux
features into their apps. That way you don't have to deal with the
limitations that are imposed by the operating systems on backgrounded apps.

Hilton from WingX Pro is a good example - he wrote the spec for the GDL90
status message for Stratus, and incorporated it into the app. He's also
working with us on AHRS and other features.

peepsnet, 2016-01-31:
That is a valid point but you know you will never get all the EFB's to add all the features of the Stratux. Even the experimental ones we dream up here.  I guess one easy option is a faux app that just brings up the webgui(for those who dont know about IP addressed) in its own web browser wrapper.

Does the WebGUI have the capability of monkeying around with all the options. Can the WebGIU include an SSH page?

Things like that. Changes that the EFB's wont want mess with.

ssokol, 2016-01-31:
Sure - you can do that with something like Apache Cordova or just using a
webview in a very simple native app.

I'll probably wind up building some basic apps to help FlightBox customers
install updates without having to mess around with computers, micro SD
cards, etc. The app would receive a push notification that an update is
available. When it opens, it would download the files(s). When it connects
to the Stratux system it would upload the files, etc. Can't do it easily
with a pure web app because of the limits that Apple puts on mobile Safari.

peepsnet, 2016-01-31:
That sounds like a great direction!  I just always forget the command line access to all the settings and configurations in Unix things. 

What about even allowing the WIFI adapter being able to check for its own updates through the web interface. If you added a 2nd WIFI adapter and have it connect to a network and check and download updates?

bkwny, 2016-03-20:
I think it is great to consider a large companion app such as this.  However it has again come up on the Reddit that someone has a legitimate need to change the SSID and the channel number.  If you guys could see your way clear to put at least those two fields onto the WebUI Settings page I think you would be doing your community a huge favor.  Thanks for listening.

peepsnet, 2016-03-21:
I did at least add this to the WIKI... Crude but more then having to repeat yourself time after time. 

https://github.com/cyoung/stratux/wiki