cyoung/stratux #229: [Request] Weather page detail

peepsnet, 2016-02-02:
I would like a little bit more detail on the weather page. Instead of just grouping each message under recent messages could we split them up in to types of messages? It shows that my TFRs have not updated in 5 hours but I can't tell if any are coming through because so much goes by so fast.

I would like to break it out in each message type. 

[ ] 1. - METAR
[ ] 2. - TAF
[ ] 3. - NOTAM
[ ] 4. - TFR
[ ] 5. - WINDS ( also I am NOT getting any winds aloft information)
[ ] 6. - NEXTRAD DATA ( I don't necessarily need this drawn out but just to know that the date is there)

Thanks!
Don

Galaxy Tab A
Ver 0.6r1
Avnav EFB
PI2
Dual sdr nano2
Stock antennas
Standard WiFi dongle

bradanlane, 2016-02-02:
The purpose of the Stratux WebUI is for configuration and diagnostics. The full UI for weather and traffic is provided by the various EFB / Navigation Apps.

cyoung, 2016-02-02:
@peepsnet - I assume your EFB tells you that TFRs have not been updated since 5 hours ago? If you see weather scrolling in the stratux web interface, you are receiving data. TFRs are only sent if they exist and the ADS-B tower you are receiving from has it in its coverage area. There isn't any "TFRs active: NONE" type of message sent that you can use to positively see that no TFRs are applicable in your area. Is that what you're looking for?

peepsnet, 2016-02-02:
@cyoung - 

> I assume your EFB tells you that TFRs have not been updated since 5 hours ago?
That is correct. Same for winds aloft.
 I want to compare (@bradanlane - for diagnostics) if any TFR's/WindsAloft/whatever's are being received versus a bug in my EFB software.  

http://imgur.com/QyZj6lq (time is at the bottom of the screen)

cyoung, 2016-12-01:
UAT stats from @jamez70 in the developer menu should help with this.

https://github.com/cyoung/stratux/pull/519

closed 2016-12-01 by cyoung