cyoung/stratux #232: [Request] 192.168.10.1/logs/startux sorting

peepsnet, 2016-02-03:
Can we sort this page alphabetically?? Its a PITA to find anything.

kjablonski, 2016-02-03:
+1 or sort by type (UAT, 1090, GPS)

bkwny, 2016-03-01:
+1.  +100 if I could.  I just now pulled logs for @cyoung.  What a pain it was.

peepsnet, 2016-03-05:
The only way I can see to get this going is json. 
Can we get a json array of files in the log folder.
we can parse with angularUI and list sort and filter. then we can build a link around each file .

I don't know how to do that in any timely manor but I know we could take that approach

Thoughts???

cyoung, 2016-03-05:
Don't bother too much with it, in the next release we'll be using a webserver (not Go webserver) to serve the webui files + logs. What you are requesting is a fairly standard feature of most webservers.

peepsnet, 2016-03-06:
Great.

cyoung, 2016-03-08:
[image]

Any better?

bkwny, 2016-03-08:
Oh yes!

cyoung, 2016-03-08:
Accessible at http://192.168.10.1/view_logs. Linked also in the "Logs" tab of the main interface.

closed 2016-03-08 by cyoung

bkwny, 2016-03-08:
I have 796d0a2 but my /logs/ tab does not link to /view_logs.  And in /logs/ when I click on "SDR, AHRS, and GPS logs" I see the old-style listing of log files.

cyoung, 2016-03-08:
The first part is not intended behavior, the second part sounds like your browser may have cached the files.

Did you try going directly to the link? http://192.168.10.1/view_logs/

bkwny, 2016-03-08:
Yes, when I tried to go directly to the link, the next thing I saw in the browser window was the Web UI's "Status" tab.

commit 796d0a26 (2016-03-08): /view_logs page.

Fixes #232.