cyoung/stratux #233: CPU temp "bottoming" out

Ergonomicmike, 2016-02-03:
Just saw this today. CPU temp in the web UI sometimes alternating from actual to -99 degrees C/-146.2 F. (Which I presume is the lower limit of the calculation?) Thought it might be low Wi-Fi signal strength. (But why not dashes then if it misses an update?) But can't reproduce the problem now.

Five second video here:
https://vid.me/s8Kt

The only change I recall making recently was that I had GGA info for the GPS updating every 5 Seconds instead of the usual every 1 Second.

Not critical. Simply reporting it. Close this issue if you wish.

cyoung, 2016-02-03:
-99 is our "invalid temperature". The sensor on the RPi doesn't give a read sometimes. Not sure why. It's not your WiFi.

Ergonomicmike, 2016-02-04:
Thanks. First time I've seen/noticed it. I

There are more important issues so I'll close this out. (And nothing you can do about a bad sensor on the Pi.) But I presume there's a code reason why it doesn't show all negative signs (that is, dashes) for invalid temp?

closed 2016-02-04 by Ergonomicmike