cyoung/stratux #234: Request for .klm file for CloudAhoy

cfirandy, 2016-02-04:
Attempting to use Stratux "flight" with CloudAhoy.  Stratux uses WiFi and not bluetooth.  As such, it's not recognized by iOS and so it's not recognized by CloudAhoy.  Would like to log my "Stratux" flight and import its data to CloudAhoy.  Thoughts?

cyoung, 2016-02-04:
http://help.cloudahoy.com/importing-flight-data/

http://www.grtavionics.com/datarecording.html - maybe we could start using a file output like this? We'll likely have some further use other than just GPS tracks in the future.

jpoirier, 2016-02-04:
Outputting data in GPX format is fairly straight forward. An example is this plugin for x-plane (https://github.com/jpoirier/DataLogger/blob/master/main.cpp) that saves the flight data to a gpx formatted file which can be imported by cloudahoy or other clients that recognize the format. Look for writeFileProlog, writeData, and writeFileEpilog.

ScraboTower, 2016-02-04:
So you are trying to use Stratux as a flight track logger - something you can view in for example Google Earth ?  I have WingX and just use the "Replay" function to record all that data, can dump it as GPX I believe 
[image]

skypuppy, 2016-02-04:
You might be on your own with this, but install a (low power) bluetooth
dongle instead of the wifi dongle, than just change where the data is
sent.  Place to start.

Bearcode, 2016-08-24:
I took a stab at adding this functionality in my pull request #481 

I implemented a way to export GPS location of the Stratux device (aka Ownship) to a KML file that can be imported to CloudAhoy. I also provided a way to dump Replay Logs to KML with Ownship + Traffic so you can visualize issues when debugging antenna setups and GPS interference.

Give it a spin and see if there are any other features I am missing.

vushalflyboy, 2021-05-05:
@alexberryman can you tell how to add the replay and record feature