cyoung/stratux #235: Web UI GPS Status not reporting loss of GPS

Ergonomicmike, 2016-02-04:
(I thought I saw a template for submitting New Issues, but I can't remember where. Am copying from a previous issue posted by Chris.)

    Stratux version: v0.6r1 (with modified ry385ai.go and powersave governor)
    Stratux config: dual SDR, GPS BU-353-S4, AHRS no, battery Anker E5, battery cable Anker 21 AWG
    EFB app and version NA. EFB platform NA. EFB hardware NA.
    Description of your issue. "When you unplug the GPS USB cable, the GPS information (solution & satellites) on the web UI Status page does not show the loss of GPS. (The information stays as it last was before GPS is lost.) Same comment for the GPS/AHRS page in the web UI. See https://www.reddit.com/r/stratux/comments/442lnn/please_pull_your_usb_gps_and_report_what_happens/  for more info."

cyoung, 2016-02-04:
When you click "New Issue", you'll see "Please review the guidelines for contributing to this repository." There's a link to auto-fill the issue with that there.

Ergonomicmike, 2016-02-04:
Ahh... okay. I saw the message about guidelines. But like most Americans, I didn't bother to read the instructions. :-(

mooneyguy, 2016-02-04:
Just wanted to let you know I just read the "Statux ADS-B: DIY and Save Big". It was a good article supporting Stratux and a great reflection on the brains working on it.

 

Thanks for all of your work!

 

Mark

AvSquirrel, 2016-02-05:
See #238.

PR #238 (2016-02-05): Detect disconnected GPS. Update web UI and attempt reconnect.
Fixes #235.

This change detects if it's been more than five seconds since a valid NMEA message has been received.

If so, solution status is changed to "Disconnected", mySituation.quality and satellite counts are zeroed, and we break out of the gpsSerialReader() routine. This allows the GPS to reinitialize without spawning multiple instances of this goroutine.
closed 2016-02-05 by cyoung