cyoung/stratux #236: Any alternative to "shutdown" in webUI to accomodate Anker users?

Ergonomicmike, 2016-02-04:
1. Stratux version: v0.6r1 (with modified ry385ai.go and powersave governor)
  2. Stratux config: dual SDR, GPS BU-353-S4, AHRS no, battery Anker E5, battery cable Anker 21 AWG
  3. EFB app and version NA. EFB platform NA. EFB hardware NA.
  4. Description: "It has been well documented in the reddit forum that something in the Anker E series power packs causes the Stratux to restart after a Shutdown is commanded from the web UI. (See https://www.reddit.com/r/stratux/comments/3z16ca/stratux_wont_shut_down_with_ui_shutdown_and_anker/ ) Interestingly, when one sends the shutdown command to the Pi via SSH Ethernet (with a 1 minute timer to allow me to unplug the Ethernet cable), the Pi "shuts down" without restarting. ("Shuts down" in quotes because the SDR's stay lit, GPS says powered, and the Edimax continues to flash. (But no Wi-Fi signal.)) Similarly, when I use Juice to send "sudo shutdown 0," the Pi also "shuts down" without restarting. (Again, LED's stay on.) Since we can't change the Anker, and since the recent survey showed a significant number of users use the Anker, is there a way to emulate the SSH shutdown command from the web UI?

wakkow, 2016-02-04:
This might be the difference between poweroff and halt. The go code sends a 'LINUX_REBOOT_CMD_POWER_OFF' command, whereas I'm pretty sure that the shutdown command (shutdown 0) is doing a halt. Switching the code to do a halt [would be easy](https://github.com/cyoung/stratux/blob/1be927e672cddbbd2d5312b7f55c71ba4c6510ca/main/managementinterface.go#L248), but it's taking away the true poweroff from everyone else that doesn't have an Anker. Can you try running the poweroff command instead of shutdown on the command line while on the anker and see if it restarts?

Ergonomicmike, 2016-02-04:
I did "poweroff." It causes the same behavior as the web UI shutdown. So it appears that the UI is either sending "poweroff" or "shutdown 0 -p"

As I understand it, the reason for the Shutdown command in the UI was not so much to provide a way to shut down the SDR's and GPS, but rather to provide a way to pull the power to the Pi without corrupting the file system. (Since the Stratux is not RO.) 

Even for the substantial number of users using ship's power for the Stratux, they still have to pull the plug/turn off a switch, because even the poweroff command does not power off all power to the Pi. (If nothing else, the red LED stays lit.)

So I'm thinking, change the Shutdown button to send "shutdown 0" without the -p? Or, alternatively, a new Halt button in the UI. That way, Anker users can use HALT. Everyone else can use Shutdown, same as before. (Although I'm adverse to cluttering up the UI.)

skypuppy, 2016-02-04:
Sometimes the USB ports stay powered as long as there is power to the
computer.  This happens on laptops, desktops, embedded, etc.  Sometimes,
they don't.  I don't know what the rules are, if any, but that is normal.

I've noticed that my RPi2's keep the red LED lit even after shutdown; I
don't know why.  At first, I thought it was still up but after several
attempts at communicating with it, I decided it was indeed shutdown. 
That red LED only goes off if power is removed.   Strange.

Skypuppy

Ergonomicmike, 2016-02-04:
Yep. I just tested it again after your post. My GSP also stays on and blinking after a shutdown command. So, from an ergonomic perspective, it might be better to change the web UI button to "Halt" simply because the "Shutdown" command doesn't really do what most people expect it to do.

1-prop-head, 2016-02-05:
I know this is common knowledge but just to refresh our memories the best procedure when using halt is 
# sync
# sync
# sync
# halt

This should help to get all the pending disk writes on the sd card

Ergonomicmike, 2016-02-05:
Do you know if Raspi Check does it this way in its Halt command?

skypuppy, 2016-02-05:
Doesn't "shutdown -h now" do the same thing as "halt," but is a graceful
shutdown and saves the filesystem?

skypuppy, 2016-02-05:
In some environments, it is sync, sync, sync, haltsys.

Ergonomicmike, 2016-02-05:
I don't know anything about Linux. [This website](http://linux.about.com/library/cmd/blcmdl8_sync.htm) makes it sound like sync is called automatically when a halt command is sent?

skypuppy, 2016-02-05:
Not true.

cyoung, 2016-02-05:
This conversation is funny, it brought me back to the days of actually typing "sync" and listening for the crackle of the hard drive to finish before shutting down.

jzeevi, 2016-02-05:
Yep....

Ergonomicmike, 2016-02-05:
Found this in the Stratux code by searching here for shutdown:
===
 main/managementinterface.go
Showing the top two matches. Last indexed 11 days ago.
```
func handleShutdownRequest(w http.ResponseWriter, r *http.Request) {
syscall.Sync()
syscall.Reboot(syscall.LINUX_REBOOT_CMD_POWER_OFF)
```
I vaguely remember now that someone here told me (in a closed issue somewhere) that Stratux was making a kernel call to shutdown. If it has to stay a kernel call (which would make the following easy to change), then I found this on the web:
===
LINUX_REBOOT_CMD_HALT
 	(RB_HALT_SYSTEM, 0xcdef0123; since 1.1.76). The message ‘System halted.’ is printed, and the system is halted. Control is given to the ROM monitor, if there is one. If not preceded by a sync(2), data will be lost.

Ergonomicmike, 2016-02-05:
There is also 
***
LINUX_REBOOT_CMD_POWER_OFF
 	(0x4321fedc; since 2.1.30). The message ‘Power down.’ is printed, the system is stopped, and all power is removed from the system, if possible. If not preceded by a sync(2), data will be lost. 
***
But I presume that doesn't work on the Pi?

Ergonomicmike, 2016-02-05:
Huh - anytime I use the halt command (like, in shutdown -h, the Anker causes a restart of the Pi. The only way I can reliable stop the Pi to pull the plug is with a plain shutdown command via SSH. Even tho the SDR LED's stay on, the Edimax light turns off and there's no signal on Channel 1.

Ergonomicmike, 2016-02-26:
So I know that all you Unix guys know this. I just mention it for brainstorming purposes. Puppy linux uses SIGTERM followed by SIGKILL. I tried both from puTTY. No joy. (Command not found.) Can they be installed and is it worth doing?

cyoung, 2016-02-26:
SIGTERM and SIGKILL are signals. So kill -SIGTERM procnum. "kill" or associated commands would be the ones to use. What is Puppy Linux?

Ergonomicmike, 2016-02-26:
Puppy Linux is yet another [Distro of Linux.](http://distrowatch.com/table.php?distribution=puppy) Fairly popular among noobs. I boot it when I need to play with go files for the Stratux or to gparted another partition on the microSD card for my tablet. (Or, possibly to create another partition for the Stratux, if that will be the optional path someday for users who want Read Only.)

cyoung, 2016-03-05:
Probably will be moving to read-only anyways. Fairly certain there's nothing further we can do to improve this.

closed 2016-03-05 by cyoung

jzeevi, 2016-03-05:
There's always a shutdown button on a GPIO.....

Josef
Sent from a phone with a tiny screen

kdknigga, 2016-03-05:
Probably all distros issue SIGTERMs and then SIGKILLs as part of the
graceful shutdown process.  SIGTERM tells processes to nicely exit.  If any
processes ignore this, SIGKILL comes along and isn't so nice.
(Tangentially related: see this man page to find out why I always caution
against people getting into the habit of using the killall command.
<http://h20565.www2.hpe.com/hpsc/doc/public/display?docId=emr_na-c02259676&lang=en-us&cc=us>
)

Now, shutdown has a couple of different flags you can give it that make it
interact with the hardware differently.  Back in the dark ages, computers
didn't have electronically controllable power supplies.  You might remember
shutting down your OS and getting a prompt that "it is now safe to turn off
your computer" in these bleak times.  That prompt was because the OS had no
way to tell the power supply to turn off, so it relied on you physically
throwing a switch to remove power.  Once more advanced hardware was
developed, the OS could then signal the power supply to turn off and no
more "it is now safe to turn off your computer" screens.

The -h flag to shutdown assumes that such a controllable power supply
exists and will try to issue those power off commands once the OS is down.
On the other hand, -H assumes there is no controllable power supply and
simply downs the OS and prints a "it is now safe to turn off your computer"
message.  My Pi+Anker don't have this problem for some reason so I can't
test this, but you might try using the 'shutdown -H' command (with a big
H).  Perhaps that will help.

mhoran, 2018-01-10:
Hey @cyoung, given you've chosen not to implement a read-only filesystem as per #78, do you plan to revisit this? I still have one of the Anker devices (originally recommended on stratux.me) that results in my Pi rebooting on shutdown. I've dealt with some corrupt sdcards in the past so I'm trying to be careful, but given my Pi reboots I don't always catch it before it starts back up after shutdown.

pp4life, 2018-01-27:
@mhoran I'm still in the same boat.