cyoung/stratux #237: Help Page for Logs Shows Last Help Page Opened

pp4life, 2016-02-04:
Stratux version: v0.6r1
Stratux config: dual SDR, GPS none, AHRS none, battery Anker E5
EFB app and version NA. EFB platform NA. EFB hardware: iPad Mini 3.
Description: Open the Stratux webpage at 192.168.10.1 in a web browser (have reproduced on Safari on iPad Mini 3 and Chrome on Samsung Galaxy S5). Hit the "HELP" menu on any page other than "Logs", then close that Help menu. Then go to Logs and hit "HELP" and the help menu for whatever page you hit the help menu on last will be displayed.

peepsnet, 2016-02-07:
Replicated on 0.6r1
Seems to be that there is no help on the "Logs" page. This means nothing populates the Log Help when you go to that page. The fix should be as simple as to add a " "(Space) to the log help page!!

peepsnet, 2016-02-11:
There is no logs-help.html 

So it would show the last help loaded!!

Kinda a bug!!

bradanlane, 2016-02-17:
fixed with #257

closed 2016-02-17 by bradanlane