cyoung/stratux #239: prioritizing a dongle with a valid id string over an anonymous one

jpoirier, 2016-02-06:
when a dongle with a stratux id string is introduced, or exists in the config list, it should override and/or take the place of an anonymous dongle

bkwny, 2016-02-09:
And if the newly-freed anonymous dongle could be used to satisfy an frequency enabled on the WebUI but not having a dongle, put the anonymous dongle to use for it.

jpoirier, 2016-02-09:
On Mon, Feb 8, 2016 at 8:27 PM, bkwny <notifications@github.com> wrote:

> And if the newly-freed anonymous dongle could be used to satisfy an
> frequency enabled on the WebUI but not having a dongle, put the anonymous
> dongle to use for it.
>
thanks.

I updated the initial post's verbiage. Though the current code will
re-config the anonymous dongle either in the current pass or the next pass,
depending on what frequency the dongle with the id was configured for, it's
not a bad idea to explicitly state the requirement .

commit bec0cef6 (2016-02-26): rework the dongle and settings handling code, fixes #239
closed 2016-02-26 by cyoung