cyoung/stratux #240: CPU load doubles when USB GPS improperly unplugged

Ergonomicmike, 2016-02-08:
Probably not a legitimate issue since users will probably not be unplugging their GPS in flight. And even if they do, it didn't cause a catastrophic failure. And I suspect that's this is only relative to the GlobalSat GPS. (Although since Chris is working on something else that communicates via serial, it could be a legitimate issue down the road.) So I leave this for Chris to close if a non-issue.

I've got a GlobalSat BU-353-S4 GPS that connects physically via USB. Internally, it uses a serial protocol to communicate with the Pi. 

Now, under normal circumstances, my CPU load is 30%, as reported by htop, 

But if I unplug the GPS without first toggling the GPS Off in the Stratux UI settings page, the CPU load doubles. [Here's a screen shot of htop.](http://s14.postimg.org/6qv16xx5t/stratux_gps_unplugged_cpu_issue.jpg)

I apologize for the hard to see load numbers. (I've since changed the color theme on htop.) The core utilization numbers are:
68.6%
62.5%
58.9%
62.6%

(That's with powersave governor limiting the CPU to 600 MHz.)

My uneducated guess is that this is a polling problem, like I used to see when DOS was waiting on an input from the keyboard?

The CPU does not come back to normal after I plug the GPS back in. I have to do a reboot.

AvSquirrel, 2016-02-10:
Grumble, grumble, zombie goroutines. Yeah, this should be fixed.

Apparently the `gpsSerialReader()` goroutine never actually returned / terminated when the GPS was disconnected. It just sat, waiting on a blocked `scanner.Scan()` that never came. Then, my new `isGPSConnected()` function triggered a new instance that sent the old one off to go limbo. The serial port never get closed, so go can't do garbage collection. This can get really ugly if you're talking over the onboard serial port (ttyAMA0) -- multiple instances all try to read the port at the same time, with predictably unpredictable results.

So, connect / disconnect the adapter enough times, and you end up with dozens of zombie subprocesses in top. I didn't see a doubling of CPU load with my u-blox receivers, but it's possible that the driver for the Prolific USB-to-serial bridge didn't play as nicely as the cdc_acm driver. 

Give this commit a try. I've tested with a M8N over serial (RY835AI) and a U7020 over USB (VK-172), but I need feedback on the BU-535 before I'll ask Chris to pull it into the master branch.

https://github.com/AvSquirrel/stratux/commit/211d3a29ed3b3b1ab5c45fdd98b4a7bed1dbbe56

skypuppy, 2016-02-10:
> AvSquirrel@211d3a2

Might be time to re-evaluate using "Go."

Ergonomicmike, 2016-02-10:
@AvSquirrel, I reimaged to v0.6r1, did a git pull, and then did a make with your two files. (BTW, is there an easy way to download just the files from the github? When I try saving the files to my desktop, I get html links to the files. I have to go to "Browse Files" and then I have to download the ZIP and extract the files I want.) Anyway, I rebooted, then plugged and unplugged the GlobalSat BU-353-S4 GPS. htop reports the CPU stays the same (low). And the Stratux status page is now showing when the GPS is connected/disconnected - and reconnected. I would say "ship it."

Ergonomicmike, 2016-02-10:
I've been running v0.6r1 outdoors with the latest (14) commits plus AvSquirrel's two new files for five hours now. (On battery, trying to reproduce Wi-Fi disconnect.) Interestingly, [memory usage](http://s10.postimg.org/jsm5ww8bd/stratux_memory_after_avsquirrel.jpg) is already higher than [my previous test run of 48 hours in the house.](https://github.com/cyoung/stratux/issues/230). But not as high [as after two hours of flying.](http://s27.postimg.org/v3asbplwj/5_stratux_memory_2_hr_flight_uat.jpg)

I expect some of the memory difference is due to ADS-B traffic and towers. I don't get any towers during my ground testing and had a peak UAT of only 60.

Thread count is up with AvSquirrel's changes.

AvSquirrel, 2016-02-11:
PR #243 

Here's my htop on the ground, with AHRS, GPS, and 1090 receiver, after running for an hour:
[image]

And here's the process list after disconnecting / connecting my VK-172 ten times, and setting the wrong baud rate on my RY835AI ten times. No change in the thread count; PID 2395 was the newest gen_gdl90 thread on both snapshots.
[image]

Ergonomicmike, 2016-02-11:
Thanks! (And I started out thinking that this wasn't an important problem.) Closing now that a pull request has issued.

closed 2016-02-11 by Ergonomicmike