cyoung/stratux #241: A/C freezing after landing but not timing out

peepsnet, 2016-02-10:
1. Stratux version: 0.6r1
2. Stratux config: dual SDR, GPS: no , AHRS no.
3. Galaxy Tab A, AvNav/FltPlan GO
4. Description of your issue. If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

I was using stratux at a local airport last night and had an issue with aircraft landing and then getting stuck on the runway and not moving...

An a/c would land... get to under 100knots and then just sit on the runway and not move again. looking at the webiu I would see the time updating for the a/c indicating it was still receiving data from the a/c but no change to the a/c position.  As the a/c cleared the runway and taxied to the gate the AvNav and Fltplan GO continued to show both on the runway. So it appears there is something not updating with the data.

[image]

peepsnet, 2016-02-10:
I could not upload the ZIP so I added ".txt" 

[logs.zip.txt](https://github.com/cyoung/stratux/files/124011/logs.zip.txt)

Ergonomicmike, 2016-02-10:
How is it that one aircraft is below sea level?

Ergonomicmike, 2016-02-10:
If they're both on the ground, why is one at 0 feet & the other at -75?

peepsnet, 2016-02-10:
GPS error.

peepsnet, 2016-02-10:
There are actually about 5 A/C piled up on the ground there.

peepsnet, 2016-02-10:
[image]

cyoung, 2016-02-15:
Do either of these EFBs show ground traffic?

peepsnet, 2016-02-16:
All of the EFB's I use(FtlPlan Go, DroidEFB, AvNav, Avare, Naviator) show the same thing. The position stops updating when the speed is below 100knots(probably the signal to switch from airborne to on the ground) and this is also when the A/C shows up in the EFB's

Ergonomicmike, 2016-02-16:
I haven't looked closely for this, but a couple weeks ago I noticed that I could watch Pipers doing T'n'G's on the runway (1090) on iFly. Is there a ground proximity switch on Air Transports where they stop updating their position after landing?

peepsnet, 2016-02-16:
I was watching the traffic screen in Stratux. The time field was changing(I believe indicating that I was still receiving 1090 hits from the a/c) but the lat/Lon/speed/alt were not updating. 

Its as if this was keeping the a/c in the list but not moving. If there is an airborne flag should the logic remove the a/c from the list when that flag is false? Should we have an option(in settings) to feed this traffic to the efb's and force updates of the lat/Lon/speed/alt if selected/possible?

Axtel4, 2016-02-16:
1090ES only, or also with 978UAT traffic?

peepsnet, 2016-02-16:
Haven't seen any 978 traffic landing at FLL  I might try FXE and see what happens!!

skypuppy, 2016-02-16:
it seems that sometimes the other aircraft does not send any data or
sends the wrong format.

peepsnet, 2016-02-23:
Well I have more info on this issue.  It seems when the a/c switch from airborne to on ground stratux is not removing them from the active list. The a/c's are still transmitting position data but no speed or heading.

This is a list of a few a/c that were arriving in each file:

I am attaching he logs from about 3 hours of flights into and out of FLL

The github system doesnt allow me to upload rar's so I added the .txt extension. Please remove it

[stratux_air-to-gnd-logs.rar.txt](https://github.com/cyoung/stratux/files/143199/stratux_air-to-gnd-logs.rar.txt)

AvSquirrel, 2016-02-25:
#285 will cause targets to properly time out if no valid position fix is received, even if speed or altitude data is being received.

cyoung, 2016-12-01:
@AvSquirrel fixed this a while ago.

closed 2016-12-01 by cyoung