cyoung/stratux #242: Wingx crashes with GPS enabled RY835AI

kkeebb007, 2016-02-10:
Hi everyone,
First off I just wanted to say thank you to everyone who has put in the time and testing for this!

I am using my iPad with the lastest OS along with WingX. I am able to successfully use the AHRS on the ground along with getting a good GPS signal.

Although when I enable gps with or without AHRS enabled and open AHRS in wingx, it immediately crashes. I am able to use wingx with the gps as long as I do not open synthetic vision or attitude.

Now this only happens when I have gps signal. If I block the receiver, to get the msg No GPS, AHRS works....but now I don't have GPS. Once I unblock the receiver and I have a GPS signal, wingx crashes. Again only with the synthetic vision or attitude screen open.

Is this known issue? Am I doing something obviously wrong that I don't see it? This happens either soldered or via usb. Maybe A defective RY835AI?

Any info or direction would be greatly appreciated! I can upload logs if that helps.

Thanks,
Kevin

kkeebb007, 2016-02-10:
Side note - I did speak to WingX tech support, they referred the issue Stratux or the RY835AI

skypuppy, 2016-02-10:
Sounds like a Wingx issue.

skypuppy, 2016-02-10:
But , of course they do.  :)

Sorry, dude.

kkeebb007, 2016-02-10:
Sadly you are correct so far. I was able to get my hands on an IPad mini and WingX and Stratux worked perfectly. 
Now to figure out why it doesn't work on the my iPad...lol

HiltonG, 2016-02-10:
What Stratux version are you running?  Are you able to capture the raw data stream?  I have a small app that'll do that if you'd like. Then send me the data and I will be able to see and hopefully reproduce any problems you're having.

Hilton

ScraboTower, 2016-02-10:
I have the same setup and have no problem, nor can I replicate this issue on my setup

kkeebb007, 2016-02-10:
I have taken several picture to hope help answer your questions. I am not sure which logs you would like to have. I have no problem using your app to capture the data.

Thank you,
Kevin

kkeebb007, 2016-02-10:
To clarify the issue.

When connecting stratux to my iPad Air, it works perfectly until I open the Synthetic Vision or Attitude screen. At that point WingX crashes taking me to my iPad home screen. If I disable the GPS from the browser or have no GPS signal, Synthetic Vision and the attitude screen works. Once I enable the GPS within Stratux (192.168.10.1) or I place the GPS receiver at a window to receive a signal, WingX crashes. I have reinstalled WingX, tried different setting on my iPad but with the same crashing result.

Meanwhile, on my iPad mini, everything works flawlessly. 
Same issue happens with the RY835AI hard wired or usb to the stratux.

Again thank you,
Kevin
480.440.3773

kkeebb007, 2016-02-10:
[image]
[image]
[image]
[image]

ScraboTower, 2016-02-10:
You tried starting WingX in safe mode first ?

kkeebb007, 2016-02-10:
Just tried with the iPad in safe mode. No go.
Unless you are talking about WingX being in safe mode, which I do not know how to do.

jzeevi, 2016-02-10:
How do you put an iPad in safe mode?

kkeebb007, 2016-02-10:
https://youtu.be/nejNvK_mNuE

AvSquirrel, 2016-02-10:
@kkeebb007 -- so you jailbroke the iPad (CriOS; safe mode, etc.)?

kkeebb007, 2016-02-10:
Negative. My experience is more with android than mac/apple. But by doing
as this person demonstrates, it should be in safe mode....there was no
indication on his or mine that it is safe mode.

Regardless, how do you start WingX in safe mode?

HiltonG, 2016-02-11:
WingX Pro7 Safe Mode: Tap on iPad Settings (not in WingX Pro7), tap on WingX Pro7, scroll down to the bottom of WingX Pro7's options (on the right), turn on Safe Mode.  Restart WingX Pro7 (be sure you close it first; i.e. double-tap Home button slide WingX Pro7 up, then run it again).

Does this fix the problem?  Either way, turn off Safe Mode otherwise you'll lose various settings each time you restart WingX Pro7.

Hilton

kkeebb007, 2016-02-11:
Just tried it in safe mode. Unsuccessful. Same crashing result.

cyoung, 2016-03-05:
@kkeebb007 - maybe you can take a screen capture of this, from start to crash?

Anyone else having an issue?

ScraboTower, 2016-03-05:
Nope and I have been using every version  from the latest release to the latest beta.

cyoung, 2016-12-01:
@kkeebb007 - fixed? If no, please re-open.

closed 2016-12-01 by cyoung