cyoung/stratux #247: GPS issues in 0.6r1

braddward, 2016-02-13:
Hi all,

I was finally able to fly this morning and despite frozen toes was able to capture some data.  The air vents on the Skyhawk seemed to be working too well - much better than the heater.

Unfortunately the GPS stopped working.  I rebooted and also had my passenger pull the plug and recycle to no avail.

Of course after I got home its working fine from the desk.

This is also the first flight with the dmurray antennas.  I had UAT off and on today (probably more off than on).  It seems that with the stock antennas I got better reception.

I've also noticed that the AHRS pitch and roll are now way off on this version.  Not sure if that's my chip or that this is still being worked on (previous versions seemed to work).
I noticed lots of invalid NMEA messages in the stratux.log (snippet below)

 2016/02/13 17:08:40  - Last GPS fix: 51 minutes ago, GPS solution type: 0 using 0 satellites (0/0 seen/tracked), NACp: 0, est accuracy 0.00 m
2016/02/13 17:08:40  - GPS vertical velocity: 0.00 ft/sec; GPS vertical accuracy: 0 m
2016/02/13 17:08:47 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:08:48 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:08:49 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:08:54 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:08:54 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:08:56 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:08:57 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:08:58 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:09:02 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:09:02 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:09:03 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:09:07 GPS error. Invalid NMEA string: Invalid NMEA message
2016/02/13 17:09:10 stats [started: 52 minutes ago]
2016/02/13 17:09:10  - CPUTemp=32.55 deg C, MemStats.Alloc=5.0 MB, MemStats.Sys=14 MB, totalNetworkMessagesSent=45,890

Log files attached.
[logs.zip](https://github.com/cyoung/stratux/files/129350/logs.zip)

Ergonomicmike, 2016-02-13:
Since I'm about to go flying, was this a vanilla v0.6r1? Or does it have the latest commits?

And do you get the invalid NMEA's only when flying? Or at your desk too? (Although I don't know how the Stratux would know the difference.)

Ergonomicmike, 2016-02-13:
Since you remarked about frozen toes, could the Stratux (or your GPS) have been too cold and out of temp spec? Condensation inside something?

braddward, 2016-02-13:
Hi Mike,

It was plain vanilla v0.6r1.  I don't think it was the cold but who knows.  It was inside and in a warm vehicle prior to the flight so I don't think it was that cold to have caused condensate.

The CPU temp was 35C according to the logs.

Looking at the log it looks like I've had that a few times before, just not to this degree.  It looks like it's on startup and then goes away.  On this flight it kept repeating.

Have a safe and fun flight!

Ergonomicmike, 2016-02-13:
Thanks. If I get invalid NMEA messages in flight, will I notice it at the Presentation/Application level? That is, will my ownship jerk or something? (Might be a function of EFB.) Or will I only notice invalid NMEA if I look at the gps log?

braddward, 2016-02-13:
I just noticed I lost my position in Foreflight.  I went to the web status page and GPS showed no satellites.

224XS, 2016-02-13:
Silly question but does the GPS module face upward and have absolutely nothing between itself and a wide angle view of the sky save a few thin layers of plastic and plexiglas? For example if the Stratux is in the back seat of a 172 the GPS would absolutely not work.  On the glare shield, facing up you get may 200 degree field of view which is probably enough, most of the time.

The 3rd party UHF antenna is not a factor, except insofar as it is a placebo in this application for relevant air-to-air 978 traffic and ground tower reception. Might help a tiny bit for picking up FL420 traffic at
150 SM, but who cares.

braddward, 2016-02-13:
It does face up correctly and has worked well until this update and changing the antenna at the same time. 

It was initially in the back seat and that's where I have generally had it. But when it wasn't working I had my passenger put it on the front dash and that didn't help.  

Generally at home I'll pick up gps in the house.

AvSquirrel, 2016-02-13:
For posterity's sake, the logs indicate that you're running with:
- v0.6r1 release image https://github.com/cyoung/stratux/commit/bbbcd3bbe16f67febb9955ff7d92908b82d2fa24
- a u-blox GPS (RY835AI module?) connected via /dev/ttyAMA0 (i.e. onboard GPIO)

During the run starting at 1553Z, the GPS was communicating with the Pi, but never got a position fix.

````
Excerpt from 0013-gps.log

12208464787,$PUBX,03,11,8,-,301,13,,000,10,e,322,65,,000,12,-,109,-1,,000,14,-,242,48,,000,15,e,058,21,,000,18,e,074,70,,000,20,e,107,10,,000,21,e,174,44,,000,22,-,293,17,,000,24,e,074,46,,000,27,e,270,18,,000*59
12254054006,$PUBX,00,155351.40,4128.55351,N,09026.41639,W,185.677,NF,2065001,1460177,0.000,0.00,0.000,,99.99,99.99,99.99,0,0,0*38
12270632391,$PUBX,04,155351.40,130216,575631.39,1883,17,-272,-208.000,21*2F
````

I spot checked a few of the PUBX,03 (satellite information) messages during that log, and you were receiving weak signals from only 1-2 satellites at a time. It was enough to set the clock, but never enough to generate a position fix. I'll echo @224XS ' comment -- how / where was the GPS mounted -- not only within the airplane, but also relative the RPi? 

Anyway, it looks like you ran for about 50 minutes without a position lock, and then rebooted a couple times. On the second reboot where you were seeing the continuous NMEA messages, the Pi was getting gibberish over the serial port:

````
0015-gps.log

17955638535,%бФ▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!c%1▒!!▒▒б▒б5!Ж╠%1▒Ф╓Ф─1!1Х15б1!!беее!1ъ$1▒%%Ф─1!1Х15б1!!беее!1у$1▒%%Ф─1!1Х15б1!!беее!1╘$1▒!╓Ф─1!1Х15беб%1бее!1№$1▒%%Ф─1!1Х11е%е!беее!1№$1▒%%Ф─1!1Х11е%е!беее!1№5еебе5c%1▒%%Ф─1!1Х11е%е!беее!1№$1▒!╓Ф─1!1Х11е%е!беее!1╒$1▒%%Ф─1!1Х11е%е!беее!1№$1б!╓Ф─1!1Х11е%е!беее!1╠%бФ▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!c%1б!!▒▒б▒е5!F╠%1бФ╓Ф─1!1Х11е%е!беее!1№$1б%%Ф─1!1Х11е%е!беее!1╒$1б%%Ф─1!1Х15б1!!беее!1╘$1б!╓Ф─1!1Х15б1!!беее!1╨$1б%%Ф─1!1Х15б1!!беее!1у$1б%%Ф─1!1Х1─е%е!беее!1№5еебе5c%1б%%Ф─1!1Х1─е%е!беее!1╒$1б!╓Ф─1!1Х1─е%е!беее!1№$1б%%Ф─1!1Х1─е%е!беее!1№$1б!╓Ф─1!1Х1─е%е!беее!1╠%бФ▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!c%1б!!▒▒б▒─5!╠%1бФ╓Ф─1!1Х11е%е!беее!1╒$1б%%Ф─1!1Х11е%е!беее!1№$1б%%Ф─1!1Х11е%е!беее!1№$1б!╓Ф─1!1Х11е%е!беее!1╨$1б%%Ф─1!1Х11е%е!беее!1╨$1б%%Ф─1!1Х11е%е!беее!1№5еебе5РбЖ%%Ф─1!1Х11е%е!беее!1№$1б!╓Ф─1!1Х11е%е!беее!1╘$1б%%Ф─1!1Х15б─ДЖ%еее!1у$1▒!╓Ф─1!1Х15б─ДЖ%еее!1Р !е1▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!c%1▒!!▒▒б▒Ж5!Ж╠%1▒Ф╓Ф─1!1Х15б1!!беее!1№$1▒%%Ф─1!1Х15б1!!беее!1╒$1▒%%Ф─1!1Х11╡%е!беее!1╒$1▒!╓Ф─1!1Х11╡%е!беее!1№$1▒%%Ф─1!1Х11╡%е!беее!1№$1▒%%Ф─1!1Х11╡%е!беее!1╒5еебе5
18954499576,%1▒%%Ф─1!1Х11╡%е!беее!1№$1▒!╓Ф─1!1Х11╡%е!беее!1№$1▒%%Ф─1!1Х11╡%е!беее!1╒$1▒!╓Ф─1!1Х11е%е!беее!1╠%бФ▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!c%1▒!!▒▒б▒Д5!Ж╠%1▒Ф╓Ф─1!1Х11е%е!беее!1№$1▒%%Ф─1!1Х11е%е!беее!1╒$1▒%%Ф─1!1Х15беб%1бее!1╨$1▒!╓Ф─1!1Х15беб%1бее!1╘$1▒%%Ф─1!1Х15беб%1бее!1у$1▒%%Ф─1!1Х11е%е!беее!1ъ5еебе5
27954919364,%1▒%%Ф─1!1Х11е%е!беее!1ъ$1▒!╓Ф─1!1Х11е%е!беее!1╨$1▒%%Ф─1!1Х1!б1!!беее!1ъ$1б!╓Ф─1!1Х1!б1!!беее!1c%бФ▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!c%1б!!▒▒б▒╞5!Ж╠%1бФ╓Ф─1!1Х1!б1!!беее!1╨$1б%%Ф─1!1Х1!б1!!беее!1╒$1б%%Ф─1!1Х1!б1!!беее!1╒$1б!╓Ф─1!1Х1!б1!!беее!1ъ$1б%%Ф─1!1Х1!б1!!беее!1╘$1б%%Ф─1!1Х1!б1!!беее!1ъ╡еебе5РбЖ5!%Ф─1!1Х1!беб%1бее!1╨$1б!╓Ф─1!1Х1!беб%1бее!1ъ$1б%%Ф─1!1Х1!беб%1бее!1ъ$1б!╓Ф─1!1Х1!беб%1бее!1Р▒Ф1▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!c%1б!!▒╡б▒Д5!╞╠%1бФ╓Ф─1!1Х1!б1!!беее!1╒$1б%%Ф─1!1Х1!б1!!беее!1№$1б%%Ф─1!1Х1!б1!!беее!1№$1б!╓Ф─1!1Х1!б1!!беее!1╒$1б%%Ф─1!1Х1!!еб%1бее!1№$1б%%Ф─1!1Х1!!еб%1бее!1╒╡еебе5Р !╡%%Ф─1!1Х1!!еб%1бее!1№$1б!╓Ф─1!1Х1!!─ДЖ%еее!1╨$1б%%Ф─1!1Х1!!─ДЖ%еее!1у$1б!╓Ф─1!1Х1%!─ДЖ%еее!1╠%бФ▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!Р !5!!▒╡б▒Д5!Ж╠%1бФ╓Ф─1!1Х1%!1!!беее!1№$1б%%Ф─1!1Х1%!1!!беее!1╒$1б%%Ф─1!1Х1%!1!!беее!1╒$1б!╓Ф─1!1Х1%!1!!беее!1№$1б%%Ф─1!1Х15!1!!беее!1у$1б%%Ф─1!1Х15!1!!беее!1╒5еебе5╠%1б%%Ф─1!1Х1─!б%1бее!1ъ$1б!╓Ф─1!1Х11!б%1бее!1№$1б%%Ф─1!1Х11!б%1бее!1№$1б!╓Ф─1!1Х11!б%1бее!1c%бФ▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!Р !5!!▒Фб▒%5!F╠%1бФ╓Ф─1!1Х11!б%1бее!1у$1б%%Ф─1!1Х11!б%1бее!1╒$1б%%Ф─1!1Х15!1!!беее!1╘$1б!╓Ф─1!1Х15!1!!беее!1╨$1б%%Ф─1!1Х11!б%1бее!1╨$1б%%Ф─1!1Х11!б%1бее!1╒5еебе5РбЖЖ▒Ф─1!1Х11!б%1бее!1╨$1б!╓Ф─1!1Х11!б%1бее!1№$1б%%Ф─1!1Х11%б%1бее!1╨$1▒!╓Ф─1!1Х11%б%1бее!1Р▒Ф1▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!Р !5!!▒Фб▒%5!Ж╠%1▒Ф╓Ф─1!1Х15!1!!беее!1ъ$1▒%%Ф─1!1Х15!1!!беее!1у$1▒%%Ф─1!1Х11%б%1бее!1у$1▒!╓Ф─1!1Х15!еб%1бее!1№$1▒%%Ф─1!1Х15!еб%1бее!1╒$1▒%%Ф─1!1Х15!еб%1бее!1у5еебе5c%1▒%%Ф─1!1Х1─Фб%1бее!1╨$1▒!╓Ф─1!1Х11Фб%1бее!1№$1▒%%Ф─1!1Х11Цб%1бее!1╒$1▒!╓Ф─1!1Х11Цб%1бее!1Р▒Ф1▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!Р !5!!▒Цб▒5%5PРбЖФ╓Ф─1!1Х11Цб%1бее!1у$1▒%%Ф─1!1Х15!1!!беее!1№$1▒%%Ф─1!1Х15!1!!беее!1╒$1▒!╓Ф─1!1Х11Фб%1бее!1№$1▒%%Ф─1!1Х11Фб%1бее!1у$1▒%%Ф─1!1Х11Фб%1бее!1╒5еебе5Р !5%%Ф─1!1Х11Фб%1бее!1№$1▒!╓Ф─1!1Х15!1!!беее!1№$1▒%%Ф─1!1Х15!1!!беее!1╒$1б!╓Ф─1!1Х15!1!!беее!1╠%бФ▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!c%1б!!▒Цб▒%5!╞╠%1бФ╓Ф─1!1Х11Цб%1бее!1№$1б%%Ф─1!1Х15!1!!беее!1ъ$1б%%Ф─1!1Х15!1!!беее!1ъ$1б!╓Ф─1!1Х1─Фб%1бее!1№$1б%%Ф─1!1Х11Фб%1бее!1№$1б%%Ф─1!1Х11Фб%1бее!1№5еебе5РбЖ5!%Ф─1!1Х11Фб%1бее!1№$1б!╓Ф─1!1Х11еб%1бее!1№$1б%%Ф─1!1Х15!еб%1бее!1╒$1б!╓Ф─1!1Х11еб%1бее!1Р▒Ф1▒!е▒▒!б!е!б!б!▒б▒!б▒б!б1▒!б!б!б1б!▒!▒!б!б!б1б!▒!б!б!▒!c%1б!!▒Фб▒Д5!Ж╠%1бФ╓Ф─1!1Х11─ДЖ%еее!1ъ$1б%%Ф─1!1Х11─ДЖ% [... cut ...] !▒!б!б!б1б!▒!б!б!▒!c%1▒!!▒Фб▒%5!╞╠%1▒Ф╓Ф─1!1Х1!!1!!беее!1╨$1▒%%Ф─1!1Х1!!1!!беее!1№$1▒%%Ф─1!1Х1!!1!!беее!1№$1▒!╓Ф─1!1Х1!!1!!беее!1№$1▒%%Ф─1!1Х1!!еб%1бее!1у$1▒%%Ф─1!1Х1!!еб%1бее!1у5еебе5
````

That's a fairly good sign of a baud rate mismatch between the GPS and Pi. I've seen similar errors when the GPS is transmitting at 38400, but the GPS is receiving at 9600. A hard power cycle is usually a sure-fire way to resolve; commits post-0.6r1 will also attempt to reinitialize the GPS connection.

wakkow, 2016-02-13:
> This is also the first flight with the dmurray antennas. I had UAT off and on today (probably more off than on). It seems that with the stock antennas I got better reception.

As an aside, my first longer flight with the dmurray antennas and I noticed the same thing. :disappointed:

braddward, 2016-02-13:
This is the case I had printed:
http://www.thingiverse.com/thing:1013857

The GPS chip is mounted in the top lid with the antenna for it facing up.
I generally have the stratux in the back seat and it's always worked well
(since the last few updates).   When I noticed it wasn't working it was
moved to the front dash to see if that would resolve it and power cycled.

I'll double check my connections.  Perhaps one of the connections is
marginally connected now since everything was moved around installing the
new antennas.    Seems weird that it is working fine on the desk.

Axtel4, 2016-02-14:
@braddward 

> This is also the first flight with the dmurray antennas. I had UAT off and on today (probably more off than on). It seems that with the stock antennas I got better reception.

Attracted is a Tower Signal Strength plot for log file 0013-uat.log.  It shows that you were receiving the signal from a tower located closed to Hillsdale, IL at a moderate signal level.  In my experience, typically signal strengths above 50 will provide solid Tower data to the EFB App.  

This log is only about 14 minutes long and it does shows some drops in signal between 5286 and 5288 minutes and between 5288 and 5290 minutes.  This could be normal as your relative position to the tower can affect the signal strength.

Also looking at your UAT log I see that you were tracking 51 unique ADS-B Downlink messages from UAT equipped aircraft.  I do not know what your antenna configuration was when this log was recorded so I don't have anything to compare against.  This is the only UAT log in your zip file that had any data.

[image]

braddward, 2016-02-14:
Very cool data.  Thanks

cyoung, 2016-02-16:
@braddward - v0.7b1 for the GPS improvements mentioned by @AvSquirrel.

Also see the settings page for the update function - another UI feature that needs fixing up if you're interested...

braddward, 2016-02-16:
Great.  I'll take a look at both.  Thanks

braddward, 2016-02-16:
Initial couch flight testing is looking good.  GPS is locking on 7 satellites (by a window) and both FlyQ EFB and Foreflight are showing 1090 traffic. 

Also ran through the update process which also seemed to work well.

Ergonomicmike, 2016-02-25:
@Axtel4 You said that signal strength over 50 is generally good for tower data. Do you know if those calculations changed in v0.7b3? I got a message from iFly today that I had lost a tower connection. Signal strength was about 400 and below. iFly reported the connection back when the signal strength of 500. Maybe the calculations changed by a factor of 10?

Axtel4, 2016-02-26:
Its about 25 - 50 per the UAT log.  Reference the attached signal strength plot.  The signal values were taken  form the UAT log.  The peaks and dips were because of driving over hills and through valleys.

[image]

cyoung, 2016-02-28:
Fixed post-v0.6r1, also some GPS reception issues.

Please use the issue format next time to document the hardware, version, and steps to reproduce - that way we don't have to dig.

closed 2016-02-28 by cyoung