cyoung/stratux #250: Test network packet format changes.

cyoung, 2016-02-15:
#249. Any EFB issues?

Axtel4, 2016-02-15:
I'll download and test DroidEFB today.

peepsnet, 2016-02-15:
Fltplan Go, AvNav, and Naviator all seem to display 1090 traffic just fine. I am not receiving any 978 from the ground at my house.

Axtel4, 2016-02-15:
Seems to be working with DroidEFB and 1090ES traffic (tracking 31 targets).  Will head toward the airport later today to test 978UAT.

[image]

peepsnet, 2016-02-15:
AVARE seems to be acting normal on 1090. Haven't been able to receive any 987 data at my location yet

Ergonomicmike, 2016-02-15:
iFly latest beta (57b) - 1090 on the ground. Seems to work okay. Call signs took about a minute or two to populate. But now that they're there, no problems. (Could just have been an initialization thing with iFly. First time I've really studied Traffic right after start up.) Am waiting for UAT traffic to fly by.

Ergonomicmike, 2016-02-15:
Got UAT direct.(Albeit 1, with a max UAT rate of 47.) Paints okay in iFly.

Ergonomicmike, 2016-02-15:
BTW, I had a couple of AvSquirrel's test files in my file structure. The new build process called them out, told me to stash them, and then aborted. I'm thinking that's a good thing. (Integrity check?)

skypuppy, 2016-02-15:
Mike, I found the nearest 978 tower to me, which was on the way to
somewhere I was going, and spent about a half hour watching, from inside
the truck.  Very informative.

cyoung, 2016-02-15:
Update to v0.7b1 to test:

`service stratux stop ; wget -O - https://github.com/cyoung/stratux/releases/download/v0.7b1/update-v0.7b1-02bf2251d8.sh | sh ; reboot`

Ergonomicmike, 2016-02-15:
Average CPU load at home is down from 30% to 20%.

I'm probably fooling myself into thinking that the blue light on the Edimax is blinking more deliberately. (It might be cooler too.)

Nokomis449, 2016-02-16:
I used:

service stratux stop ; wget -O - https://github.com/cyoung/stratux/releases/download/v0.7b1/update-v0.7b1-02bf2251d8.sh | sh ; reboot

to update with no issues once I prefaced it with "sudo su -" (operator headgap error).  Back porch testing shows the normal 1090 traffic and the VK172 got a pretty quick WAAS lock, so pending a UAT flyover I'd say all systems are nominal.  

Looking forward to testing the update function.  Great work!

Axtel4, 2016-02-16:
No issue with DroidEFB.  Also, no disconnects this evening during drive and flight testing.

AvSquirrel, 2016-02-16:
Great feedback so far. What @cyoung and I are trying to pin down is:
- Does the traffic display in the EFBs match the list of targets in the web UI traffic page, and
- Are you seeing the full FIS-B weather data stream on your EFB (METARs, TAFs, NEXRAD, etc.), especially after switching tasks or resuming from suspend.

I'll add the Android version of Fltplan GO to the "works" list. It's slow, but it shows all of the traffic targets I threw at it, and seemed to show all of the available weather data when I replayed a uat.log.

Ergonomicmike, 2016-02-16:
Match the list in number? Or in lat/long? I'm guessing the latter. 

As for FIS-B, I'll try to sneak a flight in Thursday.

pp4life, 2016-02-16:
Seems to work in Foreflight with 1090 traffic. The headings in Foreflight are 3° less on all aircraft from what is shown in the Stratux WebUI. I hadn't noticed if that was the case with previous releases. If you want me to check on that, I can pop in a 0.6r1 SD card and wait for some more traffic to fly by. I hope these changes fix the crashing WiFi issue. I had the WiFi crash as I was flying very near an ADS-B tower and I assume it was pulling all kinds of data and crashed the WiFi. 

[image]
[image]

Ergonomicmike, 2016-02-25:
Unfortunately, still got a blue light of death on the Edimax today. It happened with two tablets and when tower signal strength dropped below 400 and iFly started reporting Connection lost with the towers. This is with my voltage bypass for the USB. I plan to fly again tomorrow with only one tablet to see if this problem is following multiple clients. I will also send the iFly adsbraw.bin file to iFly and ask if they see any clues as to why the disconnect.

FWIW, I checked the stratux.log. Not quite the same NAK stuff as the last time this happened.

This time it says

`
2016/02/24 19:48:39 spsSerialReader () scanner loop iteration -=63600

And then an Invalid NMEA error. And then Entered UAT shutdown.

cyoung, 2016-03-08:
Thanks for testing this, everyone. This was a big improvement.

closed 2016-03-08 by cyoung