cyoung/stratux #252: v0.7b1 Question

Axtel4, 2016-02-16:
I want to see if I am doing this correctly.

The Web UI shows v0.7b1:

[image]

However in the Raspberry Pi git describe and git log show:
>root@raspberrypi:~/stratux# git describe --abbrev=0 --tags
>v0.6r1
>root@raspberrypi:~/stratux# git log -n 1 --pretty=%H
>a4f5696a297c818f3c572fb3fe0625c0b0119614
>root@raspberrypi:~/stratux#

Did Stratux move to some other place than /root/stratux?

Attached is a screen shot of the update sequence.
[v0.7b1_update.txt](https://github.com/cyoung/stratux/files/131821/v0.7b1_update.txt)
 
Also, in Chrome the "Choose File" button is truncated:
[image]

Thanks in advance.

cyoung, 2016-02-16:
If it is showing v0.7b1 in the web interface then you've upgraded - so that seems OK.

The upgrade file just contains binaries and won't upgrade your source tree. You will have to maintain that separately - git pull, etc. Our source tree right now is about 95M uncompressed, 51M compressed. Without it, the update is 6.1M.

Axtel4, 2016-02-16:
So, does it hurt anything to git pull / make / make install in parallel to keep the source tree up to date?

cyoung, 2016-02-16:
No, it will not hurt. The "make install" method will keep you up-to-date with stratux software and related libraries. System settings aren't managed with the Makefile, so if you don't update via an image or the update files and we make system-wide changes (dhcp config, sysctl settings, etc) then you'll miss those changes.

Ergonomicmike, 2016-02-16:
My "Choose File" button is cut off after the F.

cyoung, 2016-02-16:
Sounds about right, that's my best interface work.

cyoung, 2016-02-16:
Reopen if any further issue(s).

closed 2016-02-16 by cyoung