cyoung/stratux #254: Stratux WebUI Stops Showing Traffic

pp4life, 2016-02-17:
1. Stratux version: 0.7b1
2. Stratux config: dual SDR, GPS no, AHRS no, battery Anker, battery cable, Anker Standard
3. EFB app and version Foreflight 7.5.2. EFB platform (iOS 8.3 on iPad Mini 2, iOS 9.1 on iPod Touch 5th Generation). EFB hardware iPad Mini 2, iPod Touch 5th Gen.
4. Description of your issue.
See linked video.
https://youtu.be/8MTvH-tOQk4
       Ignore the Skywest target in the upper left of Foreflight. It got stuck there, I don't know why, but it was still there after the lower right target went out of range. Took a restart of Foreflight for it to go away.
        All webUI stuff is done on the iPod Touch, and all Foreflight stuff is done on the iPad mini. But the the general problem is that the traffic page on the Stratux WebUI stops showing traffic, but Foreflight will continue to show the traffic.
       If I tap menu on the WebUI and switch over to the Status page then flip back to the Traffic page it will again show the traffic that I'm seeing in Foreflight. I would have never noticed without having two devices sitting right next to each other so I could test for issue #250. It's like the Traffic page stops updating.

AvSquirrel, 2016-02-17:
Noticed this last night while debugging #251.

Every minute, [this bit of Javascript](https://github.com/cyoung/stratux/blob/master/web/plates/js/traffic.js#L110-L126) looks for any traffic entries that haven't been refreshed in more than three minutes. Problem is that JS uses runs on the client (i.e. iPad), and therefore uses that device's time / date. Traffic time is based on an uptime counter that starts at 0, and therefore times out all traffic each time it's called.

This doesn't affect the targets send to ForeFlight, and traffic does come back on the UI after a second or two, but it should (and will) be fixed.

Ergonomicmike, 2016-02-17:
In the comment in the java script code above it says "stail aircraft." I politely ask if that's supposed to be "stale"?

bradanlane, 2016-02-17:
there are a few things happening here. First, the web ui is attempting cleanup after 180 seconds whereas the stratux system code is every 60 seconds. this is an easy fix.

the bigger challenge is the definition of "time". this one keeps biting stratux. there are several definitions of time:

"Stratux Time" - the clock time on the Raspberry Pi, which is undefined if we don't have a gps - it could be just about anything
"Data Time" - the time stamped by the FAA
"Receive Time" - when the data was received by Stratux and is stamped using "stratux time"
"Device Time" - the time that the mobile device thinks it is

The best or worst we can do is perform all WebUI calculations based on "Stratux Time". This just requires the getStatus() interface to include whatever the raspberry pi thinks the time is.

skypuppy, 2016-02-17:
@bradan, I think we need a better "time" philosophy.  Partying puns
aside, we need standardization, and across all equipment.  I think we're
going to have to make it a requirement for all users/buyers of stratux
equipment.  If  they don't have a gps and/or can't wait for gps to
lock-in to get the actual time, then they must get a battery backed-up
RTC chip.  Alternatively, the Pi might be able to set it's time via the
web interface from the tablet or what-have-you on the other end.  (A
sanity check would be needed in the Pi code but not a lot of effort
needed for that.)  2nd alternative, the user must set the Pi time via
his watch before any other operations begin.
And that is a key phrase -- "no stratux operations to begin until Pi
time is correct or at least matched up with *something* external."  This
is going to bite us in the rear end in the long term if we don't.

We could insert a small task in front of stratux that checks current Pi
time against Pi time stored from last stratux run.  A suitable delay
needs to be determined.  I suggest the stored time format be UTC, ZULU,
or normalized system clock ticks, which is a very large integer (can we
even get that from Raspbian?)

Sure wish Pi2 had a battery backed-up RTC built-in.  <sigh>

Skypuppy

bradanlane, 2016-02-17:
The "time problem" is really only in the webUI and it is of our own making. The data on the EFB App is correct. 

If we (me) had not attempted to display so much info in the web experience, none of this would be under discussion.

Speaking personally, I want Stratux to be lite and simple. I'm happy to strip the webUI way back to the bare minimum of setup and diagnostic.

Axtel4, 2016-02-17:
>"Data Time" - the time stamped by the FAA

This is the time the FAA puts on the Up Link FIS-B data.  For example it is the METAR timestamp such as: 171753Z.

In actuality, does it matter if the Pi has a RTC?  (Other than building the code on the Pi.) What most users are interested in, is knowing how old the WX data is on the EFB.  This is accounted for in the EFB by comparing the connected devices internal clock to the time stamped WX data.

Maybe if the time in Stratux is that important to users the cheapest solution is to require a GPS reciever for a Stratux build.  What I have seen is the time on my system is synced to the GPS each time I start Stratux.  With the GPS, I have the correct time with or without an internet connection.

cyoung, 2016-02-17:
>"Stratux Time" - the clock time on the Raspberry Pi, which is undefined if we don't have a gps - it could be just about anything

@bradanlane - this has changed recently. Around 0.5b5 I switched all clock references in the code to "stratuxClock" time. This is actual time since start of stratux and is not affected by system time shifts (after getting a GPS lock, etc). So things like uptime are now reporting reliably.

How about in the TrafficInfo struct that I just add a "SecondsSinceLastUpdate" field and you display just that in the column (if not eliminated)? That plus a "StaleTarget" bool which gets sent once over the websocket when stratux stops tracking the target - so that we're not maintaining two different "stale" decision-makers.

bradanlane, 2016-02-18:
Quick question: if stratuxTime is "time since start" then if it is treated as a real datetime does it compute the same as epic (time since 1-Jan-1970)?

As for displaying a delta, that sounds like the simplest option.

skypuppy, 2016-02-18:
@bradanlane, What I'm saying is the "time" discrepancy is going to bite
us in the butt if we don't standardize.

Skypuppy

skypuppy, 2016-02-18:
If the GPS is in a cold start, it can take, by specification, up to 45
MINUTES (and in some cases -- *hours*) to get a fix.  Very few pilots
know that and even if they did, who wants to wait around on the flight
line waiting for the GPS to find itself (not metaphorically speaking)?

Skypuppy

cyoung, 2016-02-18:
We don't use GPS time nor RPi system time for these functions anymore, @skypuppy.

skypuppy, 2016-02-18:
@cyoung
So raspbian linux can give you clock ticks since start?  I wasn't sure.

Thanks,
Skypuppy

cyoung, 2016-02-18:
@bradanlane - here's what it looks like after 30 seconds uptime:

`fmt.Printf("%v\n", stratuxClock.Time)`
Output:
`0001-01-01 00:00:30.1 +0000 UTC`
and
`fmt.Printf("%d\n", stratuxClock.Time.Unix())`
Output:
`-62135596770`

So no, it's not time since epoch per se. You could extract the equivalent by getting stratuxClock.Time.Seconds().

cyoung, 2016-02-18:
Fixed with PR #259.

closed 2016-02-18 by cyoung