cyoung/stratux #255: WebUI:  On Firefox on Ubuntu, Weather and Traffic screens are stuck at "Disconnected" and Browser Console claims connections were interrupted

bkwny, 2016-02-17:
1. Stratux version:
0.7b1+11 commits

2. Stratux config: single/dual SDR, GPS yes/no (type?), AHRS yes/no, battery, battery cable.
Dual SDR Nano 2.  RY835AI via GPIO.  No AHRS.  External power.

3. EFB app and version (e.g., WingX Pro7 8.6.2). EFB platform (e.g., iOS 9.2). EFB hardware (e.g., iPad Mini 2).
N/A for this issue.

4. Description of your issue. If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

https://www.reddit.com/r/stratux/comments/45zfi4/weather_disconnected_or_connected/

FYI in the browser window titled "Browser Console" (BC) I also see occasionally:
x. The connection to ws://192.168.1.106/status was interrupted while the page was loading.
For example, in the left sidebar click on "Weather" and observe that the red icon says "Disconnected".  While I am observing this I see in the BC window the error about /status getting interrupted.  So click on "Status" and observe its icon is red.  But in a very short period, less than one second, its icon will turn green and say "Connected".  But same won't happen for Weather or for Traffic.  They are stuck at the red "Disconnected".

None of this happens on Chrome on Win 7.  On that setup the icons stay green, "Connected".

Ergonomicmike, 2016-02-17:
I saw the same Wx "Disconnected" behavior yesterday in the car using my (old) Nook HD+ with Dolphin browser. I figured it might be browser specific, because, on the GPS page, where I usually see my rough location on a map, there was no target symbol. Also, there was no AHRS indicator. Instead, there was text saying that I needed a (presumably newer) browser that supported GL.(Or words to that effect.)

cyoung, 2016-03-08:
I do believe that this was fixed, correct?

bkwny, 2016-03-08:
Yes, it is repaired.  OK to close.

closed 2016-03-08 by cyoung