cyoung/stratux #256: File system corruption on power down

jzeevi, 2016-02-17:
Using .7b1, I flew for .9 hours with no apparent issues (although ADS-B towers kept disappearing and re-appearing). When the flight completed, I simply removed power to the Stratux (as is usually what I do).

This morning, I wired up a fan to the case I built and when I attempted to restart the device, it would not boot (as evidenced by the green light on the RPi dropping much quicker than it does when the system boots and the lack of the SDR's or Wifi chip to come up).

After several restart attempts, I re-imaged my microSD card and things are back to normal.

Obviously, I can shutdown with the UI in the future, but agree with past posters that it might be wise to have an RO parition and a RW partition that can always be rebuilt if it is corrupted.

cyoung, 2016-02-17:
>although ADS-B towers kept disappearing and re-appearing

Are you referring to the ForeFlight display? If so, that's an app issue - they changed the way they display towers a few versions ago.

jzeevi, 2016-02-17:
Thanks - yes, I'm using FF. I forgot to mention that above. Do you have any insight as to why they now come and go?

And, FWIW - I wanted to justify this as a bug report as it actually crashed the RPi. I know there's an enhancement request but I wanted to track an actual failure. :-)

cyoung, 2016-02-17:
No clue. Use our web interface to get accurate reception info.

jzeevi, 2016-02-17:
LOL. Thanks. I'll ask (the QA manager is a friend)

closed 2016-02-18 by cyoung