cyoung/stratux #262: support for wifi "profiles"

dickinsonmg, 2016-02-18:
Hi,

Wondering if it's possible to add the concept of "wifi profiles", with support of switching profile (and adding?) via the web-interface.

For example, one profile would be the standard "stratux" SSID as is default at present. A second profile that is creatable by the user would allow the unit to join a wifi network (like a home network or personal hotspot) to allow updating without opening cases/plugging in ethernet. Once an update is complete, then the user could switch profiles back to the default "stratux" profile.

The profile switcher could be responsible for ensuring the appropriate configuration files are in place, followed by a "reboot" to that profile. This would alleviate any problems with the network being in a strange state with regard to static IP config, then switching to DHCP, etc and having to figure out state.

thanks.

bkwny, 2016-02-23:
+1.  SSID, whether hidden, channel number, encryption level, password, list of authorized MAC addresses.  Or, if client mode, a list of known SSIDs along with the credentials needed to connect to them.  IMHO the AP stuff is more important than the client stuff.

cyoung, 2016-03-08:
Consolidating to #228.

closed 2016-03-08 by cyoung