cyoung/stratux #267: Lost ground link multiple times in v0.7b2

GrayFlyer, 2016-02-18:
I flew v0.7b2 today with iFly on my '720 and a Android device.  In less than one hour I lost ground reception at least 10 times on both devices while receiving 4 to 6 towers.  It always restored connection without any intervention on my behalf in 30 to 180 seconds.  Everything on the Web UI looked fine.  Everything on the iFly connected devices page looked fine except for "ground link status"

jzeevi, 2016-02-18:
I also observed this on .7b1 last weekend.

cyoung, 2016-02-18:
What indicators were showing you that you "lost ground reception"?

jzeevi, 2016-02-18:
WX reception would disappear (I normally fly with FF’s Categories selected) and the indicator of how many towers were being received (top left of the FF map) went away, then came back red – showing 0 towers, then shortly showed a larger number and operation was restored.

 

I did see on the stratux webUI that I had towers. They just disappeared from FF.

GrayFlyer, 2016-02-18:
iFly has a connected devices page:
[image]
I also got flags on the nav screen to that effect on both devices.  This situation is different from what I have reported before where wifi connection was lost; no heartbeat, no GPS.  Now I maintain heartbeat, GPS, 1090es, just no ADSB ground link, but as I said it restores itself, only to lose it again in just a few minutes.

cyoung, 2016-02-18:
I'll give it a try with iFly on Android. Is it the same binary you use for your Fire?

GrayFlyer, 2016-02-18:
Yes.  I have the same version of iFly on both my rooted Fire, and my
    Samsung Tab S.  It runs fine on both.
    
    On 2/18/2016 5:55 PM, cyoung wrote:
    
    
      I'll give it a try with iFly on Android. Is it the same binary
        you use for your Fire?
      —
          it on GitHub.

Ergonomicmike, 2016-02-19:
I plan to go flying in about six hours. I'll pull the latest 23 commits. Aside from the usual logs, is there any other testing that I can do to help isolate what's causing this problem? iFly creates a adsbraw.bin file. But I expect that that is proprietary. (Although if I send them that file, they might be able to tell me why they report no ground connection.)

Ergonomicmike, 2016-02-20:
I flew v0.7b2 + 23 commits with iFly v9.4rc1 today. Flew two half hour legs with a one hour stop over. I left the Pi on the whole time and i left my tablet on (although in sleep during the stop over.) I toggled back and forth between the Pi status web UI and iFly in flight to exercise things. (Only one tablet in flight today tho.)

The Stratux worked perfectly. Absolutely no problems in flight. No tower drop outs, wx coming in fine, etc.

It wasn't until the end of the flight, when I was getting ready to shutdown the Pi with Raspi Check that I noticed the blue light on the Edimax wasn't flashing. I tried Axtel4's trick of shutting off the Wi-Fi on my tablet. But that didn't work.

I would post the stratux.log and the syslog. Unfortunately, they both appear to be corrupted. "Does not look like a text file" is the message in Linux. (I had deleted the old stratux.log before the flight to keep things short.) I'm guessing that I had some kind of catastrophic failure that corrupted these files? 

I seem to be the only getting the blue light out problem. So I'm going to assume for now that this is an Edimax hardware problem. I am planning to borrow a neighbor's Edimax to test. 

Could it be a bad SD card? I should swap that out too.

Ergonomicmike, 2016-02-20:
P.S. The guy flying with me today brought his Stratus v2 along. I was going to compare tower reception. But as I was getting 5 towers today in the Phx area, I totally forgot. And too, every target his FF showed with his Stratus, my iFly showed with my Stratux.

cyoung, 2016-02-20:
> I would post the stratux.log and the syslog. Unfortunately, they both appear to be corrupted. "Does not look like a text file" is the message in Linux. (I had deleted the old stratux.log before the [...]

Should be able to download it anyways, try using the web interface. http://192.168.10.1/logs/stratux.log.
Not sure what editor you're using but it sounds like it simply has some non-ASCII characters in it.

skypuppy, 2016-02-20:
Battery too low for the stratux at the end?

skypuppy, 2016-02-20:
You might want to run an fsck first, Mike.

Ergonomicmike, 2016-02-20:
@cyoung  Thanks. You're correct. The log appears in the Stratux web UI. However, it's not like the other logs that I can download. Best I can think of doing now is cut and pasting. But now that the log is showing, I'll copy the file in Linux to a flash drive and upload it on the web.

@skypuppy On fsck - I forget - does the Stratux already do that automatically on every boot? Or when a dirty flag is set? (Until RO comes to Stratux.) As for the low battery - I've had excellent service with the Anker. It normally goes for 10 hours. It only had 2 on it today and is still showing 3 bars.

Ergonomicmike, 2016-02-20:
stratux.log and syslog zipped and posted at 
https://www.sendspace.com/file/mn1i4a

Found that they open in a web browser. (Geany and "open as text" barfed.)

Looks like the problem occurred at 23:17Z. I'll be interested to hear what you all say happened.

skypuppy, 2016-02-20:
Mike,
It may be too careful, but if I think my sd card has been hosed, I'll
leave the RPi2 off and put the card into a Linux PC for fsck (without
mounting, IIRC.) 
Some versions of Linux do autocheck the filesystem on boot, but I think
there is a flag somewhere that must be turned on for that to happen. 
(been a long time for that one, since I'm so cautious.)  :)
However, I have had root files corrupted and the system went kinda crazy
and couldn't fsck itself properly.  So I had to take the hard drive out
and put it in another Linux box to fix it, or dig out a Linux recovery
CD/USB stick to boot a live OS from in order to fix it.  <sigh>
Windows is even worse, though.

Makes me wish the RPi2 had "multi-boot" capability, where I could let it
boot from a USB drive, and it not found, boot from the SD card.  I know
it can be done but it's kinda finicky...

Skypuppy

skypuppy, 2016-02-20:
If I EVER get an IDE working with cross-compiles, I intend to make an
automagic stratux log reader/reporter, with summaries, patterns, and
etc.  Don't hold your breath, though.  <very big sigh>

Skypuppy

Ergonomicmike, 2016-02-20:
Ran fsck -f -v on both partitions. The second partition had some problems with inodes and fsck fixed them.

Whether the fsck problems were the cause of my Edimax problem or a result, I do not know.

cyoung, 2016-02-20:
I'm having trouble finding clues of some abnormality at ~2016/02/19 23:17, what tipped you off that it happened around this time? Was it just around when you noted that it happened?

AvSquirrel, 2016-02-20:
@Ergonomicmike , @cyoung -- look at syslog starting at 23:10:19Z. I think your USB controller had a meltdown.

````
Feb 19 23:10:19 raspberrypi kernel: [11488.416218] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416271] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416309] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416401] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416438] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416477] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416562] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416598] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416635] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416716] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416755] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416795] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416878] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416917] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.416955] NYET/NAK/ACK/other in non-error case, 0x00000002
Feb 19 23:10:19 raspberrypi kernel: [11488.516313] usb usb1-port1: disabled by hub (EMI?), re-enabling...
Feb 19 23:10:19 raspberrypi kernel: [11488.516359] usb 1-1: USB disconnect, device number 2
Feb 19 23:10:19 raspberrypi kernel: [11488.516377] usb 1-1.1: USB disconnect, device number 3
Feb 19 23:10:19 raspberrypi kernel: [11488.516664] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet
Feb 19 23:10:19 raspberrypi kernel: [11488.516730] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Feb 19 23:10:19 raspberrypi NetworkManager[2095]:    SCPlugin-Ifupdown: devices removed (path: /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1:1.0/net/eth0, iface: eth0)
Feb 19 23:10:19 raspberrypi NetworkManager[2095]:    Ifupdown: get unmanaged devices count: 1
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <info> (eth0): now managed
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <info> (eth0): device state change: unmanaged -> unavailable (reason 'managed') [10 20 2]
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <warn> (2) failed to find interface name for index
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: (nm-system.c:685):nm_system_iface_get_flags: runtime check failed: (iface != NULL)
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <error> [1455923419.153507] [nm-system.c:687] nm_system_iface_get_flags(): (unknown): failed to get interface link object
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <info> (eth0): bringing up device.
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <info> (eth0): deactivating device (reason 'managed') [2]
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <warn> (2) failed to find interface name for index
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: nm_system_iface_flush_routes: assertion 'iface != NULL' failed
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <warn> (2) failed to find interface name for index
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <info> (eth0): now unmanaged
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <info> (eth0): device state change: unavailable -> unmanaged (reason 'removed') [20 10 36]
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <warn> (2) failed to find interface name for index
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: (nm-system.c:685):nm_system_iface_get_flags: runtime check failed: (iface != NULL)
Feb 19 23:10:19 raspberrypi NetworkManager[2095]: <error> [1455923419.178341] [nm-system.c:687] nm_system_iface_get_flags(): (unknown): failed to get interface link object
Feb 19 23:10:19 raspberrypi dhcpd: receive_packet failed on wlan0: Network is down
...
````

All of the USB devices in stratux.log also shut down.
````
2016/02/19 23:10:18 On  192.168.10.10:4000,  Queue length = 0 messages / 0 bytes
2016/02/19 23:10:18 On  192.168.10.13:4000,  Queue length = 24825 messages / 10926175 bytes
2016/02/19 23:10:19 Exiting gpsSerialReader() after i=229431 loops
2016/02/19 23:10:19 Entered UAT shutdown() ...
2016/02/19 23:10:19 UAT shutdown(): calling uat_wg.Wait() ...
2016/02/19 23:10:19 UAT read(): shutdown msg received...
2016/02/19 23:10:19 UAT shutdown(): uat_wg.Wait() returned...
2016/02/19 23:10:19 UAT shutdown(): closing device ...
2016/02/19 23:10:19 Entered ES shutdown() ...
2016/02/19 23:10:19 ES shutdown(): calling es_wg.Wait() ...
2016/02/19 23:10:19 ES read(): shutdown msg received, calling cmd.Process.Kill() ...
2016/02/19 23:10:19 	 kill successful...
2016/02/19 23:10:19 ES shutdown(): es_wg.Wait() returned...
````

Ergonomicmike, 2016-02-20:
Ah, so 23:10Z per my update above. (Sorry, Chris, for the initial false lead.)

Okay, USB controller = hardware, right? So hardware failure means I need to buy another Pi?

Or - could one of my USB devices be causing the controller to fail? (Probably my first SDR, since I think that I've been having this meltdown from the beginning.) It's clearly intermittent. Would these logs catch that failure or are the logs not granular enough for that detail? 

And thanks to both of you for looking into this.

cyoung, 2016-02-20:
I don't think it is failed hardware.

https://github.com/raspberrypi/linux/issues/35 -- they theorize that it is not power related, no resolution.
https://github.com/raspberrypi/linux/issues/205 -- no resolution.

http://www.netnode.de/howto/USB_modem_power_issues.html -- power related.
Couple others show power related.

Anyone know of a good add-on to the RPi to start logging voltage? There's this: http://amzn.com/B00ORN78WE, but it doesn't help much with intermittent problems. Perhaps a low voltage cut-off so that it's obvious when it happens?

skypuppy, 2016-02-20:
this is how I would diagnose it, Mike.

I would check power issues first.
Then look at connectors.
Next, software timeouts on the hardware (difficult.)
Temperature related issues (even more difficult.)
Finally, A/B parts swapping.

skypuppy, 2016-02-20:
> raspberrypi/linux#35 <https://github.com/raspberrypi/linux/issues/35>
> raspberrypi/linux#205

Be better if it had logging capabilities.

Hmm, I remember something from Sparkfun or Adafruit that watches power
and connects to I/O pins, with internal voltage protection...  Maybe
DigiKey?  If it hooks to an Arduino or another RPi2, then you can get
the logging.
That would answer the nagging power question.
The only other option I can think of at the moment is a logic analyzer,
say on the power and the USB bus.  There are some kits that turn an RPi2
or BeagleBone Black into a logging logic analyzer...

if it is temperature related, and so on.

Ergonomicmike, 2016-02-20:
I will put the Edimax on a short USB cable in case it's an EMI problem.
I can buy a powered hub, power it w/ the KMASI and power the Edimax with it, to isolate it from the Pi.
If this is a controller issue, and if a certain confluence of bits overwhlem the hardware controller, would others get the same meltdown if they played my logs? (Looking to reproduce the problem.) If so, please tell me what to upload. (I suppose that I should be able to replay my own logs and reproduce the problem if it's bits vs. the controller. But I might need to call someone to talk me thru the commands.)
Which chip on the Pi is the controller? (My Pi is wrapped in copper and I can't see the PCB.) I assume the chip closest to the USB ports? I could put a heat sink on it. (But I don't think this is heat related. Someone in the reddit says he put his in an environmental chamber and ran it at 70 C (IIRC). Mine sat in the AZ sun for a while on  a 90 degree day and didn't fail. I will let it bake some more today. (So if heat doesn't kill the controller, I'm thinking it's not power related either, since R goes up w/ T.)) Having said that, these heat stress tests aren't apples to apples. They're not like flying, where lots of bits come in to test the controllers integrity. So it could be heat related when the controller is working hard to sort bits.

Ergonomicmike, 2016-02-20:
I remember reporting blue light out when coming back from Big Bear at night in early Feb at 9500. It was cold up there. That would a point in favor of not heat related.

Ergonomicmike, 2016-02-20:
I see the UAT shutdown reports in the stratus.log. Is the a way that the software can either restart the controller or command a self-reboot when it sees that report? The latter isn't elegant, but at least would recover from the failure.

Ergonomicmike, 2016-02-20:
Can we call this issue "The Blue Light of Death"? (Parody-ing the MS BSOD.)

cyoung, 2016-02-20:
We should look into reinitializing the wifi dongle when disconnected, as
you suggested before. That would likely have allowed the wifi to recover in
your situation, whatever the cause was.

Ergonomicmike, 2016-02-21:
I looked at AvSquirrel's analysis of syslog for the case of the Edimax physically disconnected. There is a "USB diconnect" message - which also appears in  syslog when the controller melts down. So I presume that one code trigger off that message to try to reconnect in both cases.

Now, naturally, I'm all for addressing [Issue 246(https://github.com/cyoung/stratux/issues/246). But in the case of a melt down, we don't know if the controller will accept commands (to reconnect), or if it's in some indeterminate (software?) state. If in an indeterminent state, then wouldn't a reboot be necessary? But if some transistor or gate in the hardware is stuck on a rail, then it might be that not even a reboot would fix it. (In which case, ironically, the Anker's foible that causes a restart after shutdown, would come in handy.)

Ergonomicmike, 2016-02-21:
So I'm looking at [this solution](http://www.netnode.de/howto/USB_modem_power_issues.html) that Chris found. My first thought was to run some 22 gauge wire from the pads near the Pi's micro-USB to the 5v and gnd of the USB connectors. But then I remembered seeing a line in the stratux.conf about increasing current to the USB. So does that mean that the Pi is somehow involved in regulating power to the ports?  (Why?) If so, then I ought to Xacto the power run(s) to the ports and just wire them directly to 5v in at the micro-USB. Please, before I do this, would someone tell me why this is a bad idea?

Ergonomicmike, 2016-02-21:
[This site](http://www.lucid-control.com/raspberrypi-2-b-power-supply-usb-io-module/) says GPIO pin 2 is the way get power to the Pi 2 for USB.

jzeevi, 2016-02-21:
USB provides IIRC 100ma to the ports for detection of insertion. At  that point the device tells the port (electrically) how much current it wants, up to a limit for each type (1.0, 1.1, 2.3, 3.0) and the port can switch to that. I'm not sure what the .conf is doing, but it is risky to ask a controller to overdrive the port. The transistors that drive power have some overhead room, but it's not always a good idea to ask for more. Think of doing so as overclocking. For parts designed to do that, it's ok. For others, there's the risk of damaging the output driver. The solution of providing additional power (in parallel) from another USB port or 5v supply is a better choice, but be wary of simply asking too much from the pi. 

If you remove the USB port on the Pi from realizing currents being pulled through it, you're likely to cause the part to not work right. 

Josef
Sent from a phone with a tiny screen

Ergonomicmike, 2016-02-21:
"2. Any data bursting on any port (especially SD writes) spikes internal buses - you aren't going to see that with a typical multimeter." Commentbas of Feb 18, 2016  at https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=135214&sid=af90061d8268c1b4ebeff6788148f497
 
I don't know what writes the Stratux is bursting. But if true, and if the blue light out is power related, then going to RO might mitigate the problem.

Ergonomicmike, 2016-02-21:
@jzeevi  The link here http://hackaday.com/2015/04/06/more-power-for-raspberry-pi-usb-ports/ explains what the config file is doing. (Switches a resistor on or off that controls a current limiter.) A commenter has an even cleverer way to change only one resistor so that the soft switch in config.txt still works.

Ergonomicmike, 2016-02-21:
4.76 V at the USB w/ fully charged Anker, as others have measured before. Right on the low edge of the USB spec. Man, those surface mount resistors are tiny. I'm gonna have to buy a smaller soldering tip and hope that my eyes & hands are up to the task.

Ergonomicmike, 2016-02-21:
Sometimes the more practical solution is the better one. I soldered a jumper. http://s10.postimg.org/bf2aokkax/5v_jumper.jpg 

Picked up 100 mV at the USB ports. Up to 4.85 volts now, under load. We plan to fly Wednesday. So hopefully will have a report by then.

skypuppy, 2016-02-21:
Before cutting those itty-bitty traces, I recommend you verify it a
current (no pun intended) problem first.  I've got some sample inline
current A/D devices around here somewhere.  Maybe i could put them
inline via the usb-dongle.  Uh boy.  I guess we're going to wind up with
a hardware test suite after all.  <sigh>

Mike, you must have access to some way to measure current in-line, don't
you?

Regarding the Anker "3 lights," I have stopped trusting indicators on
equipment without independent instrumentation verification at least
once.  <very big sigh>

Skypuppy

Ergonomicmike, 2016-02-22:
No, sadly I don't have an easy way to measure current. If this issue persists, I might buy one of those USB measurement tools  that Chris linked to. I would probably have to open it up to get a 'scope on the shunt or Hall Effect or whatever they're using to detect current.

Ergonomicmike, 2016-02-22:
Data point: The Edimax seems to be "smart." Plug it into a power pack. The power pack indicator will light up. The Edimax never does. (Nothing to xmit.)  So, within a few seconds, the power pack indicator light will go out. Which means no power. Can the Edimax be restarted from this state?

AvSquirrel, 2016-02-25:
Found a bug in traffic.go. Fix for this will be rolled into PR #285.

https://github.com/cyoung/stratux/blob/65559a38fd681cf99b9119fa857925b9cac3a75a/main/network.go#L296-L298

FIS-B queue send interval (`pd`) is supposed to be set to 0.1 seconds if the message queue is empty. It was being set to ten seconds. This caused messages to back up in the queue, then all be released at once.

Depending on how quickly iFly is "timing-out" stale uplinks, this may be enough to show you losing towers.

Ergonomicmike, 2016-02-25:
Excellent find! Looking forward to seeing if that solves both the iFly "Connection lost" problem and the BLOD problem. (Possibly a burst of data overwhelming the USB controller?)

To be clear, iFly makes a distinction between tower count (which I'm pretty sure they get direct from Stratux) and Connection. (Which, IIRC, is a new parameter, that they're inferring from data throughput.)

GrayFlyer, 2016-02-25:
@AvSquirrel 
When would this bug have been introduced?  While I have seen wifi disconnects before v0.7, I didn't see this problem until using v0.7.  My experience base is limited to the one flight that I reported due to aircraft issues / weather. My plane goes into annual the end of this month, so it will continue to be limited. I plan to be flying this weekend though with the latest version.

@Ergonomicmike 
The "ground link status" has been there ever since I started flying Stratux early last Oct; it was the "tower count" that was added to the "Connected Devices" page recently.  (Tower count was available elsewhere before then, but a bit harder to bring up)

Thanks for everyone's input and help!

AvSquirrel, 2016-02-25:
@GrayFlyer -- looks like it was around v0.5b2 (https://github.com/cyoung/stratux/commit/e6ad9aa830b87d19b1a6731857b75f53d5bb26a1#diff-4ba3b85a06589d714ee8177de6236fd0), but I'm guessing it didn't manifest until message batching in the v0.7 releases allowed the queue to empty.

Ergonomicmike, 2016-02-26:
@GrayFlyer Thanks for correction. I had it backwards.

GrayFlyer, 2016-02-26:
I flew this morning with v0.7b3 for about 1.5 hours plus the system up during a fuel stop.I had absolutely no problems with wifi connection or loss of ground reception.  I had two devices connected; an iFly 720, and a Fire tablet.  Unless there is some significant difference between b2 and b3, I'm now wondering if what I saw before with repeated ground reception drops may have been FAA system glitches?

Thanks for everyone's input and work!

cyoung, 2016-02-27:
@GrayFlyer - the issue hadn't been addressed in the code you tested, so it may have just been luck that it worked (unless there's some other code you were testing?).

The proposed fix will be in the next release - would be great if you could give it a try when it's out.

GrayFlyer, 2016-02-27:
@cyoung - No other code involved, no changes to my hardware.  I'm looking forward to flying the next release, but it will be after my annual which begins 3/1, before I can.

cyoung, 2016-02-28:
I think it was related to the network issue mentioned above by @AvSquirrel. That has been addressed, reopen if it happens again.

closed 2016-02-28 by cyoung