cyoung/stratux #269: v0.7b2 Observations

Axtel4, 2016-02-19:
1. Stratux version:
> root@raspberrypi:~/stratux# git describe --abbrev=0 --tags
> v0.7b2
> root@raspberrypi:~/stratux# git log -n 1 --pretty=%H
> 7a2e49867aba68f5cdaa5d28fb5c7d23e35f2a29

2. Stratux config: single/**dual** SDR, GPS **yes**/no (Vk-172), AHRS yes/**no**, battery, battery cable **ship power**.
3. EFB app and version (DroidEFB). EFB platform (Kit Kat). EFB hardware (ASUS TF-300).
4. Description of your issue. If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

While performing signal strength testing and RF characterization of the dongles I noticed that the Status page (on Chrome Browser) constantly reports 1 1090ES message and will intermittently jump to 2 messages (see screen shot).  The recorded logs are empty and the EFB App shows **ADSB NO TRAFFIC TARGETS**.  (Note1: the antenna terminals are terminated into 50 ohm loads and the dongles are shielded from external signals. Note 2: It behaves the same on the tablet browser and the Desktop browser.)

Also, the Stratux clock and the Device clock have never synced closer than 25 seconds and difference appears to be increasing a few milliseconds at each successive boot.  I don't know if this is normal but thought I'd mention it.
 
[image]

Ergonomicmike, 2016-02-19:
So, dumb question: Your device is querying ntp time servers? Even if yes, are you using satellite for Internet? (Trying to explain a 25 sec delay.) If you manually set your device time, can you force the delta to be less than 25 seconds? I've gotten mine down to 8 seconds by tapping at the right time. (No pun intended.)

cyoung, 2016-02-19:
re: Clock; "Device Clock" is the clock of the device that you are accessing the web interface with. It likely has more to do with that then with stratux - Windows/MacOS/iOS? See about turning on clock sync on your desktop/laptop/tablet and see what the difference is like. Or try a different device.

Axtel4, 2016-02-19:
This project is always presenting something new to lean.  The tablet was .67 sec, the desktop was 29.8 sec.  Resynced the desk top and I am now down to .71 sec.  Thanks.

Axtel4, 2016-02-20:
Update: I rolled back to v0.7b1 and the 1090ES messages stay at zero.

AvSquirrel, 2016-02-20:
> While performing signal strength testing and RF characterization of the dongles I noticed that the Status page (on Chrome Browser) constantly reports 1 1090ES message and will intermittently jump to 2 messages (see screen shot). 

That's actually [by design](https://github.com/MalcolmRobb/dump1090/blob/master/mode_s.c#L1899-L1917). dump1090 sends a null message every ~59 seconds to keep TCP connections open.

cyoung, 2016-02-20:
Hmm, @Axtel4 - how did you downgrade? Is it possible you used an old dump1090 binary? We've upgraded to the MalcolmRobb branch since a while ago. Wonder why it wasn't showing before?

Axtel4, 2016-02-20:
I reverted to a previous image I burned.

Prior to updating the SW, I make and archive an image of my current build.  That way if I need to I can quickly go back to the pervious version.

Ergonomicmike, 2016-02-22:
So, should this issue be closed?

closed 2016-02-22 by Axtel4