cyoung/stratux #270: v0.7b2 Web Interface Missing Time and 2nd SDR PPM Correction

pp4life, 2016-02-19:
1. Stratux version: 0.7b2
2. Stratux config: single/**dual** SDR, GPS yes/**no** (type?), AHRS yes/**no**, wall power
3.EFB app and version: Foreflight 7.5.2. EFB platform: iOS 8.3 on, EFB hardware: iPad Mini 2.
4. Description of your issue. I updated to 0.7b2 (with no commits) with the .sh method and I'm not seeing the "Stratux Clock" and "Device Clock" in the WebUI on Safari.  Is that normal without GPS? I've tried deleting the shortcut from my home page and clearing the cache and history for Safari on the iPad mini. I've also tried this from an Android phone with Chrome and from a desktop with Firefox. Same result. 

I am also not seeing anywhere to put in the 2nd SDR PPM correction.

Lastly, from the iPad (or I assume any iDevice) when you click on the "Click to select System Update file" button, it only lets you select from your photo album or to take a picture. Searching around, this seems like a limitation of iDevices. Super annoying. I did the *.sh update though my android. 

[image]
[image]

Axtel4, 2016-02-19:
Did you clear the Browser cache and reload the and Stratux Web UI?

ssokol, 2016-02-19:
I can confirm this. Flushed and I still get:

[image: Inline image 1]

[image: Inline image 2]

cyoung, 2016-02-19:
The feature was added by @yihchun in PR #248 for those that program their SDR strings for tuned antennas. No feature has been added to the web interface to make use of the functionality as yet.

pp4life, 2016-02-19:
Ah OK. I have written strings to the SDRs for the dmurray antennas, but they are the standard "stratux:XXXX". I now see how @yihchun has implemented the tuning per SDR with the strings. I'll have to rewrite the strings to take advantage of the per SDR PPM settings.

pp4life, 2016-02-19:
@sskol, your images didn't come through on the github site

pp4life, 2016-02-22:
This is what it looks like after the 0.7b3 "sh" update. All looks good!

[image]

cyoung, 2016-02-28:
Expecting feature that never existed.

closed 2016-02-28 by cyoung