cyoung/stratux #275: 1090 sdr not initiating

peepsnet, 2016-02-21:
I'm running a dual sdr stratux with ry835ai 
Version . 7b2

I believe the issue is related to the new dual SDR PPM logic that was introduced in the last release

When booting the device the 1090 radio never initiates although it shows active on the Status page. I only get 978 data. 
 
Here's a video of the walkthrough for debugging purposes
https://drive.google.com/file/d/0B8TVybIHchkfYWZBMHRGUFh6UEk/view?usp=docslist_api

yihchun, 2016-02-21:
In #248 I noted
"If you don't need to specify a frequency, pick any number other than 978 and 1090 (0 recommended)" and "Doesn't allow for your format, but stx:0:-27 should work."

This will allow for better debugging:
1. After boot, run `ps auxww | grep dump1090`
2. Insert the debugging lines 
`log.Printf("===== UAT Device serial: %s =====\n", u.serial)` at the beginning of `func (u *UAT) sdrConfig() (err error)` and
`log.Printf("===== ES Device serial: %s =====\n", e.serial)` at the beginning of `func (e *ES) sdrConfig() (err error)`. Recompile, install, and report the results of `fgrep serial /var/log/stratux.log`

peepsnet, 2016-02-21:
Results from execuiting commands

pi@raspberrypi ~ $ sudo su -
root@raspberrypi:~# ps auxww | grep dump1090
root      2201  0.0  0.1   3560  1796 pts/0    S+   20:47   0:00 grep dump1090
root@raspberrypi:~#

I will ado the rest and report

peepsnet, 2016-02-21:
And here is the rest

pi@raspberrypi ~ $ sudo su -
root@raspberrypi:~# fgrep serial /var/log/stratux.log
2016/02/21 20:54:30 ===== UAT Device serial: stx:0:-28 =====
root@raspberrypi:~#

peepsnet, 2016-02-21:
The per sdr ppm logic is incorrect in some manor. If I set my serials to stx:978:-24 and stx:1090:-28 then the system boots properly and I get both sdr's receivng data.

peepsnet, 2016-02-21:
I have the serials set as above.
I turned off both sdr's in the webui settings. I then turned only three 978 sdr and it did not activate. I could still rtl_eeprom to both sdr's

peepsnet, 2016-02-21:
Just FYI. Updated to 0.7b3 no change. But I didn't expect any. Just so we are on same page

peepsnet, 2016-02-21:
Correction. In version. 0.7b3 the system now boots up with both sdr's on and correctly assigned with frequency specific serials (978 & 1090). But once booted from the settings webui if I turn off both sdr's then turn only only 978 then no sdr's are activated.

Changing the serial to 978 & 0 the systsm boots up with only the 0 serial sdr to set to 978 with both sdr's on in settings

yihchun, 2016-02-21:
I've committed a change for you momentarily that includes SDR counts. I suspect it'll help for other reasons too.

It seems like your ES is never launching. I've set my radios up the same way as you and cannot reproduce your bug. Easy thing to try is to change your stx:978:-24 to stratux:978 and set the global PPM to -24 and see if that helps.

bradanlane, 2016-02-22:
My case may or may not be related ...

I was away from Stratux for several weeks so I jumped up to 0.7r1 image. My system is 2 SDRs and the RY835AI. When I boot up, dump1090 is "defunct". If I manually start it to use "device 2" I get an error. I can manually start it using "device 1". When I use `lsusb` I notice my second SDR now has a different type. now, one is reporting as RTL2832U but the other is reporting as RTL2838UHIDIR  Previously, both reported as RTL2382U. Both SDRs are the same so there is no reason the type should have changed. 

I have not assigned serials to the SDRs.

peepsnet, 2016-02-29:
This seems to be fixed in the latest commits and will be in the next release

closed 2016-02-29 by peepsnet