cyoung/stratux #276: Webui showing working sdr when not in use by stratux

peepsnet, 2016-02-21:
Can we get a bit more info on the Status page regarding the sdr's. I would like to know if the sdr is responding. In the above issue the Status page shows the sdr is just not receiving any signal not that the sdr is not even in use by stratux! It would be nice to show an sdr is selected but not functioning

This is from the issue I am having on this issue:
https://github.com/cyoung/stratux/issues/275

skypuppy, 2016-02-21:
Those test are better served by other, diagnostic software.  Search for
"rtl-sdr" or "rtlsdr" and raspbian.

Skypuppy

peepsnet, 2016-02-21:
What I mean is.  I booted my stratux and looked at the Status page. The uat sdr showed tons of traffic. The 1090 sdr showed 0 messages. I then sshed  into the stratux and used rtl-eeprom and found that stratux had let go of the 1090 radio and yet still believed it was connected in the Status page.

The bug in #275 just showed that stratux could let go of the sdr and not show the sdr down but lead you to believe it is actually functioning

Ergonomicmike, 2016-02-22:
I was going to ask for a similar feature because one of my USB extension cables was a liite loose. Consequently my 1090 SDR sometimes backed out a little in flight. (Free form Stratux - no case.) It wasn't enough to kill power to the SDR, and the status page still showed 2 dongles connected. But when I noticed that I wasn't seeing as many aircraft as I should, I checked the status page again. The same status page showed no 1090 messages. When I pushed the SDR back in, the traffic count immediately jumped up. So the info I needed was on the page all along.

I figure there's only so much that the developers can do to make something foolproof, and I didn't think it fair to ask them to protect this fool. (Me.) I've since taken a dental pick and flexed the prongs in the USB connector.

peepsnet, 2016-02-22:
If its possible to do and it happened then it will happen again. You would not want something like this to happen if stratux is being demoed to some investor and it ruins the pitch!!

Ergonomicmike, 2016-02-22:
If the Stratux were a production unit that I paid a lot of money for, I would want it to be fool proof. (With a one year guarantee.) But I want to be realistic. This is an open-source project, brought to us by the good graces of a few - for free. And it's built with non-purposed, non certified parts. I'm often surprised at the requests (demands?) for features that some will ask for in the reddit. There are no guarantees.
I'm happy that it's cheap. I'm happy that it works. I'm happy that Chris & friends are working to stomp out the biggest bugs, like the random disconnects that we don't have control over. For now, I would rather not burden them with an issue that I do have control over, that I could prevent with more diligence on my part.

cyoung, 2016-02-22:
We can make the UAT/ES message rate "bars" turn red or something when the respective frequency is enabled but not receiving. It is a little misleading now.

Nokomis449, 2016-02-22:
While you're in there, can you also make the numbers in the bar bold or a size larger?  Old eyes sometimes can't make them out when they have the colored background.

bradanlane, 2016-02-22:
as I noted on issue #275 it is possible for Stratus to spawn the dump1090 process and not know it is defunct. What I do n ot understand is why Stratux is not getting the error messages which should be available on either on stdout or stderr - both of which are processed by Stratux.

jpoirier, 2016-02-25:
Currently, neither the 1090 nor the 978 handlers abort when there are errors, partly because it's not well understood what an error encompasses. If all errors, or the majority for that matter, are unrecoverable it's easy to just abort otherwise no so much. 

Assuming all or most errors deem the current state of the dongle unrecoverable:

If dump1090 is using stderr properly, when readToChan sees a stderr message it could send a message to closeCh to shutdown the 1090 process. Note it'd probably be easier to just split readToChan to a read stdout and read stderr version.

For 978, when an error is identified in "nRead, err := u.dev.ReadSync(buffer, rtl.DefaultBufLength)" we can return rather than break. We could always set an error counter that increments and decrements and only returns when it overflows.

closed 2018-01-17 by cyoung