cyoung/stratux #282: v0.7b3 Web UI traffic page shows nothing

GrayFlyer, 2016-02-23:
The status page shows up to 1725 msg, iFly shows traffic moving, traffic page blank??

Ergonomicmike, 2016-02-23:
First, did you close and/or clear your cache's browser? I didn't get any Traffic in the web UI until I did that.

Had you done any git pulls before upgrading to v0.7b3? I had done an *.sh update from v0.7b2 to v0.7b3 and that didn't entirely work for me. (Even tho I saw v0.7b3 on the status page.) It was because I had done git pulls beforehand and/or also because I had inserted some .go files from AvSquirrel. I had to rename the files I had added and then I did a git pull and everything is working as advertised. (I see the cool new icons in the Traffic page.)

If you still can't get Traffic to display correctly, then I suggest re-imaging back to 0.7b1 and either doing the .sh update or git pull. (But not both.)

GrayFlyer, 2016-02-23:
I cleared the browser cache, and have not done any git pulls.  I did upgrade0.7b3 via *.sh from 0.7b2 which was via *.sh from v0.7b1.  What do the red and green ((ConnectState)) buttons both being on mean?

Ergonomicmike, 2016-02-23:
I don't know. There's another guy in the reddit who is [reporting the same problem](https://www.reddit.com/r/stratux/comments/478tj1/v07b3_web_page_traffic_issue/).

Some browsers and/or OS's are not compatible with the web UI. On my old Nook HD+, running Android JB, I get a message about needing some type of web GL and the GPS page doesn't paint correctly. I presume that you're using the same tablet that you've always used? But if you have another browser or tablet, try that and see what happens. Traffic displays fine for me.

Ergonomicmike, 2016-02-23:
If all else fails, I suggest re-imaging to v0.7b1 and then doing git pull to the latest. That's ultimately what I had to do to see the new Target icons.

AvSquirrel, 2016-02-24:
> What do the red and green ((ConnectState)) buttons both being on mean?

It's a sign that one of the scripts is breaking. As @Ergonomicmike says, this will happen with sufficiently old browsers. A page that works fine in Chrome or iOS 9 Safari will completely break on an old Android 2.3 phone or first-generation iPad I have sitting around.

What browser and OS are you running on? Can you copy / paste the user agent string from the bottom of the Logs page?

GrayFlyer, 2016-02-24:
Thanks guys! It turned out to be the Dolphin browser.  I tried Chrome and the traffic page works just fine.  The tablet is a Samsung Tab S, running Android 5.0.2  

BTW, I also tried it on my rooted Fire using the Silk browser and it works fine there.

cyoung, 2016-02-24:
It was broken for previous version on Dolphin, wasn't it?

GrayFlyer, 2016-02-24:
I can't say for sure, but it probably was.  I flew v0.7b2 using Silk on my Fire and it was fine, I thought I ran it at home on the Tab S with Dolphin, but maybe not.  I've had problems with the Web Ui and Dolphin before, and just forgot to use Chrome.  The problem Dolphin is the latest version.

Did you ever fly v0.7b2 with iFly to see if you got the repeated loss of ground stations that I was getting?  I haven't flown since, but when I do it will be with b3.

cyoung, 2016-02-24:
Haven't tested it yet

Ergonomicmike, 2016-02-24:
I flew iFly. (A v9.4 release candidate, IIRC.) No problems.

GrayFlyer, 2016-02-24:
I was also flying with a v9.4.x beta connected to my 720 and an Android tablet.  They were going in and out of ground link reception together.  How many devices were you connected to on your flight?

Ergonomicmike, 2016-02-24:
I was flying 63rc1 and had only one tablet (Android) that flight. Planning to fly Wed w/ 65rc2 w/ two tablets. It's on my Test Card to check the Connections page every now & then to see if I get what you got.

GrayFlyer, 2016-02-24:
If you don't already have it selected, there is an ADSB Status
    instrument available on iFly and it also pops a banner warning on
    loss of ground stations after so many seconds.
    
    
    On 2/24/2016 1:10 AM, Ergonomicmike
      wrote:
    
    
      I was flying 63rc1 and had only one tablet (Android) that
        flight. Planning to fly Wed w/ 65rc2 w/ two tablets. It's on my
        Test Card to check the Connections page every now & then to
        see if I get what you got.
      —
          it on GitHub.

Ergonomicmike, 2016-02-25:
Okay, I had my first "GrayFlyer" disconnect message today in iFly. (At least, the first that I've seen.)

To answer @GrayFlyer directly, yes, I have had the ADS-B Instrument up ever since the Stratux.

Today I got your disconnect message. I was descending into a back country airport in the desert SW. (P23, Seligman, AZ.) When I got the disconnect message in iFly's Connection window, it showed two towers, but no Connection. I quickly switched to the Towers page in the Stratux UI. Both towers had a signal strength below 400. (Then down to one tower at 272.) The disconnect message went away when the signal strength came up to 500 for a short time. 

Frankly, it might be that this is expected behavior with iFly. Weak signals, no data. (IIRC, the release notes in the v9.4 betas for iFly have mentioned tweaking the ADS-B stuff.) Not quite the same as displaying a disconnect when there are no towers around. More conservative. Unless this happens to me again with a tower booming in, I don't see it as a problem at this point.

Ergonomicmike, 2016-02-26:
Apparently I am fooling myself that I lowered the gain of the SDR. Axtel4 had asked what my dump-1090 log showed. It reports a max gain of 49.60 and that my gain is set for 49.60. So tgain in sdr.go not working? Or tgain is only affecting the gain of the 978 SDR? (I looked in the uat log for a listing of gain, but did't see anything about it.)

cyoung, 2016-02-28:
Old browser issues.

closed 2016-02-28 by cyoung