cyoung/stratux #283: the Status websocket is reopenning after being explicitly closed

bradanlane, 2016-02-23:
It turns out the resolution for #164 which was committed as [1c231dd](https://github.com/cyoung/stratux/commit/1c231dd301c48c579250097f219dff382b98ad32#diff-709e310a9fdf1c8b5ad0d6d26d76e128) has introduced a similar bug.

Now, when a user legitimately navigates away from the status page, the status websocket is reopened even though the status page is no longer current.

You can view the bug in the browser console. Start on the status page, and then navigate to the log page (or any other page). The console will continue to show `Received status update`

I'm not yet sure what the right mechanism is to differentiate between the socket being closed by the user vs otherwise. It may just be a flag.

ssokol, 2016-02-23:
Not really a big deal but, would it be feasible to build this as a single
page app with each of those tabs simply being a different view, rather than
a different page? That way you could use a single websocket that connects
when the page loads or resumes from background, and there would be no
reconnect time when switching views.

bradanlane, 2016-02-23:
I'm by no means an AngularJS guru so I would not know what pitfalls will come from trying to merge everything at this stage.

I removed the timeout line in `status.js` and made all files served from `managementinterface.go` with no-cache, no-store. Thus far, my testing is showing not bad side effects.

cyoung, 2016-02-28:
What do you suggest to fix this?

We need a similar fix for #255.

bradanlane, 2016-02-28:
I'll need to attempt reproducing the Firefox+Ubuntu case to see if it's the same. In this latter case it's not a production configuration so it may be less urgent. Hopefully the fix for #283 and #255 will be the same/similar.

Ergonomicmike, 2016-03-05:
I don't know where a performance issue would manifest itself, and I know it's more elegant to use elegant solutions instead of kludges. But to put performance in perspective, I only see 30% ave CPU load in flight. And that's with the powersave gov loafing to 600 MHz.So if a hammer bumped that up to 50%, there's still plenty of headroom.

bradanlane, 2016-03-05:
OK. I'll create a branch with my "fix" and submit the pull request.

Ergonomicmike, 2016-03-05:
Worth a try. Still in beta, after all.

westphae, 2017-05-16:
Old issue, but I noticed that the gps socket persists after the page is changed too.  Both status and gps sockets only persist if the user stays on that page for a few minutes before switching.  Interestingly, I have been unable to get the traffic socket to persist.  Its onclose event is handled a bit differently, but modifying the status and gps sockets' onclose methods to be like traffic's didn't seem to resolve the issue.  But maybe the traffic socket's better behavior is a clue to resolving.