cyoung/stratux #284: Video - is this Edimax blue light behavior expected?

Ergonomicmike, 2016-02-23:
This is mostly for @AvSquirrel.  A [short video](https://vid.me/4ZKK) of the Edimax.

(I'm trying to reproduce a problem [that bkwny reported](https://www.reddit.com/r/stratux/comments/46u7hv/wanted_experimenters_to_jump_power_for_edimax/d0a1jh3).)

When you first start the Stratux, the blue light in the Edimax flashes like it usually does. In the video, it flashes, even tho there is no client connected. I don't know who it's sending to. But okay. It doesn't seem to hurt anything by flashing to no one.

Then, after six hours (and there's probably nothing special about six hours - it's just when I came back to check things out) the blue light isn't flickering anymore. Instead, it's just pulsing regularly. Probably the heartbeat?

Since it's the first time I've noticed this - I mean, who leaves their Stratux running in use without a client hooked up to it? - I'm asking if this is expected behavior? It seems to me that when the Stratux isn't connected to a client, it ought to go into heartbeat only mode right away. 

Again, it doesn't seem to be breaking anything. Although bkwny is reporting that, after a night of sitting, his Stratux will not connect to his iPad, even tho his blue light is pulsing. (I'm trying to repeat his experience. Which is why I've got the Stratux running for hours by itself.)

Ergonomicmike, 2016-02-24:
I restarted the test after making the video. Edimax is doing heart-beat only flash at 4 hours. 
I don't know if the time until heart-beat only is a function of raw up time or of a cache filling up. (Or other?)
Anyway, I plan to let it run until morning to see if I am locked out or not after 12 hour plus.

Ergonomicmike, 2016-02-24:
Maybe this is a manifestation of the DHCP lease expiring in 12000 seconds (3.3 hours), per Chris?
https://www.reddit.com/r/stratux/comments/4607bz/display_dhcp_clients_mac_addresses/

Ergonomicmike, 2016-02-24:
16 hours later of the Stratux running without a client, I was able to connect with Stratux. I've noticed that the two people who have reported that they have not been able to connect in a long term test are iOS users. (I use Android.) Possibly something with iOS that, when the Stratux DHCP lease expires, iOS will not renegotiate the connection?

AvSquirrel, 2016-02-24:
@Ergonomicmike -- For what it's worth, I've had mine running overnight for several weeks now, but that's because I'm developing. :-) I don't usually keep any WiFi devices connected. Tonight, I can set mine up to replay a UAT log in a loop, and try to get my Nexus 7 (Avare or Fltplan Go -- if I can get it to stay awake) and iPad Air 2 (ForeFlight) to stay connected overnight.

Regarding the video, the GDL90 output is paused if no active devices are connected, and network.go pings the client table only every five seconds. Maybe it's the UI broadcaster? @cyoung / @bradanlane -- do UI updates get paused when devices are asleep?

AvSquirrel, 2016-02-27:
@Ergonomicmike - I finally got a chance to run that overnight test yesterday:

- v0.7b3 (latest AvSquirrel dump1090-mutability commit). This included the fix to change the default queue time from 10 sec to 0.1 sec.
- Dual SDRs enabled. 1090ES saw typical message rates of 1000+ messages/minute, and 4500+ messages / minute. Unfortunately, I am out of range of the ground stations, so I wasn't able to pick up any UAT messages.
- Traffic demo mode enabled, creating 18 fake traffic targets
- UAT replay enabled, running a ~5 hour loop of prerecorded UAT traffic and weather (1x replay speed). Message density in the recorded stream was anywhere from 1-5 towers, and upwards of 1500 messages / minute peak.
- Three connected devices: a 2012 Nexus 7 running Avare, an iPad 3 with ForeFlight 7.5.2, and an iPad air 2 with ForeFlight 7.5.2

Both Avare and ForeFlight (iPad 3) crashed once during the test, but all three stayed connected and updated for a full 12 hours last night. I left the iPad 3 and Nexus 7 connected when I was at work today, and it was still running strong at 6 PM tonight. That's almost 24 hours of solid connection.

I can't rule out an interaction with high actual load on the UAT radio, but a moderate-to-high message count alone wasn't able to kill the connection.

Ergonomicmike, 2016-02-27:
Thanks. Someone in the reddit said that he has streamed Blu-ray data streams over the Edimax without problems. (Albeit with an Intel based something or other.) That shows that the Edimax proper is capable of high data rates without crashing. That leaves the USB controller on the Pi itself and/or the kernel. (Or power. But I'm beginning to discount power at this point (even tho power can be marginal), because others report running a single client with the KMASHI and running for hours in flight without a BLOD.)

In addition, I think a pattern of failures is beginning to emerge. 

I don't recall anyone getting the BLOD testing on the ground without UAT towers. It always happens in flight. (That could be because they hardly test on the ground, and the place where they really notice a failure is in flight. And they report failures in flight.)

I don't recall anyone getting a BLOD running a single client. It's only with more than one client that the BLOD happens. But as you found, and as others report also, it doesn't happen when a user runs two different clients and/or OSs. (Don't know which one matters.) For example, I run two Android tablets with the exact same version of iFly. I will get the BLOD. But if someone runs FF on iOS and something else on Android, they don't get a BLOD.

I'm trying to gather data to see if having logging enabled is associated with the BLOD. (Back to a power issue. Some in the Pi community report that massive writes to an SD card can cause current dip. I plan to eliminate that possibility by changing a resistor on the Pi to allow the current limiter chip to deliver more current. So far, my wire bypass, which brings the USB port up to 4.85 volts didn't prevent the problem.)

skypuppy, 2016-02-27:
Not that I know much about it, but I would not recommend changing 
hardware on the Pi.  For one thing, no other Pi's will have that mod.

Ergonomicmike, 2016-02-27:
I agree. It's a test to isolate power as an issue. If it can be shown that power is an issue, then the COTS solution might be a 12v power pack and voltage regulator.

Ergonomicmike, 2016-02-27:
Quoting bkwny, "every user who binds to the WAP immediately creates a UDP broadcast burden for Stratux ..." I don't know how big that burden is, but I'll run 3 tablets today to see if I get the BLOD. ('Course, that'll be on the ground testing (no towers). But towers seem to be a requirement for the BLOD.)

Ergonomicmike, 2016-02-29:
"However, since the USB can be constrained by the speed of the CPU, the added processing capacity means the work required to run the USB will be a lower proportion of total CPU time, so higher network and USB speeds are expected. For example, for instance a USB SSD goes from about 27MB/s on Pi1 to 31MB/s on Pi2." From https://www.raspberrypi.org/forums/viewtopic.php?p=683295#p683295

So I will try the performance governor if I get the BLOD after v0.7b4.

bkwny, 2016-02-29:
Just to comment on something ErgonomicMike said a couple of days ago:  I have now experienced the BLOD twice on the ground out of earshot of UAT towers.  Our Reddit thread https://www.reddit.com/r/stratux/comments/482va9/i_got_the_blod_this_morning/ has more information.

Axtel4, 2016-02-29:
Here is an interesting comment from an article about the new R-Pi 3B:

> We also ran into one problem with the onboard Wi-Fi that only occurred whilst running the new board in headless mode. This ended up being caused by a power-saving mode on the connectivity chip. Effectively the board became unresponsive in headless mode if it was connected to the network only via Wi-Fi.

http://makezine.com/2016/02/28/meet-the-new-raspberry-pi-3/

Ergonomicmike, 2016-02-29:
I recall duecedriver [sic] asking about power saving mode for the Edimax here in the github. I looked it up when he did and it seems that the Edimax is either on or off . (That is, no 50% power mode.) And the code in the Stratux is currently set to keep it on all the time. (Although as I write this, I wonder if it's possible that the code that turns the Edimax off is generated randomly?) 

Anyway, I was hoping that the Pi 3 might be a way around the problem. Given the article, perhaps it won't. (We don't have the ability to set  "'keep-alive' in our ssh client.") But just in case the same thing is happening to us, is there a way to fake out the Pi 2 in software to tell it we're not in headless mode? So far it has been noted that the BLOD never occurs when a Pi is hooked up to Ethernet. (Presumably a continuous SSH connection.) As a test for this, is there a resistor that I could put from keyboard to ground that makes the board think I'm not headless?

bkwny, 2016-02-29:
I have experienced BLOD with my wired Ethernet connected.

Ergonomicmike, 2016-02-29:
I saw in the reddit that you weren't able to cure today's BLOD via going to the web UI via Ethernet. Have you gotten the BLOD when you're running a console via SSH?

Ergonomicmike, 2016-02-29:
Pin 19 on an HDMI connector is "Hot Plug Detect." If I put a pull up resistor on that pin, will the Pi think I have a monitor hooked up? Or does the Stratux code set headless in software. (Don't think it does 'cause in the early days, people were hooking up monitors to set things.)

cyoung, 2016-02-29:
Stratux doesn't do anything with the HDMI output. We can turn off some of it to save a little power, but have not. Take a look at the `tvservice` command if you're playing with it.

bkwny, 2016-02-29:
I have not kept track of whether I've had an ssh session open into Pi when the Edimax blue light extinguishes.  I can tell you, though, that if I did, I would have been ssh'ing into Pi through the wired Ethernet.

mooneyflier, 2016-03-01:
I ran across this information on how this person kept his wifi adapter from going to sleep. The info about the adapter going to sleep is about half way down the page. This looks like it may be relevant, disregard if not: http://www.jeffgeerling.com/blogs/jeff-geerling/edimax-ew-7811un-tenda-w311mi-wifi-raspberry-pi

Ergonomicmike, 2016-03-01:
Thx. Those lines are already in the config. (I'm currently trying one more switch that might keep the USB from sleeping.)

I'm guilty of confusing the BLOD issue a bit. I don't think that the core problem is that the Edimax is going to sleep. Rather, I think something causes the USB controller to hang/fail, and that, in turn, causes the Edimax to put itself to sleep. (It's easy to get misdirected by focusing on the blue light. But the light going out is a symptom. Not the disease.)

Nokomis449, 2016-03-03:
To add some mud to the water, I just looked closely at the flashing light and realized that if I ever get a BLOD, it will actually be a GLOD because I've been using a TP-LINK dongle with a green light for some time now.  I must've accidently slipped it in there when I wasn't looking.  It's the same dongle that was removed from the "parts" list a few months ago due to everybody having issues with it but I never did.  At least no issues that I could definitely blame on the TP-LINK and not inadequate power.  Not sure why or how long it's been in this particular Stratux, but I'll slip an EdiMAX in there for this weekend's flights and see what I get.  FWIW, the only issue I get these days, and it also happens on a Stratux with an EdiMAX, is the 1090 SDR will randomly stop working.  Doing nothing but going into the Setup page and cycling the 1090 off & back on will get it working again.

Ergonomicmike, 2016-03-03:
The 1090 dropping out might be a new issue w/ v0.7b3 alpha.

I thought it was just me. But  sometimes it seems like it doesn't get started on boot. I reboot and suddenly there's traffic. But I'll have to fly with it confirm.

In the meantime, will be interested to see what you find w/ the Edimax. I just upgraded my kernel to jessie. So will be interestdd to see if that fixes it.

skypuppy, 2016-03-03:
Mike, please keep me informed about your efforts with jessie.  In fact, 
it might be fun to email each other directly when the discussion doesn't 
apply to stratux as a whole.

I don't know if this is wise to post on here (or even if reddit will cut 
it) but my email is ainut@hiwaay.net.

Skypuppy aka David M.

Ergonomicmike, 2016-03-09:
@Nokomis449 I saw that you did a lot of flying recently. Did you put the Edimax back in? If you did, I trust no BLOD's? (Or we would have heard about it.)

Nokomis449, 2016-03-09:
Yes; EdiMAX and rock solid Stratux performance.  Up until these flights I've had occasional/random SDR lockups where traffic went to 0.  I used to resolve these by rebooting the Stratux, but then I discovered they could be resolved by simply going into the setup page, toggling off the 1090, and toggling it back on.  Before these latest flights I reimaged with the latest image as of last Wednesday, and then did an .sh update as of last wednesday.  Like I said, rock solid performance.  I ran the same system overnight for 2 nights, rock solid.  I'm very grateful to the devs and people like you that have tested the bejezus out of this.  Anxiously awaiting the RO milestone.

Ergonomicmike, 2016-03-09:
Based on what Jeff reported and my last flight, I'm going to close this issue now. (Although bkwny has been reporting BLOD's.) If I or anyone else should get a BLOD with  v0.8r1 or greater, we can report it as a new issue.

closed 2016-03-09 by Ergonomicmike