cyoung/stratux #286: Not able to download  v0.7b3

mooneyguy, 2016-02-24:
I have been download the image updates since v0.5b1 just fine. Now when I try to save v0.7b3 I get a quick blink and then a notice it is saved but when I go to the usual zip file the file seems it is just a small or corrupt with a white page instead of the yellow zip folder and no application I have opens it.

skypuppy, 2016-02-24:
Wait a few seconds after it says done before you try to open it. 
Sometimes it is still in the download buffer and not written fully to 
the file.  Other times the download gets corrupt so wait a minute and 
try to download it again.

mooneyguy, 2016-02-24:
@skypuppy I have waited over night and it is still the same thing. Tried it all over today and got the same result.

skypuppy, 2016-02-24:
Sounds like a corrupt file, then.

mooneyguy, 2016-02-24:
It will usually take a few minutes to down load and then says the down load was successful but now it is all over in 3 seconds and says it was successful.

peepsnet, 2016-02-24:
Try a different browser. My firefox doesnt like github and I have to use chrome...

mooneyguy, 2016-02-24:
I have tried internet explorer and chrome and have the same results.

wakkow, 2016-02-24:
I think you're downloading an update scripts. You'll need to download the last full release v0.7b1, then update it via the web interface with the update script. I haven't tried it yet, so I don't know if you hae to do b1 -> b2 -> b3 or if you can go b1 -> b3

cyoung, 2016-02-25:
@wakkow is correct.

And v0.7b1 -> v0.7b3 is fine, you don't have to do one version at a time.

closed 2016-02-25 by cyoung