cyoung/stratux #287: Can't make the latest commits to v0.7b3

Ergonomicmike, 2016-02-26:
Not really a code issue, but since I'm not on Slack, though this was the next best place to post.

I am trying to do a git pull on the latest commits. At make I get

`recipe for target 'xdump1090' failed`

I tried a few times. Is it me or something wrong with xdump?

cyoung, 2016-02-26:
You pulled just in the middle of me reorganizing some things in the repo. Give it a try now.

https://github.com/cyoung/stratux/commits/master

You tried here, I assume: b1cbc74b543b3c7507f30e1a12579c7436e6cd65.

Should probably just delete the whole directory and start with a fresh clone. `rm -rf stratux`, `git clone https://github.com/cyoung/stratux --recursive`.

Still working on it, though.

closed 2016-02-26 by cyoung

Ergonomicmike, 2016-02-26:
I saw AvSquirrel's detailed instructions to Axtel4. But I figured that since you incorporated his commits into the Release page, that it would be the run of the mill git pull; make; make install.

I'll wait. Won't be flying for a few days.

Ergonomicmike, 2016-02-27:
Later Friday: I did a git pull; make; make install. That works now.

Unfortunately, Stratux stopped reporting traffic after that. No messages in either UAT or 1090. No traffic on the Traffic page and no traffic making it to iFly.

In the Settings page, the SDR, GPS, and logging enabled are constantlly disabled. When I enable one of them, they alll slide to enabled, the way I had them before. But the settings donlt persist. But when I come back to the Settings page, they're disabled again. (Although the status page shows both gas gauges. It also shows my correct GPS status.)

I've cleared the browser cache. N/C. 

I do not have the SDR's named. (Never wrote to them.)

peepsnet, 2016-02-27:
Ergo,

Same for me here!!  I have the same issues

I got a new SD card and loaded the v0.7b1 img and git pull to 0.7b3-latest commits and the same

Ergonomicmike, 2016-02-27:
I didn't notice weird icons. Where do I look?

peepsnet, 2016-02-27:
checkout my stratux
http://stratux.no-ip.org

peepsnet, 2016-02-27:
[image]

Ergonomicmike, 2016-02-27:
Actually, it looks like it's working for you. If by "weird" icons, you're talking about the new icons on the Traffic page, those are new. They depict where the original message originated. (Ground or air.)

Ergonomicmike, 2016-02-27:
I'm finding that the new changes have made the web UI even more browser specific. Foe example, the behavior I reported above were on the latest version of Chrome for Android. Whereas I get better behavior in a different Android browser.

JohnOCFII, 2016-02-27:
> checkout my stratux
> http://stratux.no-ip.org

Looks normal to me.  I'm using Safari on a Mac to look at the site.

peepsnet, 2016-02-27:
Is the settings page loading for you??  I get nothing on PC/chrome PC/Firefox Android/chrome android/internet android/firefox but the icons are correct in PC/firefox and Android/Firefox

peepsnet, 2016-02-27:
And the GPS AHRS page has no graphics

Ergonomicmike, 2016-02-27:
Everything presents fine here on you UI when I link to your connection. (Spooky. I was able to disable your dongles? Could I command a reboot?)

Question: Had you done the eeprom thing to your SDR's?

I'm still not getting anything on iFly here. I suppose my 1090 SDR could have died.

(BTW, what's the Watch List on the Settings page?)

cyoung, 2016-02-27:
@peepsnet: cached files in your browser? Looked at your stratux link, works fine for me.

peepsnet, 2016-02-27:
Ok someone load my page and turn off both my SDR's... let me know when and I will rtl_eeprom and see if they have really disconnected

Ergonomicmike, 2016-02-27:
Since the commits are working for you (at least thru your internet interface), I'll reimage and then pull to latest. (Swapping SDR's here made no difference.)

ayarbro, 2016-02-27:
I'm having similar issues with the latest as well.  Same error Erg has, with the error 2.  I can make install after the error, but 1090 no longer works, and the web interface is very slow/unresponsive in chrome, the icons have become squares.,

cyoung, 2016-02-27:
`rm -rf stratux ; git clone https://github.com/cyoung/stratux --recursive` then try again

ayarbro, 2016-02-27:
Sorry cyoung, same "error2"

root@raspberrypi:/home/pi/stratux# rm -rf stratux ; git clone https://github.com/cyoung/stratux --recursive
Cloning into 'stratux'...
remote: Counting objects: 3477, done.
remote: Compressing objects: 100% (100/100), done.
remote: Total 3477 (delta 61), reused 0 (delta 0), pack-reused 3377
Receiving objects: 100% (3477/3477), 23.42 MiB | 2.31 MiB/s, done.
Resolving deltas: 100% (2182/2182), done.
Checking out files: 100% (185/185), done.
Submodule 'dump1090' (https://github.com/AvSquirrel/dump1090) registered for path 'dump1090'
Submodule 'linux-mpu9150' (git://github.com/cyoung/linux-mpu9150) registered for path 'linux-mpu9150'
Cloning into 'dump1090'...
remote: Counting objects: 4354, done.
remote: Compressing objects: 100% (57/57), done.
remote: Total 4354 (delta 36), reused 0 (delta 0), pack-reused 4297
Receiving objects: 100% (4354/4354), 5.68 MiB | 2.26 MiB/s, done.
Resolving deltas: 100% (2955/2955), done.
Submodule path 'dump1090': checked out 'b360db74b99a408cb83833af802042d29cccffd9'
Cloning into 'linux-mpu9150'...
remote: Counting objects: 186, done.
remote: Total 186 (delta 0), reused 0 (delta 0), pack-reused 186
Receiving objects: 100% (186/186), 102.30 KiB, done.
Resolving deltas: 100% (100/100), done.
Submodule path 'linux-mpu9150': checked out '56658ffe83c23fde04fccc8e747bf9b7c1e184ea'
root@raspberrypi:/home/pi/stratux# make
make xdump978
make[1]: Entering directory '/home/pi/stratux'
cd dump978 && make lib
make[2]: Entering directory '/home/pi/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o 
make[2]: Leaving directory '/home/pi/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
make[1]: Leaving directory '/home/pi/stratux'
make xdump1090
make[1]: Entering directory '/home/pi/stratux'
git rm --cached dump1090
fatal: pathspec 'dump1090' did not match any files
Makefile:20: recipe for target 'xdump1090' failed
make[1]: *** [xdump1090] Error 128
make[1]: Leaving directory '/home/pi/stratux'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2

peepsnet, 2016-02-27:
Is this stratux.conf correct??
{"UAT_Enabled":true,"ES_Enabled":true,"GPS_Enabled":true,"NetworkOutputs":[{"Conn":null,"Ip":"","Port":4000,"Capability":5,"LastUnreachable":"0001-01-01T00:00:00Z","SleepFlag":false},{"Conn":null,"Ip":"","Port":4001,"Capability":5,"LastUnreachable":"0001-01-01T00:00:00Z","SleepFlag":false},{"Conn":null,"Ip":"","Port":4002,"Capability":5,"LastUnreachable":"0001-01-01T00:00:00Z","SleepFlag":false},{"Conn":null,"Ip":"","Port":43211,"Capability":5,"LastUnreachable":"0001-01-01T00:00:00Z","SleepFlag":false}],"AHRS_Enabled":true,"DEBUG":true,"ReplayLog":false,"PPM":0,"OwnshipModeS":"000000","WatchList":"KFLL KFXE KMKY KAPF"}

AvSquirrel, 2016-02-27:
Something changed in Chrome today. I reverted to v0.6r1 and saw the same thing (boxes replacing unicode characters, JS not updating).

cyoung, 2016-02-27:
@ayarbro: you're in the wrong directory.

`cd .. ; rm -rf stratux ; git clone https://github.com/cyoung/stratux --recursive ; cd stratux`

peepsnet, 2016-02-27:
This is interesting also...

[image]

peepsnet, 2016-02-27:
@cyoung my copy of stratux is on /root/stratux

Ergonomicmike, 2016-02-27:
@peepsnet Yeah, I get that too with some browsers. The Project might need to specify minimum system requirements.

peepsnet, 2016-02-27:
ok well 
192.168.1.51/getSettings
does not match
/etc/stratux.conf

aren't these the same file?

ayarbro, 2016-02-27:
@cyoung wow, what a dumbass. K, makeing now.  Thank you!

Ergonomicmike, 2016-02-27:
Reimaged to v0.7b1. Set the date to GMT. In stratux directory. Did a git pull, same as I always do. No joy. Tried again. Same error.

```
github.com/ziutek/mymysql (download)
Makefile:16: recipe for target 'xgen_gdl90' failed
make[1]: *** [xgen_gdl90] Error 1
make[1]: Leaving directory '/root/stratux'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2
```

cyoung, 2016-02-27:
Don't do git pull, do as above:

`cd .. ; rm -rf stratux ; git clone https://github.com/cyoung/stratux --recursive ; cd stratux`

The switch of repos for dump1090 messed up some stuff.

Ergonomicmike, 2016-02-27:
It wants a "username."

cyoung, 2016-02-27:
We've got issues on top of issues here, @ayarbro - did it work for you?

cyoung, 2016-02-27:
Well, if it's not building then do the above process, I've just committed a change - rm and re-clone.

@Ergonomicmike - no idea about the username, not sure how that came up.

I'm going to hook up CircleCI to the package builder and just have it build a new update file that you guys can download for each commit. The package is built by CircleCI anyways, and we're all using the same hardware - so no reason we should all be building intermediate versions.

Ergonomicmike, 2016-02-27:
Sorry about the username thing. Typo on my part. (So I deleted the comment.)  Reimaged again, did the rm, did the recursive. Am doing a make now. Should see in a few minutes if it works for me.

Ergonomicmike, 2016-02-27:
Rats. Still erroriing out. (Below.) So I'll reimage and do a new git pull with the fiftieth commit.

```
Cloning into '/root/go_path/src/github.com/kellydunn/golang-geo'...
error: RPC failed; result=7, HTTP code = 0
fatal: The remote end hung up unexpectedly
package github.com/kellydunn/golang-geo: exit status 128
Makefile:16: recipe for target 'xgen_gdl90' failed
make[1]: *** [xgen_gdl90] Error 1
make[1]: Leaving directory '/root/stratux'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2
```

skypuppy, 2016-02-27:
well, now I'm confused.  <sigh>  (it is a familiar state of mind)

AvSquirrel, 2016-02-27:
@peepsnet  - the UI failures appear to be an appcache issue on Chrome. I opened #289 to track.

Ergonomicmike, 2016-02-27:
Okay, some success. I reimaged to v0.7b1. I SSH'd in, via puTTY. I set the date to UTC. I did Chris's rm instruction. I did a git pull clone --recursive. I changed to /stratux and did a make; make. I got an error, like above, but it worked the second time I tried. I did a make install and rebooted.

I am now seeing 1090 messages. iFly is now painting traffic. I think the rest of the problems are with the new browser code.

For example: 

When I enter the web UI, I see messages in the Status page and I see GPS acquired.. But when I go to the Settings page, it shows both SDR's, the GPS, and Records disabled. I can move the sliders, but they don't persist. They used to, but this changed with this last build. (Chrome. I cleared all the privacy stashes.)

The Weather page shows that I'm watching KBOS KATL KORD KLAX (0), even tho the Watch lIst on the Settings page is blank.

The GPS page does not show my location or a map. I suspect this is because the GPS slider keeps setting itself to disabled.

I apologize for the false starts and fat finger errors above.

AvSquirrel, 2016-02-27:
@Ergonomicmike -- ref #289. If you `rm /var/www/stratux-appcache` and refresh Chrome (ctrl-F5 a couple times) does the UI come back?

Ergonomicmike, 2016-02-27:
Well, I got a little bit ahead of you. I sent the Shutdown command to Stratux and unplugged the power. Counted to 5 and plugged it back in - all the time with Chrome on my tablet still running. When the Stratux booted back up, the browser worked as normal. (That was with stratux-appache still intact.)

So then I removed the stratux-appache directory, powered down the Pi, turned off my browser and restarted. The web UI came back as normal. But that's not a good test, since it might have fixed itself after my first effort.

So I'll reimage back to 0.7b1, pull the clone and then I'll remove stratux-appache before I start the web UI. I'll report to you what happens.

Ergonomicmike, 2016-02-27:
Okay, when I did all the above, the web UI came up like it used to. Specifically, prior to this clone pull, my UI would come up after the first boot after an image with the SDR's enabled, but the GPS disabled. (BU-353.) It's doing that again now. (Once I enable the GPS, it stays enabled.)

Also, the Traffic page started populating right away. Of course that can be a function of nearby traffic. But it seemed laggy before. Now it's snappy.

Checked it thru a reboot of the Stratux and a wipe of browser cache. It's staying good.

ayarbro, 2016-02-27:
@cyoung we're operational now.  testing new SDR with tcxo.  Only thing strange I'm seeing are web UI issues.  Which would be boxes instead of icons in chrome, but everything normal in Safari.  However both browsers have slow responsiveness for the settings page.  Have to click a setting, then all the settings properly display.  Also the Watching list settings don't seem to be reflected in the weather page, instead I'm seeing "KBOS KATL KORD KLAX". Would you like me to open a new issue for each of these?  Or are they already filed?

Sorry for the slow responses, had a visitor show up to my house unannounced that I hadn't seen in sometime.  Few hours of catch up time.

Ergonomicmike, 2016-02-27:
Go up to AvSquirrel's last comment. You need to SSH in and delete the stratux-appcache directory.

Ergonomicmike, 2016-02-27:
Yeah, it's .appcache in mine too. I thought that's what it was, but quoted AvSquirrel's comment instead. It's been a long day & I bet we're all tired.

skypuppy, 2016-02-27:
This is just my personal opinion. but I don't think we should write to 
Chrome's limitations.  Writing to Firefox as a baseline should be our 
focus.  It is up to Google <censoreds> to match the industry, not us 
matching Googles bugs (er, features.)

Skypuppy