cyoung/stratux #288: make error with current clone of git build 2/26/16 5:30PM PST

ayarbro, 2016-02-27:
1. Stratux version: ~0.7b3
2. Stratux config: dual SDR, GPS yes (RY...), AHRS yes, AC
3. latest Foreflight/ipad air
4. I attempt to build the latest from the git repo and get this:
"
root@raspberrypi:/home/pi/stratux# make
make xdump978
make[1]: Entering directory '/home/pi/stratux'
cd dump978 && make lib
make[2]: Entering directory '/home/pi/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o 
make[2]: Leaving directory '/home/pi/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
make[1]: Leaving directory '/home/pi/stratux'
make xdump1090
make[1]: Entering directory '/home/pi/stratux'
git rm --cached dump1090
rm 'dump1090'
git submodule update --init
Submodule 'linux-mpu9150' (git://github.com/cyoung/linux-mpu9150) registered for path 'linux-mpu9150'
Cloning into 'linux-mpu9150'...
remote: Counting objects: 186, done.
remote: Total 186 (delta 0), reused 0 (delta 0), pack-reused 186
Receiving objects: 100% (186/186), 102.30 KiB, done.
Resolving deltas: 100% (100/100), done.
Submodule path 'linux-mpu9150': checked out '56658ffe83c23fde04fccc8e747bf9b7c1e184ea'
cd dump1090 && make
make[2]: Entering directory '/home/pi/stratux/dump1090'
make[2]: *** No targets specified and no makefile found.  Stop.
make[2]: Leaving directory '/home/pi/stratux/dump1090'
Makefile:20: recipe for target 'xdump1090' failed
make[1]: *** [xdump1090] Error 2
make[1]: Leaving directory '/home/pi/stratux'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2
"

Worked last night, something change in the make file?

closed 2016-02-27 by ayarbro

cyoung, 2016-02-27:
Fixed it?

AvSquirrel, 2016-02-27:
@cyoung -- I saw the same thing. Problem is [here](https://github.com/cyoung/stratux/blob/master/Makefile#L20) and [here](https://github.com/cyoung/stratux/blob/master/Makefile#L29).

`git rm --cached` removes the submodules from the project, and therefore prevents it from being able to pull them. I removed those statements and it compiled normally.

cyoung, 2016-02-27:
Perhaps - why does CircleCI not fail then?

AvSquirrel, 2016-02-27:
Nonrecursive `git clone` removes the submodule from the project before anything calls for dump1090.
````
root@raspberrypi:~# git clone https://github.com/cyoung/stratux stratux-testing
Cloning into 'stratux-testing'...
remote: Counting objects: 3477, done.
remote: Compressing objects: 100% (100/100), done.
remote: Total 3477 (delta 61), reused 0 (delta 0), pack-reused 3377
Receiving objects: 100% (3477/3477), 23.42 MiB | 1.86 MiB/s, done.
Resolving deltas: 100% (2182/2182), done.
Checking out files: 100% (185/185), done.
root@raspberrypi:~# cd stratux-testing/
root@raspberrypi:~/stratux-testing# git status
# On branch master
nothing to commit (working directory clean)
root@raspberrypi:~/stratux-testing# git submodule
-b360db74b99a408cb83833af802042d29cccffd9 dump1090
-56658ffe83c23fde04fccc8e747bf9b7c1e184ea linux-mpu9150
root@raspberrypi:~/stratux-testing# make
make xdump978
make[1]: Entering directory '/root/stratux-testing'
cd dump978 && make lib
make[2]: Entering directory '/root/stratux-testing/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o
make[2]: Leaving directory '/root/stratux-testing/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
make[1]: Leaving directory '/root/stratux-testing'
make xdump1090
make[1]: Entering directory '/root/stratux-testing'
git rm --cached dump1090
rm 'dump1090'
git submodule update --init
Submodule 'linux-mpu9150' (git://github.com/cyoung/linux-mpu9150) registered for path 'linux-mpu9150'
Cloning into 'linux-mpu9150'...
remote: Counting objects: 186, done.
remote: Total 186 (delta 0), reused 0 (delta 0), pack-reused 186
Receiving objects: 100% (186/186), 102.30 KiB, done.
Resolving deltas: 100% (100/100), done.
Submodule path 'linux-mpu9150': checked out '56658ffe83c23fde04fccc8e747bf9b7c1e184ea'
cd dump1090 && make
make[2]: Entering directory '/root/stratux-testing/dump1090'
make[2]: *** No targets specified and no makefile found.  Stop.
make[2]: Leaving directory '/root/stratux-testing/dump1090'
Makefile:20: recipe for target 'xdump1090' failed
make[1]: *** [xdump1090] Error 2
make[1]: Leaving directory '/root/stratux-testing'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2
root@raspberrypi:~/stratux-testing# git submodule
 56658ffe83c23fde04fccc8e747bf9b7c1e184ea linux-mpu9150 (heads/master)
root@raspberrypi:~/stratux-testing# git status
# On branch master
# Changes to be committed:
#   (use "git reset HEAD <file>..." to unstage)
#
#       deleted:    dump1090
#
root@raspberrypi:~/stratux-testing#
````

Recursive clone copies dump1090 into the local filesystem; gcc is happy since it has something to work on. But, once the build completes, both linux-mpu9150 and dump1090 submodules are marked for deletion. If you made a commit and pushed it, they'd be removed from the project. Circleci may not care, but you do.
````
root@raspberrypi:~# git clone https://github.com/cyoung/stratux stratux-testing --recursive
Cloning into 'stratux-testing'...
remote: Counting objects: 3477, done.
remote: Compressing objects: 100% (100/100), done.
remote: Total 3477 (delta 61), reused 0 (delta 0), pack-reused 3377
Receiving objects: 100% (3477/3477), 23.42 MiB | 2.30 MiB/s, done.
Resolving deltas: 100% (2182/2182), done.
Checking out files: 100% (185/185), done.
[... 43 lines cut ...]
make[2]: Entering directory '/root/stratux-testing/dump1090'
[... a bunch of compiler messages removed...]
make[2]: Leaving directory '/root/stratux-testing/dump1090'
make[1]: Leaving directory '/root/stratux-testing'
make xlinux-mpu9150
make[1]: Entering directory '/root/stratux-testing'
git rm --cached linux-mpu9150
rm 'linux-mpu9150'
git submodule update --init
[... a bunch of other compiler messages removed...]
root@raspberrypi:~/stratux-testing# git status
# On branch master
# Changes to be committed:
#   (use "git reset HEAD <file>..." to unstage)
#
#       deleted:    dump1090
#       deleted:    linux-mpu9150
#
# Untracked files:
#   (use "git add <file>..." to include in what will be committed)
#
#       dump1090/
#       linux-mpu9150/
root@raspberrypi:~/stratux-testing# git submodule
````

ayarbro, 2016-02-27:
Sorry @cyoung , I closed it as soon as I saw ErgonomicMike had already posted a similar issue, then fixed it.  I'll look better next time prior to posting.  Thank you for your attentiveness, on this freeware project!  I hope I can contribute by writing a how-to or tutorial at some point.