cyoung/stratux #289: Stratux.appcache changes break web UI on Chrome 48

AvSquirrel, 2016-02-27:
1. Stratux version:v0.7r3 (last @cyoung commit),
2. Stratux config: dual SDR, GPS (RY835AI GPIO), AHRS yes, 2A AC power adapter.
3. Browser / OS: Chrome 48.0.2564.116m and 48.0.2564.109 m on Windows 10 x86-64

4. Description of your issue. https://github.com/cyoung/stratux/commit/4aa70c0dea1d8f2e7b65045d344501c919e0d1d1 modifed the format of the cache manifest. After upgrading to the latest commit via 'git pull' release, the web UI is broken in Chrome 48 on Windows 10 x86-64. Icons are displayed as rectangular boxes, and the javascript console shows continuous errors.

Removing the manifest file from /var/www and hard-refreshing the page (Ctrl-F5) restores normal functionality. Reinstalling the manifest file and hard-refreshing causes the error to return.

Other browers (iOS Safari, MS Edge, Firefox) do not appear to be affected.

peepsnet, 2016-02-27:
Yup.
mv stratux.appcache stratux.appcache.off

all back to normal!!  wow!!! 

Fixed all my issues with #287 

Settings page populates on tablest/phone/pc now!!

closed 2016-02-27 by cyoung

AvSquirrel, 2016-02-27:
Excellent. Thanks.

commit e97b7f5e (2016-02-27): Fixes #289.