cyoung/stratux #290: Latest commits for 0.7b3 Build Error

Axtel4, 2016-02-27:
I am trying to build the latest version with the newest dump1090 commits.  I have the bolded error and I am not sure what to try.

root@raspberrypi:~/stratux# make
make xdump978
make[1]: Entering directory '/root/stratux'
cd dump978 && make lib
make[2]: Entering directory '/root/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o
make[2]: Leaving directory '/root/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
make[1]: Leaving directory '/root/stratux'
make xdump1090
make[1]: Entering directory '/root/stratux'
git submodule update --init
Submodule 'dump1090' () registered for path 'dump1090'
Submodule 'linux-mpu9150' () registered for path 'linux-mpu9150'
**fatal: reference is not a tree: b360db74b99a408cb83833af802042d29cccffd9
Unable to checkout 'b360db74b99a408cb83833af802042d29cccffd9' in submodule path 'dump1090'**
Makefile:20: recipe for target 'xdump1090' failed
make[1]: *** [xdump1090] Error 1
make[1]: Leaving directory '/root/stratux'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2`

Ergonomicmike, 2016-02-27:
Go here to the "closed" issue. We've been discussing it even tho it's closed.

Go to the last few comments and work your way up from the bottom. Chris explains the special git clone -- recursive method you have to do to get these commits installed.

Axtel4, 2016-02-27:
I got rid of the fatal build error.  Thanks @Ergonomicmike for pointing me to that thread.  I now appear to have a new issue.  

I no longer have the Signal Strength displayed on the Traffic Page like it was in the avsquirrel-statux version that I built the other day.  I do still have the Tower Signal Strength on the Tower Page.  I am seeing this on my device Web UI App, my device Chrome browser, and my desktop Chrome browser.

Here is the latest version reported by Stratux:
> root@raspberrypi:/stratux# git describe --abbrev=0 --tags
> v0.7b3
> root@raspberrypi:/stratux# git log -n 1 --pretty=%H
> 6762c01935c4ee7747074481590b0996718bf965

Here is a screen shot of the desktop Web UI traffic page:
[image]

Thanks.

AvSquirrel, 2016-02-27:
@Axtel4 -- the power level reporting didnt make the cut for #285. Those changes and a few other tweaks have been submitted as #292. I'll be flying in about 20 minutes to validate and can share my logs if you want to analyze them.

Axtel4, 2016-02-27:
Thanks. I was going to be at the airport later today and wanted to give it a try.  I could copy the image I have now and pull in new_dump1090 and build that version.

AvSquirrel, 2016-02-28:
@axtel4: I forgot to update the git tags on my branch before I committed it. I'll fix that, as there are a couple of other minor debug items I need to commit to #292.

In any case, here are my logs from today.
Equipment list:

- Raspberry Pi 2 Model B, clear plastic case, and Edimax adapter from the Vilros kit
- NooElec NESDR Mini 2 (qty 2; cases removed)
- VK-172 GPS
- Anker E5 battery with included USB cable

I used homebuilt quarter-wave wire dipole antennas for the first 45 minutes of testing, but replaced them with the stock NooElec telescoping antennas partway through log 13. The homebuilt antennas must be too narrow bandwidth and/or not be tuned right, since reception went from zero towers to six just by swapping out for the stock antennas.

[0013-gps.log.gz](https://github.com/cyoung/stratux/files/149661/0013-gps.log.gz)
[0013-uat.log.gz](https://github.com/cyoung/stratux/files/149630/0013-uat.log.gz)
[0013-es.log.gz](https://github.com/cyoung/stratux/files/149631/0013-es.log.gz)
[0014-gps.log.gz](https://github.com/cyoung/stratux/files/149632/0014-gps.log.gz)
[0014-uat.log.gz](https://github.com/cyoung/stratux/files/149672/0014-uat.log.gz)
[0014-es.log.gz](https://github.com/cyoung/stratux/files/149668/0014-es.log.gz)

Axtel4, 2016-02-28:
@AvSquirrel here is an interesting question,

Why does KMSP really have three unique tower locations within the airport grounds?  

These coordinates show up in both UAT logs for tower(s) located at KMSP:
44.873486, -93.22777
44.881103, -93.208222
44.884601, -93.210905

It would be interesting to know if all three coordinates have the same tower ID.

As a sanity check, the other tower locations I see in the logs are near KDTL, Woodbury, MN, Chanhassen, MN, Russell, MN.

peepsnet, 2016-02-28:
At KFLL there are 2 towers on the field also

AvSquirrel, 2016-02-28:
@axtel: SWAG -- separate transceivers for surface, terminal, and enroute service volumes? Your decoding of the locations checks out based on what I've seen.

cyoung, 2016-02-28:
Try updating using these files instead of git pull etc: http://updates.stratux.me/builds/.

They should be available within a few minutes after a new commit.

closed 2016-02-28 by cyoung

Axtel4, 2016-02-28:
If there an order in which to apply these updates?

Ergonomicmike, 2016-02-28:
You can just apply the last one (as Chris has said a few times here and in the reddit). BUT - if you did any gits to your build, then you have to image back to stock. (At least I had to to get the Edimax fix to work.)

Axtel4, 2016-02-28:
Sorry about posting in a closed thread, but I am confused on how to use the "sh" method.  In the post above cyoung states:

>"Try updating using **these files** instead of git pull etc: http://updates.stratux.me/builds/."  

"These files" implies that to update I'll need to use the multiple files that are listed.  If so, what order are they applied?  Also, the link above sends me to an update page at stratux.me, but there is also an sh update file on github.  The link on stratux.me has three files, however the github link only has one file.  The commit number in the file name on the github site does not appear to match any of the commit numbers on the stratux.me site.

The link above sends me to updates.stratux.me, however, I don't see an "update" link on the stratux.me page.  Am I missing it or is the only way there to access the download area through a typed link?

Also, why does the git pull update process break the .sh update process?

Thanks.

cyoung, 2016-02-29:
Apply only the last one, that will include all of the changes of the previous files.