cyoung/stratux #293: 1090 range problem with v0.7b3 with 52 commits?

Ergonomicmike, 2016-02-27:
1. Stratux v0.7b3 with 52 commits from last night
2. Stratux config: dual SDR, GPS yes (BU-353), AHRS no, 2A wall wart for testing.
3. EFB app and version (iFly v9.4 RC2). EFB platform (Android KK). EFB hardware (Various tablets).
4. Description of your issue.

I've been testing the last clone build at home with multiple clients to see if I can get a BLOD in the Eidimax. While testing, I noticed that I'm not getting as much 1090 traffic as I usually do during the day in Phx. Usually I can see aircraft at my local airport from here. I can't with the latest clone build.

So I took another microSD card, imaged it back to v0.7b1 and did the .sh to b3. Swapped it into the Pi outdoors. Suddenly I'm getting more traffic and can see to my local airport.

Before spending any resources on this, I suppose the smart things to do is to see if others are noticing reduced gain with the lastest commits.

In the meantime, I haven't changed the gain of the SDR. I don't recall reading that gain was changed. Could it be that traffic is being dropped if it doesn't meet a quality threshold? Or being lost by some new parsing scheme? (Wild ass guesses.)

Ergonomicmike, 2016-02-28:
So, dumb question: AvSquirrel said "#285 will cause targets to properly time out if no valid position fix is received, even if speed or altitude data is being received."

I read the changes for #285 but didn't see anything obvious about aircraft timing out. 

How long is the timer?

AvSquirrel, 2016-02-28:
@Ergonomicmike - It's less that I changed the timer, and more that I applied more aggressive criteria to remove stale traffic.

Two things that make 1090 traffic challenging are 1) that we can receive one of about nine different Mode S message formats for any given aircraft, and only one of those message types gives us position information, and 2) at least three of those message (ES position ODD, ES position EVEN, ES velocity) have to be received each second to give us up-to-date position and speed on any target. And unfortunately, ES position messages are longer than non-ES altitude messages, so we're less likely to pick them up if we're on the edge of reception.

Up through v0.7b2, 1090 traffic would be sent as soon as we received a valid position message corresponding to that 24-bit IACO address. We'd keep that airplane in memory and continue sending it each second as long as we'd received *any* Mode S messages from that target in the past 60 seconds. This led to strange situations like what @peepsnet was seeing in #241: once traffic landed, they "piled up" on the runway because he stopped getting position messages, but somehow kept picking up ident / altitude messages.

In v0.7b3, the logic was slightly modified when I added the "Age" characteristic. The timeout for sending messages was reduced to six seconds since the last Mode S message was received, but we still kept them in memory (and on the UI) for sixty.

My changes on #285 reset the criteria for what counts as a "recent message". Now, a valid *position* (not just altitude, speed, or a squawk code) has to have been seen to keep the timer alive. This is going to cut down on the number of targets that update sporadically -- e.g. if you're only picking up a position message from planes at the local airport every 20-30 seconds, they all would have been visible with the old code. With the new code, they'd get timed out. This is more consistent with what the certified units do; eventually I'm planning to put in a "coast mode" where position will be predicted as soon as we go more than a second without a position update.

Ergonomicmike, 2016-02-29:
@AvSquirrel The code today seems to work well and traffic populates quickly after I reconnect the SDR antenna. Sp feel free to close this issue after you've viewed the video below about coasting. (Or, I will close it after.)

Okay on all above. My concern was that I wasn't seeing as many 1090 messages the other day because the new code might be discarding "bad" traffic too soon (one second) if a position message was invalid. (Which, from what you said about when coast mode would kick in, seems to be the window that you're using.) 

{I saw your comment in a commit about how you flew and had to change something in the code, so perhaps that fixed the traffic count today.}

My gut says that a second or two discard is too quick, but I have no data to support that. I guessing that a valid signal could be partially corrupted by short term phenomenon - signal blockage (air carrier in a shallow bank), frequency collisions in high traffic environments, lighting burst near a line of TRW's, etc. I suppose a coast mode would mitigate my concerns about dropping aircraft too soon. Although if the time to discard were, say, 5 seconds, then you might not need to implement a coast mode.

Now, about coast mode. I don't know how long you plan to coast. I recall having this discussion before. But for completeness, here's a short video of iFly's built-in coast mode. 
https://vid.me/KpBR

So if you coast and they coast, it could make for a lot of coasting. 

From an philosophical OSI point of view, should the Stratux coast at all? I view the Stratux as a Layer 1 & 2 device, that supplies raw data to EFB's. The EFB's are high Layer devices (Application layer) that take the raw data and play with it as they wish. So I'm seeing the Stratux as "dumb," in that it doesn't operate on the raw data. It's job is to feed the raw data to a "smart" terminal. How smart should the Stratux be? 

I don't the answer to my question. Is coasting at the Stratux level is something supported in Industry specs? I did a very cursory search and found [an FAA doc for GDL-90](http://www.faa.gov/nextgen/programs/adsb/wsa/media/GDL90_Public_ICD_RevA.PDF). It says "No report extrapolation or coasting is performed." (at 3.6.)

Axtel4, 2016-02-29:
> My gut says that a second or two discard is too quick, but I have no data to support that. 

Here is my thought; on the data I've analyzed there can be as much as a 4 - 6 second gap in received 1090ES traffic so Stratux should retain the last position at least 4 seconds before droping it.  I believe the same is true for UAT traffic, but I haven't delved into those data sets yet.

I posted an analysis of this in Issue 203:

> The approximate two seconds gaps in the track plot are due to the diversity of the Intruder Transponder.  The Intruder transponder squits the 1090ES DF-17 message at a one second interval with the following pattern: BBTTBBTT.  As the Intruder aircraft gets closer to Own Ship the Top antenna 1090ES squits are shaded by the Intruder aircraft airframe.  The shorter gaps are when the Intruder is squitting on the top antenna.  The longer four to six second gaps occur when the Intruder bottom antenna squits are also missed.  Two of the more common reasons for missing the bottom squits are Intruder aircraft antenna shading when an aircraft is in a turn, or due to packet collisions at the receiving station.

1090ES Traffic Analysis (Posted 28-Jan-2016):
1090ES Traffic Analysis.pdf

> From an philosophical OSI point of view, should the Stratux coast at all? 

I'm mixed on who should coast the target.  In the certified world, TCAS and TAS coast the targets because they format the traffic data and output it to a "dumb" terminal.  In the TIS Data-link (aka TIS-A) world, the display coasts the target and the transponder only sends the raw traffic data to the display.  (Yes, the transponder is the receiving device on 1030 MHz for TIS Data Link.)  Since TIS-B / ADS-R is built off of TIS Data Link concept I believe the display device also handles the coasting.

Axtel4, 2016-02-29:
Update:
It looks like some of the changes made it in to the .sh file, but the Web UI didn't get updated.  The ES log has what appears to be the latest data:

> 281102634840,{"Icao_addr":11344262,"DF":0,"CA":0,"TypeCode":0,"SubtypeCode":0,"SBS_MsgType":7,"SignalLevel":5,"Tail":null,"Squawk":null,"Emitter_category":null,"OnGround":false,"Lat":null,"Lng":null,"Position_valid":false,"NACp":null,"Alt":31400,"AltIsGNSS":false,"GnssDiffFromBaroAlt":null,"Vvel":null,"Speed_valid":false,"Speed":null,"Track":null,"Timestamp":"2016-02-29T00:43:24.801Z"}

Axtel4, 2016-02-29:
Oh, and I just noticed the Tower page doesn't have the added Signal data either.

jpoirier, 2016-02-29:
I see stratux as nothing more than a piece of middleware and as middleware it should be making minimal decisions concerning traffic data (eg not much more than "is it valid or not valid data") with a priority being put on speed of delivery to the EFB. 

The entire point of an EFB is to render information as quickly and as accurately as possible and it should be left up to each vendor to decide on how best to do that. There should be as close to zero overlap between stratux and an EFB as possible; it's not the job of stratux to take up the slack for a poor EFB implementation, if anything it should be able to help vendors by exposing flaws in/with their implementations.

Ergonomicmike, 2016-02-29:
Excellent input guys!

AvSquirrel, 2016-02-29:
@Ergonomicmike:

> It says "No report extrapolation or coasting is performed." (at 3.6.)

The GDL90 had two modes of operation. One was a "pass-through interface", where it literally did no processing on received UAT messages before passing them on to other devices. Raw bits in, raw bits out. That is what's described in section 3.6 of the interface spec.

The other mode -- what we're using -- was called the "traffic alert interface", as described in section 3.5 of the spec. Among other things, this mode provided for traffic alerts for nearby traffic, and extrapolated position reports:

> A Traffic Report is generated for each tracked target every second, as long as the target is
identified by CSA as a priority target. If no ADS-B message is received for a tracked target, an
extrapolated report is generated for that target. Each target is extrapolated until its track is
discontinued by CSA.

Extrapolated status is shown to downstream devices by setting [the m field](https://github.com/cyoung/stratux/blob/master/main/traffic.go#L246-L263) in the traffic message. So, we're emulating a device that has the ability to coast traffic. As long as we're following a documented standard in how we present coasted traffic, I see this as a net benefit.

Ergonomicmike, 2016-02-29:
Okay. Glad to hear that it's in the spec. Do they say how long to coast? Axtel4's six seconds sounds reasonable.

I'll close this issue now.

closed 2016-02-29 by Ergonomicmike