cyoung/stratux #294: Mode C "Tis-B Shadow" problem

peepsnet, 2016-02-28:
I was flying with the Stratux for the first time the other day and noticed a target that was at my altitude and position in my EFB(Naviator). I talked to @cyoung and was told this is a TIS-B Shadow issue because there is no way to collateral my target from the TIS-B and ME(ownship) 

He also said that there is a hex code assigned to my ship by the FAA, randomly, for my mode C target. 

SO here is my offering...

Can the Stratux software do one of 2 things. 

1) try to offer to the user via the webUI what the generated hex code might and put it in the field in the settings page so it can be filtered

OR

2) @AvSquirrel  said there is a bearing and distance field on the list for the traffic page.(https://www.reddit.com/r/stratux/comments/4836z8/if_stratux_has_a_gps_web_ui_can_show_traffic/)
Could there be a clickable link on each line of traffic that would a) copy the hex code to clipboard so it can be pasted to the settings page OR b) get/post to the webserver and add the code to the /car/stratux.conf automatically.

Just my idea as to how we might be able to fix the Mode C Tas-B Shadowing issue for usres.

Don

Nokomis449, 2016-02-28:
Most EFB's have an option to remove the ownship target.  I know iFly does, but I've never used Naviator.  You might check thru the options and find it.

peepsnet, 2016-02-28:
Naviator has the options but here are the problems with that:
1) A Mode C transponder has no hex code broadcast
2) There is no callsign/hex code data from the TIS-B if you are in a Mode C transponder A/C

SO adding your callsign and/or Hex code will not remove your target since the target that is being displayed is not "connected" to your a/c

Ergonomicmike, 2016-02-29:
As Jeff said, iFly has a Ghost Filter and it works very well. I'm Mode C and 99% of the time, iFly filters me out. (Sometimes it has to think for a few seconds when I first become lighted up in someone's puck.) Since neither iFly or the Stratux know anything about me, I deduce that iFly is using my GPS position to figure out  that "that target right next to you must be you." You can try a fully operational version of iFly for Android or iOS for 30 days for free.

So, since iFly works with my Mode C, I suggest this is a Naviator problem. (As opposed to a Stratux one.)

More generally, per a comment I just posted in https://github.com/cyoung/stratux/issues/293, philosophically, how smart should the Stratux be? I'm thinking it ought to be a low level "dumb" device that simply gathers raw data from the airwaves and feeds them to a smart Application. If the Stratux gets too smart, I can see that leading to complications/incompatibilities down the road with smart EFB's. Does the [GDL-90 spec](http://www.faa.gov/nextgen/programs/adsb/wsa/media/GDL90_Public_ICD_RevA.PDF) allow for user input to help ghost filtering? When I flew shotgun in a 182 with a guy who has a Stratux V2, I didn't see him enter the plane's Mode S. (But maybe he did it the night before when he got current.)

peepsnet, 2016-02-29:
Ergo, 
I agree the "smarts" should be left to the EFB but allowing me to easily identify my Mode C target would not require a change to the data stream. I was just looking for a fast way to identify my "radar assigned TIS-B HEX code" so I can add it to the EFB App or take advantage of the Mode S field on the settings screen. 

Am i wrong in the function of the Mode S Hex box in the settings tab. Is it not to filter the ownship target??

skypuppy, 2016-02-29:
Have to agree with ErgonomicMike on this one.

Ergonomicmike, 2016-02-29:
I understand that the Stratux let's you enter your ownship Mode S. Whether it should or not, I don't know. I'll email my Stratus buddy and ask him if there's a way to do it in the Stratus. (I won't get a response for a day.) FF and Stratus are so tightly integrated, that it might be that he enters his ownship Mode S in FF, and that, in turn, gets to the Stratus. But if it does that, that's not exactly kosher. I mean, in a symbiotic system like that, where one box is necessary for the other to live, which box is which? 

Having said that, I vaguely recall a place to enter your Mode S in Garmin's Pilot. But I don't have that app anymore. I looked in iFly and don't see a place for it. But Jeff would know better.

peepsnet, 2016-02-29:
I am looking for an option for MODE C (No Mode S).

Ergonomicmike, 2016-02-29:
Oops. My mistake. I kept seeing stuff about hex code and associated that with Mode S only. 

Even if it could be done, you wouldn't (couldn't?) know your randomly assigned Mode C phantom-code until someone lit you up. And then I wonder if your phantom code stays the same from puck to puck? I mean, out here in the desert SW, we often drop off RADAR. How would the FAA know it's you again and re-assign you the same random code?

Again, I'm not seeing this as necessary. Give iFly a try, if for no other reason to see that they're are able to do what you want without you having to do a thing. If you like how it works, then tell the guys at Naviator to do it too.

peepsnet, 2016-02-29:
I believe it would be the same. I am sure one is created for every Non-Mode S target.  The question would be if the radar system is smart enough to re-associate with the correct target after loosing coverage...  Like t/g's at an apt that is below coverage. 

I am an Air Traffic Controller at KFLL and I can tell you that we "loose" a/c that are low level along the beach as they pass behind large buildings on the beach but the radar is smart enough to preposition the tag on the radar screen as long as it hasn't traveled too far or too many other targets cross/intersect/pass.

peepsnet, 2016-02-29:
@AvSquirrel was talking about having the bearing and distance in the traffic page... This alone would make the experiment of your Mode C "HEX" code staying with you testable. 

It would allow for me to pick my "shadow" out from the traffic page quickly and see how that goes

cyoung, 2016-02-29:
There's a Mode S hex code box in the Settings page for those that have ADS-B Out / Mode S equipped aircraft. GDL90 has a provision that lets the receiving end know the identifier of Ownship - presumably to filter it, because that is a display function and not a receiver function. This definitely is a feature that belongs in the EFB. @peepsnet - did you ask Naviator what might could be done here? 

@mmiller84

peepsnet, 2016-02-29:
I will look into that.

AvSquirrel, 2016-02-29:
> @AvSquirrel was talking about having the bearing and distance in the traffic page

[image]

Prototype: https://github.com/AvSquirrel/stratux/tree/gpsattitude

(Along with prototype GPS pseudoattitude, FLARM support, and ForeFlight AHRS hack. All need a little work, but coming along nicely.)

peepsnet, 2016-02-29:
@Avsquirrel
Talking to our radar tecs here at the airport they they said the Mode C a/c are assigned a random Mode S code by the radar on the initial radar hit and that code is kept with the a/c as long as the a/c is tracked on radar. 

As I am writing this I thought about that mode s code transferring from center to center or even approach to approach so when i find the guy again ill have to ask.

Barring that, it does seem we would be able to choose "ownship" from the traffic page and set that hex code in the gdl90 message

@AvSquirrel: Could you add a button/link to each a/c in the traffic page list that would call setSettings to set the ownship value? You could even comment it out in the traffic.HTML page and I could just uncomment it to test so.it's not live.

I am flying again on Tue(3/1) if you get it up by then.

Axtel4, 2016-02-29:
@peepsnet 

Question, how will you know which one is yours and not another Mode C in close proximity?

ScraboTower, 2016-02-29:
WingX is a bit of a mystery on this one, under iOS > Settings>WingX there is a spot to enter your tail number but within the app itself, there is no such option.  So I think it might have been a thought at one point but not followed thru on.

Maybe @HiltonG can clarify

HiltonG, 2016-02-29:
@ScraboTower That Tail Number option works correctly.  I left it in Settings because it would only need to be set once (an owner in his airplane) and didn't think adding an option to the UI that most pilots wouldn't use.  I am tempted to add it, but it does work as advertised.

Hilton

ScraboTower, 2016-02-29:
Thanks Hilton.  Don't think too many folks knew it was there !!

peepsnet, 2016-02-29:
@AvSquirrel have you flown in South Florida. Im almost always in someone's puck. Lol. Its almost always  shown. I flew for 2 hours and was visible for the whole flight.

peepsnet, 2016-02-29:
@AvSquirrel at a minimum the distance and bearing will allow me to test my theory

peepsnet, 2016-02-29:
@ScraboTower  that solution is for mode s or better equipped a/c.  I am looking for a solution for mode C  equipped a/c where the a/c has no callsign or hex reportING and is only seen as a TIS-B shadow

AvSquirrel, 2016-03-01:
@peepsnet -- 

I'm thinking that (given my skillset, which doesn't have a background in web design / web apps) a toggle to flip/flop between showing callsign and ICAO code would be pretty easy. 

Adding some logic to look at distance and track to identify ownship (and have the user press a button to confirm) might be a little harder, but doable.

I'll have to dig in a bit and figure out how to make targets clickable, but I like the idea of having a popup with more detailed info than what shows on the UI, and the ability to set ownship from the popup.

peepsnet, 2016-03-02:
I was looking even simpler then that.  Just a small column after all the other data in the traffic page that is a + to click on. That would be a json call to setSettings to add that icao code to stratux.conf

Every line in the traffic page would have this + sign.  To prevent the user from seeing that column we could have a toggle in the settings that would turn that column on and off and describe what it is for

peepsnet, 2016-03-02:
Ill look into it. It will take a bit of time since I will have to trace all the webUI backend and see if I can figgure it out. Can someone point me in the direction of the source for the CSS template and anything else??

peepsnet, 2016-03-02:
I don't understand the AngularJS!  I am not able to get the functionality I am looking for. I have been out of the game too long!!

peepsnet, 2016-03-02:
@AvSquirrel  what is the array that is used for the traffic page to get the data

What does data_list look like??
https://github.com/cyoung/stratux/blob/master/web/plates/traffic.html#L24

cyoung, 2016-03-02:
https://github.com/cyoung/stratux/blob/master/notes/app-vendor-integration.md

See `ws://192.168.10.1/traffic`

AvSquirrel, 2016-03-03:
@peepsnet -- busy week. Got about an hour of time to look at this the last couple of evenings.

So, I figured out how to attach an `<a>` tag to each tail number, and used that to launch a modal  popup, but wasn't able to get it to pass the 'aircraft' variable. Following a few how-tos just caused my JS to break.

Apparently, we're using Mobile AngularUI, which is an extension / subset of AngularUI-Bootstrap. The latter supports doing all the neat things in the JS code to let me pass variables between scopes, but Mobile doesn't. I think that explains my lack of success.

I'm going to try integrating Bootstrap to see if it will let me do what I want to do.

peepsnet, 2016-03-03:
And mine!! I feel just a bit better!!  just a bit, though

Ill look into it also

peepsnet, 2016-03-03:
I don't know if this is close. Sounds similar
http://stackoverflow.com/questions/20685045/pass-argument-to-http-get-through-angular-ui-modal

AvSquirrel, 2016-03-04:
I've temporarily given up on trying to make Mobile Angular UI do what I want it to, and just added a second line of text beneath the callsign to show the ICAO address. It's quick and dirty, but at least it makes the information available without cluttering up the display too much on smaller screens.

In lieu of showing the ICAO address when callsign is unknown, I'm displaying "[unknown]".

[image]

Axtel4, 2016-03-04:
I like it.

AvSquirrel, 2016-03-04:
Two more tweaks. These are pushed to my ['gpsattitude' development branch](https://github.com/AvSquirrel/stratux/tree/gpsattitude), and I'll submit a PR to bring them into the master branch once I backport my distance / bearing math.
- Slider for showing relative distance / bearing, vs lat/lon
- Detection of GPS status (won't try to show relative distance unless detected)

[image]

[image]

[image]

Axtel4, 2016-03-04:
Are you writing the range/bearing to a file? Or, will I still need to calculate that separately for analysis?

AvSquirrel, 2016-03-05:
@Axtel4 - I'm doing the calculations in traffic.go, and just updated to using a [Law of Cosines](https://github.com/AvSquirrel/stratux/blob/dadd520a25099c398976a28a6bc33c8efed8e7bd/main/equations.go#L291-L311) method, so results will now be valid for long distances (> 60 NM) along great circles. I've tested using one of @peepsnet's UAT captures from the FLL area, and the distances / bearings calculated with my method match what ForeFlight calculates.

Anyway, if Stratux has a valid GPS position, [range and bearing are written to ti.Distance and ti.Bearing each second, just prior to pushing out the traffic data.](https://github.com/AvSquirrel/stratux/blob/dadd520a25099c398976a28a6bc33c8efed8e7bd/main/traffic.go#L167-L172) The entire traffic data structure is always available over a websocket, and is also being written (JSON formatted) to stratux log every 15 seconds if the "verbose logs" option is enabled.

I've been thinking about adding a separate traffic log CSV output for your signal analysis. What fields and what output frequency would be best for you? I could have it write one row of data for each target each second, or have it generate a message each time it receives an update on the SDRs. If I did the latter, would you want to filter out the non-ADS-B Mode S message types? 

@peepsnet -- here's an updated version of my enhanced traffic UI. This adds a separate pane for non-ADS-B Mode S traffic (with squawk codes in lieu of position), and also denotes that the 24-bit code for Mode C TIS-B traffic is the randomly-assigned track file ID, rather than an ICAO code.
[image]

peepsnet, 2016-03-05:
Great stuff! Are the squawk codes in the data stream for all the traffic now? If so i want to add it to the "good" traffic also in my version. Helps me test when I am on break at work.

AvSquirrel, 2016-03-05:
> Great stuff! Are the squawk codes in the data stream for all the traffic now? If so i want to add it to the "good" traffic also in my version.

Yep. New commit (https://github.com/AvSquirrel/stratux/commit/258a9e6439016a8c14db1325f5de998f80b95cd1) looks like this:

[image]

I added the "squawk" slider to try keeping the width down for smaller tablets. This still renders OK on my Nexus 7 in landscape mode, but portrait looks a little ugly. I'm also sorting the top table by distance, and the bottom table by ICAO code.

>Maybe a sentence at the end of the mode-s traffic table saying something like "This traffic is seen by Stratux but does not report lat/lon therefore is not able to be displayed in your EFBs.

Not a bad idea. I can put a brief blurb at the bottom, with "refer to the help page for additional information."

peepsnet, 2016-03-05:
How close are we to having it in a build?? I'm excited to muck with it

peepsnet, 2016-03-05:
I had one version in my fiddling that was 2 rows per a/c

             Flight      ICAO           Squawk      Power          Age
       Speed         Alt         Lat/Lon      Range        Dist         Bearing

cyoung, 2016-03-05:
Are the distance/bearing columns still there if a GPS is not present? It would be ideal to do like on the status page - hide the elements with blank information for lesser equipped stratux setups.

peepsnet, 2016-03-05:
If this was a for profit project then I would tell you to show all the data the user is missing from not upgrading and taunt them to buy more!!

But this is not that kinda of project. So its your call

AvSquirrel, 2016-03-05:
@cyoung - I revert to showing lat/lon if GPS is disabled.

GPS enabled:
[image]

GPS disabled:
[image]

peepsnet, 2016-03-05:
Great job. I'm ready for the build!  lol

Ergonomicmike, 2016-03-06:
Should the "Traffic Marking" option be removed from the Settings page now,? Seems like it's been rendered obsolete (and deprecated) by the new Traffic page display.

AvSquirrel, 2016-03-06:
@Ergonomicmike -- I'd leave it in for now, since there are some users who want the display on their EFBs and not open a separate browsing window. There's actually a request on #218 to add target type (ADS-B vs ADS-R vs TIS-B) to the callsign so it can be displayed on a glider computer. Image below shows that.

@peepsnet -- yeah, the headers could be more clear. Intent is to convey that the top panel tracks all targets for which Stratux has received valid positions (ADS-B, ADS-R, and TIS-B), while the bottom is primarily for tracking non-ES Mode S targets.

[image]

Axtel4, 2016-03-08:
@Avsquirrel – I have been out of town for a few days, so I am catching up. . .

 

What fields and what output frequency would be best for you?

For 1090 I use the data from Message 3 in the 1090ES log.  I use the time stamp in the first column to sync between the GPS log and the ES log, the Mode S address to locate specific targets, the Lat and Lon for the range and bearing calculations, and the Altitude field to calculate the antenna view angle.  I haven’t done anything with the 978 UAT traffic as I didn’t have a good way to decode the position information.

 

I could have it write one row of data for each target each second, or have it generate a message each time it receives an update on the SDRs.  If I did the latter, would you want to filter out the non-ADS-B Mode S message types? 

I would think once a second would be plenty.  As of now, I haven’t been using the non-ADS-B Mode S message types.  If I need to see those, I could look in the ES log.  

 

A CSV file with the Bearing, Range, Lat, Lon,  Mode S, Signal Strength, Altitude, and time stamp should cover most of the data I use for analysis.  

 

 <https://cloud.githubusercontent.com/assets/14127245/13545766/e144c454-e25b-11e5-950e-ed49505a7fc4.png>

AvSquirrel, 2016-03-08:
@Axtel4 -- this should capture what you're looking for.

It outputs to a new nnnn-traffic.log file once per second. Only ADS-B / ADS-R / TIS-B targets are included (though it's just a matter of changing one line to include basic Mode S). 

Timestamp is the RPi clock time that the message was received. Power is reported in dB. ICAO codes are hex formatted. Blanks are reported for any unknown or old (> 3 sec) position / velocity data. Ages are reported relative to the internal Stratux ticker.

````
432678443,Timestamp,Source,Type,Signal Strength (dB),ICAO code,Callsign,Latitude,Longitude,Bearing (deg true),Range (NM),Age of position fix (sec),Altitude (ft MSL),Age of altitude (sec),Track (deg true),Groundspeed (knots),Vertical velocity (ft/min),Age of velocity (sec)
5433775706,2016-03-08 06:23:20.399,978,ADS-B,-10.692,A0678C,,44.829346,-93.457611,208.398,3.8355,0.900,975,0.900,101,15,0,0.900
6433682535,2016-03-08 06:23:21.418,978,ADS-B,-7.766,A0678C,,44.829327,-93.457489,208.320,3.8342,0.900,975,0.900,101,16,0,0.900
7433690561,2016-03-08 06:23:22.54,978,ADS-B,-10.604,A0678C,,44.829304,-93.457397,208.258,3.8338,0.750,975,0.750,101,16,0,0.750
8434239315,2016-03-08 06:23:23.368,978,ADS-B,-8.971,A0678C,,44.829304,-93.457306,208.206,3.8319,0.950,975,0.950,101,16,0,0.950
9433723853,2016-03-08 06:23:23.368,978,ADS-B,-8.971,A0678C,,44.829304,-93.457306,208.206,3.8319,1.950,975,1.950,101,16,0,1.950
10433719900,2016-03-08 06:23:23.368,978,ADS-B,-8.971,A0678C,,44.829304,-93.457306,208.205,3.8321,2.950,975,2.950,101,16,0,2.950
11433674801,2016-03-08 06:23:26.389,978,ADS-B,-12.879,A0678C,,44.829239,-93.456970,207.987,3.8290,0.900,975,0.900,102,17,0,0.900
12433921577,2016-03-08 06:23:26.389,978,ADS-B,-12.879,A0678C,,44.829239,-93.456970,207.987,3.8290,1.900,975,1.900,102,17,0,1.900
13433689291,2016-03-08 06:23:26.389,978,ADS-B,-12.879,A0678C,,44.829239,-93.456970,207.985,3.8292,2.900,975,2.900,102,17,0,2.900
14433927577,2016-03-08 06:23:29.406,978,ADS-B,-16.363,A0678C,,44.829197,-93.456665,207.800,3.8256,0.900,975,0.900,99,17,0,0.900
15434254404,2016-03-08 06:23:30.48,978,ADS-B,-9.370,A0678C,1200,44.829197,-93.456543,207.730,3.8233,0.800,975,0.800,100,17,0,0.800
15434646851,2016-03-08 06:23:31.1,1090,ADS-B,-20.481,AD7633,,,,,,0.900,40000,0.150,206,454,0,0.900
16434901283,2016-03-08 06:23:31.549,978,ADS-B,-12.879,A0678C,1200,44.829174,-93.456451,207.668,3.8227,0.750,975,0.750,100,16,0,0.750
16435339042,2016-03-08 06:23:32.18,1090,ADS-B,-20.994,AD7633,,45.360409,-93.225830,15.615,29.6118,0.350,40000,0.350,207,455,-64,0.100
17434473998,2016-03-08 06:23:32.433,978,ADS-B,-11.972,A0678C,1200,44.829155,-93.456329,207.591,3.8214,0.850,975,0.850,98,16,0,0.850
17434850976,2016-03-08 06:23:33.254,1090,ADS-B,-20.591,AD7633,,45.358543,-93.227165,15.569,29.4887,0.300,40000,0.000,207,454,-64,0.000
18433955515,2016-03-08 06:23:33.501,978,ADS-B,-14.289,A0678C,1200,44.829155,-93.456238,207.538,3.8198,0.800,975,0.800,99,16,0,0.800
18434373795,2016-03-08 06:23:34.24,1090,ADS-B,-19.783,AD7633,,45.356636,-93.228439,15.528,29.3637,0.250,40000,0.250,207,454,-64,0.050
19433737240,2016-03-08 06:23:34.387,978,ADS-B,-10.663,A0678C,1200,44.829132,-93.456116,207.459,3.8186,0.900,975,0.900,99,16,0,0.900
19434119947,2016-03-08 06:23:35.29,1090,ADS-B,-19.762,AD7633,,45.354401,-93.230049,15.471,29.2161,0.200,40000,0.000,207,454,-64,0.000
20433738547,2016-03-08 06:23:34.387,978,ADS-B,-10.663,A0678C,1200,44.829132,-93.456116,207.459,3.8186,1.900,975,1.900,99,16,0,1.900
20434835888,2016-03-08 06:23:36.07,1090,ADS-B,-19.359,AD7633,,45.352478,-93.231354,15.428,29.0902,0.200,39975,0.200,207,454,-64,1.000
21433714750,2016-03-08 06:23:34.387,978,ADS-B,-10.663,A0678C,1200,44.829132,-93.456116,207.461,3.8184,2.900,975,2.900,99,16,0,2.900
21434498967,2016-03-08 06:23:37.23,1090,ADS-B,-19.139,AD7633,,45.351006,-93.232391,15.391,28.9935,0.250,39975,0.250,207,453,0,0.050
22433911995,2016-03-08 06:23:37.404,978,ADS-B,-7.787,A0678C,1200,44.829109,-93.455811,207.278,3.8136,0.900,975,0.900,95,15,0,0.900
22434389702,2016-03-08 06:23:38.24,1090,ADS-B,-19.560,AD7633,,45.349094,-93.233665,15.349,28.8685,0.250,40000,0.250,207,453,0,0.050
23433789448,2016-03-08 06:23:37.404,978,ADS-B,-7.787,A0678C,1200,44.829109,-93.455811,207.279,3.8134,1.900,975,1.900,95,15,0,1.900
23434224239,2016-03-08 06:23:39.25,1090,ADS-B,-19.914,AD7633,,45.347233,-93.235008,15.300,28.7459,0.250,40000,0.250,206,454,0,0.000
24433972266,2016-03-08 06:23:39.547,978,ADS-B,-12.690,A0678C,1200,44.829090,-93.455627,207.167,3.8109,0.750,975,0.750,90,15,0,0.750
24434447837,2016-03-08 06:23:40.22,1090,ADS-B,-20.755,AD7633,,45.345325,-93.236275,15.258,28.6213,0.300,39975,0.300,207,453,0,0.050
24434716430,2016-03-08 06:23:39.79,1090,ADS-B,-28.289,A83EF2,,,,,,0.450,11025,0.450,,,,24.400
25433781542,2016-03-08 06:23:40.945,1090,ADS-B,-20.857,AD7633,AMX008,45.345325,-93.236275,15.258,28.6215,1.300,40000,0.300,207,453,0,1.050
27621604786,2016-03-08 06:23:40.87,1090,ADS-B,-26.830,A83EF2,NKS597,,,,,3.650,,3.650,,,,27.600
27622327909,2016-03-08 06:23:40.433,978,ADS-B,-3.148,A0678C,1200,,,,,3.050,,3.050,,,,3.050
27623285406,2016-03-08 06:23:40.433,978,ADS-B,-3.148,A0678C,1200,,,,,3.050,,3.050,,,,3.050
27624011706,2016-03-08 06:23:40.945,1090,ADS-B,-20.857,AD7633,AMX008,,,,,3.500,40000,2.500,,,,3.250
27624672225,2016-03-08 06:23:40.87,1090,ADS-B,-26.830,A83EF2,NKS597,,,,,3.650,,3.650,,,,27.600
[... cut ...]
28435226555,2016-03-08 06:23:43.505,978,ADS-B,-9.193,A0678C,1200,44.829132,-93.455261,206.981,3.8015,0.800,975,0.800,82,15,0,0.800
29433987657,2016-03-08 06:23:45.151,1090,ADS-B,-20.903,AD7633,AMX008,45.335873,-93.242775,15.024,28.0019,0.400,40000,0.400,206,454,0,0.150
29434400781,2016-03-08 06:23:45.05,1090,ADS-B,-27.151,A83EF2,NKS597,44.703880,-92.895905,116.080,24.6513,1.300,11125,0.300,132,309,1088,0.250
29434753176,2016-03-08 06:23:44.628,978,ADS-B,-13.271,A0678C,1200,44.829132,-93.455170,206.931,3.7995,0.700,975,0.700,87,15,0,0.700
````

Axtel4, 2016-03-09:
@AvSquirrel,

To create and access the file you have above do I need to switch to your branch and build from there?  

If I need to switch and since I have been updating via the .sh method, do I need to git pull / make / install to bring my repository up to date before switching to your branch?

Nokomis449, 2016-03-09:
Apologies if this is the wrong thread or has already been pointed out.  Today on the ground, I had a Stratux connected to iFly.  I tapped on a nearby aircraft icon that interested me and kept the info box displayed.  At first the aircraft's ID had 4 uuuu's and some data.  Suddenly, its ID changed to one u and some data, then 2 u's and some data, then 3 u's and some data, then back to 4 and some data.  After 15 seconds or so, it ran thru the process again.  After a bit, the data changed to what looked like a valid tail number.  This was on the most recent image updated to the most recent .sh file as of last Wednesday.  Screenshots below.  It's the same aircraft; the vehicle I was in was moving and turning:
[image]

[image]

[image]

[image]

[image]

[image]

peepsnet, 2016-03-09:
Do you have the traffic(something, I cannot remember) ... the setting above the logging turned on?

Look at the help menu for the explination.

Nokomis449, 2016-03-09:
Yes, I do.  I know the "u" means it's coming from the UAT SDR.  Someone was recently working on that part of the code, can't remember who.  Seems like they were cleaning it up so it wouldn't append more than one "u" anymore; that's why I'm reporting this.

AvSquirrel, 2016-03-09:
@Axtel4 
> To create and access the file you have above do I need to switch to your branch and build from there?

If you feel like living on the bleeding edge, I built an update for https://github.com/AvSquirrel/stratux/commit/fd936e7e1b71429abb4e853f8a7116078a3da718 which is my latest development commit. Unzip this file and install the .sh through the web UI : [update-stratux-v0.8r1-fd936e7e1b.sh.zip](https://github.com/cyoung/stratux/files/164529/update-stratux-v0.8r1-fd936e7e1b.sh.zip)

In addition to the enhanced traffic UI and traffic logging, this fixes the repeated "uuu" or "eee" @Nokomis449 saw when traffic markings are enabled. 

I should also explain some of the new experimental features that you'll see on the UI settings page:
- Demo traffic. Creates a couple dozen fake targets and flies them around your position at various altitudes. At various points around their circle, I turn the "position valid" flag off so they disappear, and when they reappear, they are initially shown as ground targets. This is mainly a tool to help debug traffic display on the various EFBs.
- GPS Attitude, which takes GPS turn rate, vertical speed, and groundspeed to synthesize what an AI should be doing. Think of it as a partial-panel made smart, or a "fake AHRS". This still needs flight testing. 
- Ownship Mode S / ADS-B pressure altitude and callsign detection (still needs flight testing)
- ForeFlight AHRS display, using the ForeFlight's simulator mode. This disables the Port 4000 GDL90 messages to trick ForeFlight that there's only a flight simulator attached.
- Routines for FLARM-formatted output to display ADS-B / TIS-B traffic on glider computers (in-work; for now just dumps to stratux.log.)

Ergonomicmike, 2016-03-09:
I've noticed, with the Traffic Markings off, that sometimes iFly will take a while to populate the Call Sign field. Don't know if that's an iFly problem or a Stratux problem. Given what Jeff has reported, I'm going to guess that iFly waits until the Call Sign has stabilized before it reports it.

Axtel4, 2016-03-09:
Thanks. I'll give it a try on a parallel image.

peepsnet, 2016-03-10:
@AvSquirrel Not that it matters but... is the distance calculation taking in to account altitude. Would a target directly over head and 10,000ft show as 2 miles? slant range?

AvSquirrel, 2016-03-10:
@peepsnet -- no. This is purely great circle distance. A target overhead would show as 0.0 miles.

cyoung, 2017-04-19:
#294.

closed 2017-04-19 by cyoung