cyoung/stratux #295: How is the .sh process supposed to work?

Axtel4, 2016-02-29:
Not really an issue, more of a question.

1.Stratux v0.7b3 
2.Stratux config: dual SDR, GPS yes (Vk-172), AHRS no, 2A wall wart for testing.
3.EFB app and version (NA). EFB platform (NA). EFB hardware (NA).
4.Description of your issue.

I have been trying update to the the latest version through the .sh method.  I finally gave up and rolled back to the v0.7b1 image and proceeded from there.  What I don't see is the signal strength to the Traffic page.  I previously pulled Avsquirrel's new_dump1090 branch and built it to have the signal data, but when I update the official image through the .sh method the signal strength is missing.  Does this mean the traffic signal strength isn't in the .sh updates and that I need to go back to the new_dump1090 branch to have them?  

My Stratux directory shows the following .sh files:
> -rw-r--r--  1 root root 6383266 Feb 15 18:06 update-v0.7b3-6762c01935.sh
> -rw-r--r--  1 root root 6384317 Feb 15 18:04 update-v0.7b3-857b71b716.sh
> -rw-r--r--  1 root root 6383177 Feb 28  2016 update-v0.7b3-f034df937d.sh

I assume they were installed correctly as the Startux home page doesn't show the commit version nor does git describe / git log report the updates applied through the .sh message.  Maybe there is a .sh file that picked up the traffic updates that I can't find.

Ergonomicmike, 2016-02-29:
I've got the 716.sh installed (the lastest, AFAIK.). I do not have signal strength on my Traffic Page.

skypuppy, 2016-02-29:
Please, can we "reset" this question from scratch?

How does one, using a prior downloaded stratux full SD card image, use 
the ".sh" to update to the very latest (alpha?) code, and have it 
compile properly on the Pi?  Step by step, if you don't mind.
So, here I am, sitting at the console of the Pi, so what do I do?

Thanks,
Skypuppy

peepsnet, 2016-02-29:
@Axtel4 ,
That would be my fault!
https://github.com/cyoung/stratux/pull/280

Change the name of the files to match the new format!!

update-stratux-v0.7b3-6762c01935.sh
update-stratux-v0.7b3-857b71b716.sh
update-stratux-v0.7b3-f034df937d.sh

@cyoung Can you change the name of the updates on the releases page??

Ergonomicmike, 2016-02-29:
@skypuppy I do it all from the web UI. (Not in the Pi as we've been doing with gits.)

First, download the latest .sh to your device from
http://updates.stratux.me/builds/

Start with virgin image of 0.7b1. (Although Chris said you don't have to start virgin. But I had to.) Once it's running, go to the Settings page, do the "Check to select System Update file" button, chose the latest .sh you just downloaded from the (presumably) Download directory on your device, click the button again to install, wait 60 seconds and you're set.

Axtel4, 2016-02-29:
@Ergonomicmike,
You didn't have to rename the files as posted above?

>Change the name of the files to match the new format!!
update-stratux-v0.7b3-6762c01935.sh
update-stratux-v0.7b3-857b71b716.sh
update-stratux-v0.7b3-f034df937d.sh

Also, by latest, you mean the file with the latest data and time stamp?  Does this imply you can jump in at the latest and pull in all the previous interim update?

Ergonomicmike, 2016-02-29:
I didn't change anything. And yes, I went by the latest time stamp. And also yes, Chris says that you can just do the last one. You do NOT have to update incrementally.

Axtel4, 2016-02-29:
Thanks, 

I tried it with both files names and no joy on the traffic signal strength.  I also cleared the browser cache and reloaded and still no updated traffic page.  At this point I have to assume the Avsquirrel's updates didn't make it into the .sh file.  I guess I'll go back and pull his branch and rebuild again.

skypuppy, 2016-02-29:
Ergo (no reference to the Stargate SG-1 Ergo,) didn't you say that you 
did not get the latest code with that method?

Thank you,
Skypuppy

peepsnet, 2016-02-29:
Since the change to the .sh naming was introduced in v0.7b3 then if you load any of the .sh from the github releases page without the new naming convention then it should fail. 

But if you are on v0.7b3 and try to load the .sh from http://updates.stratux.me/builds/ then I believe you should have to change the name

Axtel4, 2016-02-29:
Update:
It looks like some of the changes made it in to the .sh file, but the Web UI didn't get updated.  The ES log has what appears to be the latest data:

> 281102634840,{"Icao_addr":11344262,"DF":0,"CA":0,"TypeCode":0,"SubtypeCode":0,"SBS_MsgType":7,"SignalLevel":5,"Tail":null,"Squawk":null,"Emitter_category":null,"OnGround":false,"Lat":null,"Lng":null,"Position_valid":false,"NACp":null,"Alt":31400,"AltIsGNSS":false,"GnssDiffFromBaroAlt":null,"Vvel":null,"Speed_valid":false,"Speed":null,"Track":null,"Timestamp":"2016-02-29T00:43:24.801Z"}

Axtel4, 2016-02-29:
> Since the change to the .sh naming was introduced in v0.7b3 then if you load any of the .sh from the github releases page without the new naming convention then it should fail. 

How does one know the process fails?

peepsnet, 2016-02-29:
Now that I look closer in the commits. 

I believe the .sh naming convention change was a commit after v0.7b3 I cannot remember. So it should work for anything named update*.sh

Axtel4, 2016-02-29:
I just noticed the Tower page doesn't have the latest Signal updates either, i.e. max, average, etc.

peepsnet, 2016-02-29:
I have all the latest update/commits using git pull and there are no changes to the traffic page

cyoung, 2016-02-29:
>Startux home page doesn't show the commit version nor does git describe / git log report the updates applied through the .sh message

See /var/log/stratux.log. Will add something in the interface that shows build (commit) running.

cyoung, 2016-02-29:
@Axtel4 - the changes you're looking for are in a pending pull request. I'll review it now and merge if possible - then it will be in there.

Use the web interface for updates.

peepsnet, 2016-02-29:
where will the update be:
http://updates.stratux.me/builds/ OR with a git pull from ssh??

cyoung, 2016-02-29:
In the builds directory. Each time a new commit is made to the project, CircleCI kicks off to do a "check build". A packaging process runs after this, so within a new commit being made an update .sh file should be available within 10-15 minutes.

https://circleci.com/gh/cyoung/stratux

cyoung, 2016-02-29:
http://updates.stratux.me/builds/update-v0.7b3-a5a02ff4d5.sh just went up (latest). Yes, just get the one with the newest timestamp.

peepsnet, 2016-02-29:
if you used any of the other .sh updates would they of included the update file naming convention??

Axtel4, 2016-02-29:
@cyoung Chris, you've been busy.  Thanks.

Axtel4, 2016-02-29:
New issue, now my VK-172 GPS isn't been seen by Startux.  The green light blinks, but I am stuck off the coast of Africa.  I rebooted and reload the Web UI and no joy.  Shall I start a new issue thread?  I'll try the other Stratux with the RY-835.

cyoung, 2016-02-29:
@peepsnet is right. That's the reason your update didn't work and you had to roll back, @Ergonomicmike. I made a mistake in the file naming, they are uploaded now with the right names. Some error checking was ignoring the update file.

http://updates.stratux.me/builds/update-stratux-v0.7b3-a5a02ff4d5.sh is the latest now.

Axtel4, 2016-02-29:
The RY-835 isn't see by Stratux either.  I'll burn another image and work through the sh files again to see where is stops working.  If I still have issues, I'll start another thread.

cyoung, 2016-02-29:
As of 0d67cb855b:

[image]

Axtel4, 2016-02-29:
I am there now.  I don't know the issue with the GPS.  I back up to a clean image of v0.7b3 that I archived and built from there.  Thanks for adding the version build info.  If you want, you are more than welcome to close this issue.

[image]

skypuppy, 2016-02-29:
> versiondisplay

Um, the SD card image on Stratux.me is v.6r1.  I had .4r4 and while it's 
web ui had more stuff, there was nothing there for downloading any shell 
scripts.

Currently installing .6r1 onto the SD card and will try from there. <sigh>

Thanks,
Skypuppy

Ergonomicmike, 2016-02-29:
FWIW, the BU-353 is still working.

skypuppy, 2016-02-29:
Ok, how do I get from .6.1 to .7.3?  Just installed and expanded and 
running .6r1 and there are no shell script options.
Having multiple update methods, one with version control, one as a 
download full image, and one with multiple pull methods has thrown me 
into a place I don't know where I am.

Thanks,
Skypuppy

Axtel4, 2016-02-29:
Try installing the 0.7b1 zipped image and work up from there.  From there start with the 0.7b1 sh file working up through the updates at http://updates.stratux.me/builds/.

I think, but I didn't try it, that a person could jump directly to the most recent build from the stratux.me/builds repository from the 0.7b1 image.  0.7b1 was the last full image and included the sh update method.  The follow on beta releases were only released via source code and .sh files.

I am still unclear as to if one can jump in with the git method update and then switch back to the .sh method to update.

cyoung, 2016-02-29:
@Axtel4 - just use the .sh files as they include more changes than 'make install' will track. System level changes are not (and never were) managed by 'make install'. For example, the recent change that added the 'restart' function for WiFi would not have been made on your system if you only used 'make install'.

closed 2016-02-29 by cyoung

skypuppy, 2016-02-29:
There appear to be numerous changes/problems in the SD card image for .6r1.

1. dpcpd failed.  maybe that's what is behind all the dhcp problems?

2. i2c kernel not used and i2c not enabled.  ahrs failures?

3. using "raspi-config" to turn on graphics desktop fails, even after 
apt-getting 'lightdm'   It appears wheezy does not automatically load 
the X11 greeter.  Without graphics, we can't download the upgrade 
scripts to get to .7.3

4. after apt-getting i2c-tools, running a simple i2cdetect fails (even 
as sudo.)

5. " sudo apt-get updates" fails 11 packages!!! with virgin .6r1 SD card 
image.

6. and so on.  Many other problems cropping up from this image.

Is this just me or what?  If my SD card has gone bad...  (it is the one 
I used for all the .4 versions.

Skypuppy

skypuppy, 2016-03-01:
Ok, but how does one GET the .7b1 version???????????

cyoung, 2016-03-01:
https://github.com/cyoung/stratux/releases/download/v0.7b1/stratux-v0.7b1-02bf2251d8.img.zip

skypuppy, 2016-03-01:
Thanks, Chris.  As so many methods of update are being used, I lost all 
track of who did what to whom and how many times.

Skypuppy

skypuppy, 2016-03-01:
You guys won't believe this but I've made the first breakthrough in 
getting Qt to work between the Pi and the Linux PC.  Woohoo.

Have begun the I2C code for talking to and setting up the AHRS.

Now, only 737 1/2 steps to go.  (pun intended.)

cyoung, 2016-03-01:
The .sh updates now should be the primary means of updating once all users get up to v0.7b1+.

skypuppy, 2016-03-01:
Er, now is this 384 mb I just downloaded a full SD card image or is it 
just the stratux tools and executables?  Just wondering before I start 
on the SD card.

Axtel4, 2016-03-01:
It is the full compressed image.  Unzip the image file and burn it to the SD card.  On Windows I use Win32DiskImager.  The uncompressed file size will be just over 3 GB.

skypuppy, 2016-03-01:
Thanks.  With the name "image" in it...  Just making sure.  :)

skypuppy, 2016-03-01:
Ok, got the .7b1 installed on the SD card, expanded, and running. Two 
issues:

Using a Winblows laptop (because it is easy to move around!) and Firefox 
to view the web page and went to "settings."

Issue #1, the Adafruit Ultimate GPS, using standard NMEA sentences on 
the serial port, is still not showing up as communicating.  Yes, I 
enabled the switch on the web page.

Issue #2, hit the "update" box and browsing ONLY looked at files on the 
laptop.  Did I miss something here?  I understood it would link to the 
stratux.me website and show the current shell update files, so we could 
choose which version to download.  Then those files would be downloaded 
to the RPi2  without laptop intervention and automatically run to update 
the RPi2 stratux software.  What am I missing?  How do I actually update 
to .7.3, please?

Yes, I read back through the posts in here and don't see where I went 
wrong.  :(

Thanks,
Skypuppy
(did not want to post this on the reddit site.)

skypuppy, 2016-03-01:
Truly, I'm not normally this dense, but the dog ate my homework, I'm 
sick with the flu, a family member is sick, I got a flat tire, and I 
didn't get the memo on time.

Skypuppy

Axtel4, 2016-03-01:
Here is what I did:

From my browser on my desktop I navigated to:

http://updates.stratux.me/builds/

I right clicked and downloaded the most recent .sh file to my desktop.  I then accessed the Web IU on Stratux and used the Update button to navigate to the sh file and clicked Open.  The button changed to something like Install Updates.  I clicked in the button.  After a few minutes I logged back into Stratux, cleared the browser cache and refreshed the browser.  I then had the latest Web UI and Stratux was updated.

skypuppy, 2016-03-01:
Worked like magic!  Thanks a lot, Axtel4.

Didn't the source code used to be located underneath the /home/pi 
directory?  Where is it now?

Skypuppy

Axtel4, 2016-03-01:
It is euther in /stratux or /root/stratux

I had to find it again by using:

find / -name gen_gdl90.go