cyoung/stratux #299: WebUI uptime does not display days.

cyoung, 2016-03-02:
1. Stratux version: v0.7b3 / d714a5139e
2. Stratux config:

    SDR
      [X] single
      [ ] dual

    GPS
      [X] yes
      [ ] no
      type: Vk-172

    AHRS
      [ ] yes
      [X] no

    power source: Vilros power adapter

    usb cable: Y-cable

3. EFB app and version: FF v7.5.2

    EFB platform: iOS 9.2.1

    EFB hardware: iPad Mini 2 16GB

4. Description of your issue:

WebUI uptime does not display days.

bkwny, 2016-03-02:
I suggest ddd/hh:mm:ss for the display format.  For example, 0/11:21:45, 1/05:26:15, 13/23:55:16, or the like.

bkwny, 2016-03-03:
For /web/plates/status.js, I think:

if (uptime != undefined) {
 var up_d = parseInt((uptime/1000) % 86400),
     up_h = parseInt(((uptime/1000) - 86400*up_d) % 3600),
     up_m = parseInt(((uptime/1000) - 86400*up_d - 3600*up_h) % 60),
     up_s = parseInt(((uptime/1000) - 86400*up_d - 3600*up_h - 60*up_m);
 $scope.Uptime = String(up_d + "/" + ((up_h < 10) ? "0" + up_h : up_h) + ":" + ((up_m < 10) ? "0" + up_m : up_m) + ":" + ((up_s < 10) ? "0" + up_s : up_s));

bkwny, 2016-03-03:
For /web/plates/status.js, I think:
[status-uptime.txt](https://github.com/cyoung/stratux/files/157713/status-uptime.txt)

cyoung, 2016-03-05:
Fixed by PR #306 - thanks to @bkwny.

closed 2016-03-05 by cyoung