cyoung/stratux #304: an interesting snipet from the new pi release article .. power issues ...

duecedriver, 2016-03-03:
"The camera-shy Raspberry Pi 2 suffered problems with the switched mode power supply (SMPS) chip on the board, which turned out to be photosensitive. Too much light, such as from flash photography, caused fluctuation in the power supply and made the board reset. "

possibly part of peoples power woes if used on the glarshield in a translucent case... 

only visible light or can it be affected by "other" radiated energy.. ie leaky sidelobe or transmitter power supplies.. etc..

thoughts..

kdknigga, 2016-03-03:
https://www.raspberrypi.org/blog/xenon-death-flash-a-free-physics-lesson/

TL;DR: "the effect only happens under VERY specific circumstances. Flashes
of high-intensity, long-wave light"

cyoung, 2016-03-03:
That's a good catch. Anyone have power issues that may be light exposure related?

AvSquirrel, 2016-03-03:
To reempahsize @kdknigga  --
>I’ve said it above, but it bears repeating, because I’ve seen some of you mention this in the forums and in comments sections elsewhere: **common everyday light sources – e.g. bright sunlight, indoor lighting, angry cyclists* – don’t cause this to happen, so please don’t worry**

peepsnet, 2016-03-03:
My few flashes of brilance while working on my ppl have not seemed to of affected the stratux at all. I guess its not the correct wavelength?!

Axtel4, 2016-03-03:
Cross post from Reddit:

Mine has been running for hours sitting on a glare shield, in a translucent case, and with a DME and Mode S transponder running with no ill effects. Oh, and cruising altitude at 8k to 10k ft.

I have also run it for hours sitting on the dashboard of a vehicle with no ill effects.

closed 2016-03-05 by cyoung