cyoung/stratux #305: [Request] GPX output... Whynot... were already logging GPS data...

peepsnet, 2016-03-03:
1. Stratux version: v0.7b3-83eda624ce.sh 
2. Stratux config: 

    SDR
      [ ] single
      [ X ] dual

    GPS
      [ X ] yes
      [ ] no
      type:

    AHRS
      [ X ] yes
      [ ] no

    power source:
AC 2.1A Samsung Charger
DC 20600 Anker

    usb cable:
Portapower 20awg charge/sync (from 6" to 10')

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
iFLY GPS
Avare
AvNav
Fltplan GO

    EFB platform: (e.g., iOS 9.2)
Android
    EFB hardware: (e.g., iPad Mini 2)
Galaxy Tab A T357T

4. Description of your issue:

This is more of a feature request. 

Since we are logging the GPS Data anyways. Could we output GPX data to a separate folder also???
I found a website called CouldAhoy.com and its fascinating!!

I logged 2 flights already and the tools for debriefing your flights are unheard of!!  I know the EFB's do logging but since we're logging the GPS data anyways. why not just have another switch in the settings to turn on GPX format output. Then have it log to a separate folder so you don't have to search through all the mess in the log folder.

iFLY GPS(my main efb does not output to a readable format for the website. I can, however, import the log file into http://www.gpsvisualizer.com/ and get a gpx file that works.  I have spoke with IFLY GPS on their support forum and they tell me that other formats are in the works, 

BUT

I was looking for the feature in stratux so a flight log is not killed by say... A BLOD or a tablet battery dying, or the EFB crashing!!

Check out the website here: http://www.cloudahoy.com
Here are my flights.
Local T/G work
http://www.cloudahoy.com/debrief/?key=7uGTlrUk8n4M

X-Ctry and Brunch!
(Leg 1)http://www.cloudahoy.com/debrief/?key=q2O4EudckU0Ob
(Leg 2)http://www.cloudahoy.com/debrief/?key=YBml4bGKUjxell5x2SM

Ergonomicmike, 2016-03-03:
Is this the same request as https://github.com/cyoung/stratux/issues/234?

peepsnet, 2016-03-03:
Well. Kinda they accept gpx also. There might be some pre written code for GPX

Ergonomicmike, 2016-03-03:
There's an answer here.

http://iflygps.com/SUPPORT/Forum/tabid/108/forumid/14/threadid/20854/scope/posts/Default.aspx

cyoung, 2016-03-05:
Duplicate - we'll likely consolidate replay logs into a more manageable format and then create an export tool. @peepsnet - track progress in #234.

closed 2016-03-05 by cyoung