cyoung/stratux #307: Support for RPI3 and it's built in wifi.

jthawkes, 2016-03-06:
1. Stratux version:
2. Stratux config:

    SDR
      [ ] single
      [ ] dual

    GPS
      [ ] yes
      [ ] no
      type:

    AHRS
      [ ] yes
      [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
I am unable to boot from the provided 0.6 image SD card or build 0.7 on Jessie lite. 

pi@test-rpi3:~ $ sudo su -
root@test-rpi3:~# cd stratux
root@test-rpi3:~/stratux# make
make xdump978
make[1]: Entering directory '/root/stratux'
cd dump978 && make lib
make[2]: Entering directory '/root/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o
make[2]: Leaving directory '/root/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
make[1]: Leaving directory '/root/stratux'
make xdump1090
make[1]: Entering directory '/root/stratux'
git submodule update --init
cd dump1090 && make
make[2]: Entering directory '/root/stratux/dump1090'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/stratux/dump1090'
make[1]: Leaving directory '/root/stratux'
make xlinux-mpu9150
make[1]: Entering directory '/root/stratux'
git submodule update --init
cd linux-mpu9150 && make -f Makefile-native-shared
make[2]: Entering directory '/root/stratux/linux-mpu9150'
gcc libimu.o inv_mpu.o inv_mpu_dmp_motion_driver.o linux_glue.o mpu9150.o quaternion.o vector3d.o -shared -o libimu.so
sudo cp -f ./libimu.so /usr/lib/libimu.so
make[2]: Leaving directory '/root/stratux/linux-mpu9150'
make[1]: Leaving directory '/root/stratux'
make xgen_gdl90
make[1]: Entering directory '/root/stratux'
go get -t -d -v ./main ./test ./linux-mpu9150/mpu ./godump978 ./mpu6050 ./uatparse
go build -ldflags "-X main.stratuxVersion=`git describe --tags --abbrev=0` -X main.stratuxBuild=`git log -n 1 --pretty=%H`" main/gen_gdl90.go main/traffic.go main/ry835ai.go main/network.go main/managementinterface.go main/sdr.go main/uibroadcast.go main/monotonic.go
# github.com/jpoirier/gortlsdr
38: error: 'rtlsdr_set_tuner_bandwidth' undeclared (first use in this function)
Makefile:16: recipe for target 'xgen_gdl90' failed
make[1]: *** [xgen_gdl90] Error 2
make[1]: Leaving directory '/root/stratux'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2
root@test-rpi3:~/stratux# make

Ergonomicmike, 2016-03-06:
Some clues [here](https://www.reddit.com/r/stratux/comments/48eg03/strongly_recommend_we_all_move_to_jessie_instead/) about how to make it work. (For jessie, anyway. Pi 3 is a new unknown.)

cyoung, 2016-03-06:
https://github.com/jpoirier/gortlsdr/issues/14

cyoung, 2016-03-06:
Where'd you get a Pi3?

jthawkes, 2016-03-06:
I ordered two from https://www.element14.com/community/community/raspberry-pi
and one from http://www.mcmelectronics.com/product/83-17300 (they limited me to one)
on Monday.
The MCM one was delivered on Friday. Still waiting for the element14 models.

skypuppy, 2016-03-06:
Yeah.  :)  I can't find one anywhere.

jthawkes, 2016-03-06:
oh yeah, element 14 are out of stock and MCM has a 21 day wait.
One overpriced amazon supplier says  In stock on March 8, 2016.

skypuppy, 2016-03-06:
way overpriced.

jpoirier, 2016-03-06:
Note that we're (the stratux image builder) using a slightly modified fork of librtlsdr from here:

https://github.com/jpoirier/librtlsdr

And on a side note, an attempt to get a newer librtlsdr version released:

http://lists.osmocom.org/pipermail/osmocom-sdr/2016-February/001373.html
http://lists.osmocom.org/pipermail/osmocom-sdr/2016-March/001376.html

jthawkes, 2016-03-06:
I have spent a few hours trying all I can to get it to work.
1. installed image on RPI2 and try to upgrade it to jessie(no luck).
2. build on RPI3 with jessie, no luck.  tried a few of the suggestions. no luck.
Happy to test a little when you have some ideas.
:-)

jpoirier, 2016-03-06:
Are you getting the same error you listed in your initial post or a different one?

jpoirier, 2016-03-06:
On Sun, Mar 6, 2016 at 3:37 PM, jthawkes <notifications@github.com> wrote:

> pretty much the same error.
> I will have to start again with a clean Jessie RPI3 image.
>

Assuming you have all the necessary dependencies (eg git, golang, etc)
installed, the following steps should build gortlsdr successfully, which
you can do independent of stratux. Once you get it built then you can try
building stratux.

apt-get install libusb-1.0-0.dev

git clone https://github.com/jpoirier/librtlsdr.git
cd libstlsdr
mkdir build
cd build
cmake ../
make
sudo make install
sudo ldconfig

go get -u github.com/jpoirier/gortlsdr

jthawkes, 2016-03-06:
could you list the dependencies for me? I can install them, just need a list.

jpoirier, 2016-03-06:
On Sun, Mar 6, 2016 at 3:51 PM, jthawkes <notifications@github.com> wrote:

> could you list the dependencies for me? I can install them, just need a
> list.
>
How did you install the deps prior cd'ing to stratux and running make?

Note that it's not going to work if all you're doing is trying to build
stratux after cloning the repo on to a stock OS install; there are a bunch
of things that need to be setup/configured properly in order to create an
image that will work properly. You can see the details in the following
file stratux/image/spindle/wheezy-stage4

jthawkes, 2016-03-06:
How do I run that? I have copied the text into a file and marked it as executable but it won't run.
(not a build expert just a very long time lunix user. 20+ years).

jpoirier, 2016-03-06:
On Sun, Mar 6, 2016 at 4:14 PM, jthawkes <notifications@github.com> wrote:

> How do I run that? I have copied the text into a file and marked it as
> executable but it won't run.
> (not a build expert just a very long time lunix user. 20+ years).
>
You don't; it's a spindle file. Spindle is the tool that's used to build a
working stratux OS image that runs on an RPi2, where the OS image is setup
specifically for stratux and includes all the necessary binaries.

jthawkes, 2016-03-06:
I added a bunch of packages
sudo apt-get install -y screen  hostapd isc-dhcp-server tcpdump mercurial  autoconf fftw3 fftw3-dev  libtool git cmake libusb-1.0-0.dev build-essential autoconf fftw3 fftw3-dev  libtool git make

I now get this error. It looks simple to fix.

sudo cp -f ./libimu.so /usr/lib/libimu.so
make[2]: Leaving directory '/root/stratux/linux-mpu9150'
make[1]: Leaving directory '/root/stratux'
make xgen_gdl90
make[1]: Entering directory '/root/stratux'
go get -t -d -v ./main ./test ./linux-mpu9150/mpu ./godump978 ./mpu6050 ./uatparse
import "fmt": import path doesn't contain a slash
package fmt: unrecognized import path "fmt"
import "reflect": import path doesn't contain a slash
package reflect: unrecognized import path "reflect"
import "runtime": import path doesn't contain a slash
package runtime: unrecognized import path "runtime"
import "runtime/cgo": import path doesn't contain a hostname
package runtime/cgo: unrecognized import path "runtime/cgo"
import "strconv": import path doesn't contain a slash
package strconv: unrecognized import path "strconv"
import "syscall": import path doesn't contain a slash
package syscall: unrecognized import path "syscall"
import "unsafe": import path doesn't contain a slash
package unsafe: unrecognized import path "unsafe"
import "errors": import path doesn't contain a slash
package errors: unrecognized import path "errors"
import "log": import path doesn't contain a slash
package log: unrecognized import path "log"
import "math": import path doesn't contain a slash
package math: unrecognized import path "math"
import "time": import path doesn't contain a slash
package time: unrecognized import path "time"
import "encoding/hex": import path doesn't contain a hostname
package encoding/hex: unrecognized import path "encoding/hex"
import "io/ioutil": import path doesn't contain a hostname
package io/ioutil: unrecognized import path "io/ioutil"
import "strings": import path doesn't contain a slash
package strings: unrecognized import path "strings"
import "bufio": import path doesn't contain a slash
package bufio: unrecognized import path "bufio"
import "bytes": import path doesn't contain a slash
package bytes: unrecognized import path "bytes"
import "compress/gzip": import path doesn't contain a hostname
package compress/gzip: unrecognized import path "compress/gzip"
import "crypto/rand": import path doesn't contain a hostname
package crypto/rand: unrecognized import path "crypto/rand"
import "crypto/sha1": import path doesn't contain a hostname
package crypto/sha1: unrecognized import path "crypto/sha1"
import "crypto/tls": import path doesn't contain a hostname
package crypto/tls: unrecognized import path "crypto/tls"
import "encoding/base64": import path doesn't contain a hostname
package encoding/base64: unrecognized import path "encoding/base64"
import "encoding/binary": import path doesn't contain a hostname
package encoding/binary: unrecognized import path "encoding/binary"
import "encoding/json": import path doesn't contain a hostname
package encoding/json: unrecognized import path "encoding/json"
import "flag": import path doesn't contain a slash
package flag: unrecognized import path "flag"
import "math/big": import path doesn't contain a hostname
package math/big: unrecognized import path "math/big"
import "regexp": import path doesn't contain a slash
package regexp: unrecognized import path "regexp"
import "sort": import path doesn't contain a slash
package sort: unrecognized import path "sort"
import "unicode": import path doesn't contain a slash
package unicode: unrecognized import path "unicode"
import "io": import path doesn't contain a slash
package io: unrecognized import path "io"
import "os": import path doesn't contain a slash
[... 34 lines cut ...]
package crypto/x509: unrecognized import path "crypto/x509"
import "database/sql": import path doesn't contain a hostname
package database/sql: unrecognized import path "database/sql"
import "database/sql/driver": import path doesn't contain a hostname
package database/sql/driver: unrecognized import path "database/sql/driver"
import "encoding/csv": import path doesn't contain a hostname
package encoding/csv: unrecognized import path "encoding/csv"
import "encoding/xml": import path doesn't contain a hostname
package encoding/xml: unrecognized import path "encoding/xml"
import "image/draw": import path doesn't contain a hostname
package image/draw: unrecognized import path "image/draw"
import "math/cmplx": import path doesn't contain a hostname
package math/cmplx: unrecognized import path "math/cmplx"
import "go/build": import path doesn't contain a hostname
package go/build: unrecognized import path "go/build"
import "image/png": import path doesn't contain a hostname
package image/png: unrecognized import path "image/png"
import "image/jpeg": import path doesn't contain a hostname
package image/jpeg: unrecognized import path "image/jpeg"
Makefile:16: recipe for target 'xgen_gdl90' failed
make[1]: *** [xgen_gdl90] Error 1
make[1]: Leaving directory '/root/stratux'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2
root@stratux-test1:~/stratux#

skypuppy, 2016-03-06:
Why in the <censored> did you guys port librtlsdr to golang????  It 
worked fine in C, whether faster or not, I don't know, but considering 
there is such a small pool of programmers who work in go rather than C...
Very odd administrative decision.

Skypuppy
Not condemning, just sitting here scratching my head or something.

jpoirier, 2016-03-06:
> Why in the <censored> did you guys port librtlsdr to golang???? It worked fine in C, whether faster or not, I don't know, but considering there is such a small pool of programmers who work in go rather than C... Very odd administrative decision. 

Huh? 

gortlsdr isn't a port, it's a wrapper around librtlsdr (via go's ffi) and it allows one to drive librtlsdr natively from within go code, which is convenient since the core part of stratux is written in go. 

And, if one can (competently) develop in C then developing in Go is second nature.

skypuppy, 2016-03-19:
A wrapper around a library?  Okey dokey.
Competent in C?  Well, I couldn't testify in court that I am, but I've been paid to do just that since 1980.

cyoung, 2016-03-21:
Guys requesting RPi3 support - could you test this image?

https://www.reddit.com/r/stratux/comments/4aoocl/need_beta_testers_for_new_image/

The few that have tested it are showing good results.

I'm a little concerned about the Wi-Fi controller especially in different build configurations.

jthawkes, 2016-03-21:
It runs for me.
I will test it in the office on Wednesday.

I am too low to the ground to get towers at home.

1090 is receiving AC. Not many UAT978 AC in the area(my Piaware setup detects about one per day and it has a full 8db antenna)
GPS dongle works. I will wire in the GPS/AHRS in the next week or so.

cyoung, 2016-03-21:
I've removed the beta image, try v0.8r2 when you can (on RPi3), please.

Thanks!

JohnOCFII, 2016-03-27:
Not sure if you are still looking for RPi3 feedback, but it bench tests fine for me with 2 NooElec SDR radios and the RY835AI (with AHRS via GPIO, and also with the GPS via USB - my GPS isn't working via GPIO, but that is unrelated to the Pi).

ytwytw, 2016-04-18:
hi @cyoung 

Can you provide instruction or guide how to setup hotspot on Pi 3?

I have successfully compiled code from source and its running well on web

I have tried several methods on google to start hotspot, but not working for pi 3

Thanks

peepsnet, 2016-04-23:
Could this be closed??

cyoung, 2016-04-23:
Oh yeah.

@ytwytw - http://stratux.me, install v0.8r2. Works out of the box on the Pi 3.

closed 2016-04-23 by cyoung