cyoung/stratux #309: Hardware measurements on an  HMC5883L Digital Compass in a LEADIY-D3

Ergonomicmike, 2016-03-07:
1. Stratux version: 0.vb7+alpha to 18de.sh build
2. Stratux config:

    SDR
      [ ] single
      [X] dual

    GPS
      [X ] yes
      [ ] no
      type: BU-353

    AHRS
      [X] yes
      [ ] no

    power source:
Anker E5
    usb cable:
Anker cable
3. EFB app and version: (e.g., WingX Pro7 8.6.2)
N/A
    EFB platform: (e.g., iOS 9.2)
N/A
    EFB hardware: (e.g., iPad Mini 2)
N/A
4. Description of your issue:

First, after seeing a Stratus V2 play, I don't think that getting the digital compass to work on the Stratux is an issue. (No warning from Stratus to place the unit away from metal or to swing the compass before using.) Still, it bugs me because the digital compass is suppose to work, whether Stratux uses it or not.

So, I bought a LEADIY-D3, per https://www.reddit.com/r/stratux/comments/479wy0/ry835ai_ahrs_alternative_for_sale_2499/

(Note now that the website leadiy.com (all in Chinese) shows a -D3A now.)

It says it has the same chipset as the RY835ai.

It has (ostensibly) a Honeywell  HMC5883L on the board. (Ostensibly because it says L883 on the IC.)

Although I get a "Heading" indication in the Stratux UI AHRS page that tracks, that appears to be a (drifting) gyro output, that starts at 360 degrees when the Stratux first powers up, regardless of the orientation of the Stratux.

I got out my old 'scope. Now, my eyes are old, my hands are shaky, and these parts have gotten a lot smaller. So someone should confirm my findings.

I found that pin 15, the Data Ready pin, stays high all the time. That is, it's not going low for 5 uSec when [compass] data is placed in the registers. 
Data sheet from https://www.seeedstudio.com/wiki/images/4/42/HMC5883.pdf

Pin 1, SCL is not toggling. It stays high all the time. (The header pin on the board for SCL does pulse low.)
Pin 16, SDA is also high all the time.

Pin 2 and Pin 4 are connected by the PCB. The data sheet shows that Pin 4 is supposed to be No Connection.

Pin 14 is supposed to tie to Pin 2 in a single supply configuration. If it does, I can't see a pad. Looking at an enlargement of the same board without the L883 confirms that there is no connection on pin 14.
http://www.amazon.co.uk/LEADIY-D3C-MPU6050-Angular-Sensor-Acceleration/dp/B00WG9HUTM?tag=duc08-21

Some folks in the arduino forum have been fighting inconsistencies with data sheets and code.
http://forum.arduino.cc/index.php?topic=55159.0

They also say that the chip is susceptible to failure from reflow soldering.

If there are actual misconnections at the hardware level (on the PCB) , then it might be that no amount of code is going to make the chip with this board.

I'll leave it to Chris to close this "issue" out.

Ergonomicmike, 2016-03-07:
So now I'm wondering aloud if the Stratux waits for the Compass to self test and then if it switches the Compass from the default single to continuous mode?
***
OPERATIONAL EXAMPLES
The HMC5883L has a fairly quick stabilization time from no voltage to stable and ready for data retrieval. The nominal 56 milli-seconds with the factory default single measurement mode means that the six bytes of magnetic data registers (DXRA, DXRB, DZRA, DZRB, DYRA, and DYRB) are filled with a valid first measurement. 
To change the measurement mode to continuous measurement mode, after the power-up time send the three bytes:
0x3C 0x02 0x00
This writes the 00 into the second register or mode register to switch from single to continuous measurement mode setting. With the data rate at the factory default of 15Hz updates, a 67 milli-second typical delay should be allowed by the I 2 C master before querying the HMC5883L data registers for new measurements. To clock out the new data, send:
0x3D, and clock out DXRA, DXRB, DZRA, DZRB, DYRA, and DYRB located in registers 3 through 8. The HMC5883L will automatically re-point back to register 3 for the next 0x3D query. All six data registers must be read properly before new data can be placed in any of these data registers.

Below is an example of a (power-on) initialization process for “continuous-measurement mode”:
1. Write CRA (00) – send 0x3C 0x00 0x70 (8-average, 15 Hz default, normal measurement)
2. Write CRB (01) – send 0x3C 0x01 0xA0 (Gain=5, or any other desired gain)
3. Write Mode (02) – send 0x3C 0x02 0x00 (Continuous-measurement mode)
4. Wait 6 ms or monitor status register or DRDY hardware interrupt pin
5. Loop
Send 0x3D 0x06 (Read all 6 bytes. If gain is changed then this data set is using previous gain)
Convert three 16-bit 2’s compliment hex values to decimal values and assign to X, Z, Y, respectively.
Send 0x3C 0x03 (point to first data register 03)
Wait about 67 ms (if 15 Hz rate) or monitor status register or DRDY hardware interrupt pin
End_loop

Ergonomicmike, 2016-03-07:
BTW, in additional to the fact that the compass doesn't detect North on start up, it does not deviate when I hold a strong magnet near the chip.

Ergonomicmike, 2016-03-07:
So, since the SCL input is always high to the Compass, I have to assume that it's wired to the MPU-6050's Auxiliary i2c bus. 

I also assume that the Compass has to be switched to the Continuous Measurement Mode thru the MPU-6050's Aux Bus. (A one time event after 56 milliseconds after start up.)

On page 26 of the [MPU-6050 datasheet](https://www.cdiweb.com/datasheets/invensense/MPU-6050_DataSheet_V3%204.pdf) they discuss using their Aux i2c bus for Magnetometers. They say this is good for sensor fusion in their on-chip DMP. (But since metal in the aircraft could be a problem, the Stratux might not want that fusion.)
***
7.11 Auxiliary I 2 C Serial Interface
The MPU-60X0 has an auxiliary I 2 C bus for communicating to an off-chip 3-Axis digital output magnetometer or other sensors. This bus has two operating modes:
-  I 2 C Master Mode: The MPU-60X0 acts as a master to any external sensors connected to the
auxiliary I 2 C bus
-  Pass-Through Mode: The MPU-60X0 directly connects the primary and auxiliary I 2 C buses together,
allowing the system processor to directly communicate with any external sensors.

Auxiliary I 2 C Bus Modes of Operation:
- I 2 C Master Mode: Allows the MPU-60X0 to directly access the data registers of external digital
sensors, such as a magnetometer. In this mode, the MPU-60X0 directly obtains data from auxiliary
sensors, allowing the on-chip DMP to generate sensor fusion data without intervention from the
system applications processor.
For example, In I 2 C Master mode, the MPU-60X0 can be configured to perform burst reads,
returning the following data from a magnetometer:
-  X magnetometer data (2 bytes)
-  Y magnetometer data (2 bytes)
-  Z magnetometer data (2 bytes)
The I 2 C Master can be configured to read up to 24 bytes from up to 4 auxiliary sensors. A fifth sensor
can be configured to work single byte read/write mode.

- Pass-Through Mode: Allows an external system processor to act as master and directly
communicate to the external sensors connected to the auxiliary I 2 C bus pins (AUX_DA and
AUX_CL). In this mode, the auxiliary I 2 C bus control logic (3 rd party sensor interface block) of the
MPU-60X0 is disabled, and the auxiliary I 2 C pins AUX_DA and AUX_CL (Pins 6 and 7) are
connected to the main I 2 C bus (Pins 23 and 24) through analog switches.
Pass-Through Mode is useful for configuring the external sensors, or for keeping the MPU-60X0 in a
low-power mode when only the external sensors are used.
In Pass-Through Mode the system processor can still access MPU-60X0 data through the I 2 C
interface
***
Pass Through mode sounds more like what Chris wants. But I can't tell if you HAVE to use analog switches (on the PCB) for AUX-DA and AUX-CL. 

Even if you don't need analog switches, it appears from the traces on the LEADIY-D3 that it is wired for the Auxiliary Mode of the MPU-6050. (That is, the SCL pin from the Compass comes from the AUX-CL pin on the 6050.) So it seems that it would take a different PCB layout for Pass-Through Mode. 

So if Stratux had to have Compass, and if there are no production boards wired for Pass-Through, then someone might have to come up with a break out board specifically for the Stratux.

I'm ready to close this now.

Ergonomicmike, 2016-03-07:
Boy, they don't make this easy.

Here's the separate Register Map document for the MPU-650 that tells how to send initialization data to the Compass via the Auxillary SCL & SDA. http://43zrtwysvxb2gf29r5o0athu.wpengine.netdna-cdn.com/wp-content/uploads/2015/02/MPU-6500-Register-Map2.pdf

Ergonomicmike, 2016-03-07:
It seems to me that someone must done all this already. Here's one with compass integration.
https://github.com/wesleymilan/Dual-MPU9150-DMP-Intel-Galileo-and-Edison-CPP

Ergonomicmike, 2016-03-14:
Also, FWIW, some code here:

http://elinux.org/Sparkfun:_HMC5883L_Magnetometer

skypuppy, 2016-03-14:
Wow, another serendipitous coincidence.  My first embedded system was a 
BeagleBone Black!  I got 4 of them for different projects while learning 
this stuff.

This is relevant to the Pi because of one fantastic resource I found 
that is applicable to both worlds, a book called "Exploring Beaglebone," 
by Prof. Derek Molloy.  It's available from Amazon and elsewhere.  I've 
used it many, many times over the past year+ and it is extremely 
helpful.  It's about 550 pages of pure gold.  I have needed/read 
thousands of tech books over my years and this one tops them all and by 
a huge margin.  The first 200 pages or so are about the basics of 
Linux.  There is also a section, with examples, of basic electronics.  
Then the fun starts.  He give the theory, then the practical, a whole 
lot of practical, about using these little beasts.  While it focuses on 
the BeagleBone Black, with just a few address and pin changes, it 
applies almost 100% to the RPi2/3!  I can't say enough good about the 
information in this work of Dr. Molloy's.  I have used my first one so 
many times that the pages are falling out!!!  (Amazon, amazingly, 
replaced it at no charge, when it should have been the publishers 
responsibility, Wiley.)
It talks about interfacing with I2C, SPI, serial, using Eclipse and Qt, 
making the RPi a web server, interfacing to it via ssh with or without 
X, and so much more -- ALL with 'C/C++' code examples and a little bit 
of Java and Python now and then.  Further, all code in the book is 
backed up and kept current and corrected on the accompanying website.  
He even answers simple and complex tech questions on that eb site!!!!!  
I'm telling you, this guy is tireless.  He makes 45 minute web videos 
about many topics having to do with computers and sensors, and so on.
Remember, though, that his code is aimed at teaching an audience, not 
optimizing the source code.

The companion website is here: www.exploringbeaglebone.com, segregated 
by chapter with chapter topic very evident.

Skypuppy

Ergonomicmike, 2016-04-19:
I noticed that LithuanianAmerican no longer sells the LEADIY-D3. Last he emailed me, he was trying to talk to the Compass, without success. I noticed in the reddit that ppl are moving on to another chipset. So am closing this.

closed 2016-05-27 by cyoung