cyoung/stratux #310: Update process takes ~3 minutes to respond

ssokol, 2016-03-07:
1. Stratux version: v0.7b3-83eda624ce
2. Stratux config:

    SDR
      [ ] single
      [X] dual

    GPS
      [X] yes
      [ ] no
      type: VK-172

    AHRS
      [ ] yes
      [X] no

    power source: Anker PowerPort 4

    usb cable: Anker 1'

3. EFB app and version: N/A

    EFB platform: N/A

    EFB hardware: N/A

4. Description of your issue:

Updating from v0.7b3-83eda624ce. Attempted to install update-stratux-v0.7b3-4bf26079d1.sh which behaved differently from previous updates. Usually the upload and update process takes ~5 seconds or less. When attempting this update, did not receive any feedback from the web application after clicking the "Install update-stratux-xx" button. It showed the blue glow around the button, but it did not pop up the dialog indicating success. After several minutes a dialog appeared that simply said "error".

I went back and installed the next sequential update, thinking that perhaps you couldn't skip directly to the latest. When I did, I had a similar experience except that after ~3 minutes a dialog box appeared telling me the update was successful.

Anyone else seeing this?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2016-03-08:
Open in Chrome, right click, "Inspect".

Maybe related:

```
XMLHttpRequest cannot load http://192.168.10.1:8080/updateUpload. No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://192.168.10.1' is therefore not allowed access.
```

See what errors show in the console.

https://github.com/cyoung/stratux/commit/d1d00d5bfbadc9c728b76d1a96a6a02a6c2921db

Ergonomicmike, 2016-03-08:
If not a Chrome problem, a failing SD card?

Ergonomicmike, 2016-05-16:
I had similar behavior the other day. I was in an FBO and had the Stratux outside to receive ADS-B. Poor Stratux WiFi signal strength inside. The first time I tapped the "Install" button from inside the FBO, I got an error. The Second time I tapped it, it took longer than usual to get the acknowledgment to reboot. Usually I do .sh updates within feet of the Stratux. Am chalking it up to a poor WiFi connection.

cyoung, 2016-12-01:
https://github.com/cyoung/stratux/commit/0177402f00487c7cde186545fdabc5a7141f81f8

Also perhaps was a silent fail from `managementinterface.go` on file write, which was fixed.

closed 2016-12-01 by cyoung