cyoung/stratux #311: Seeing double targets (first time ever)

Ergonomicmike, 2016-03-08:
1. Stratux version: v0.7b3 + 18de alpha
2. Stratux config:

    SDR
      [ ] single
      [X] dual

    GPS
      [X] yes
      [ ] no
      type: BU-353

    AHRS
      [X] yes
      [ ] no

    power source:
Anker E5
    usb cable:
Anker cable
3. EFB app and version: (e.g., WingX Pro7 8.6.2)
iFly v9.4.65
    EFB platform: (e.g., iOS 9.2)
Android KK
    EFB hardware: (e.g., iPad Mini 2)
Asus ZenPad S 8
4. Description of your issue:

Coming back home to Phx. Am south of KPHX. A fair amount of traffic. Saw at least five targets painted twice. (Either that or an awful lot of formation flying going on today.)

[image]

Didn't see this on the way down thru the same airspace. Never seen it before. (First time.)

Could be an iFly problem. But I'm first guessing a stratux problem, given the recent changes. 

BTW a LOT of zeros in the uat log.

es, uat and dump logs here:
https://www.sendspace.com/file/f4ilei

cyoung, 2016-03-08:
Did you get a chance to check the web interface to see if the duplicates were showing there?

Ergonomicmike, 2016-03-08:
I had the chance, but I'm not that smart. Is there a web log I can send? 

If there's a next time, I'll be sure to check. (Although unless the traffic there presents in pairs, it won't be easy to see a pattern in flight. Will take a screen shot if there's a next time.)

cyoung, 2016-03-08:
I'll try replaying with your logs. You don't experience this problem on the ground (1090 only)?

Ergonomicmike, 2016-03-08:
I've never seen this issue before, in flight or on the ground. I flew the same route south a few hours beforehand. (But I didn't have the logging on for that. Sorry. Was trying to isolate BLOD issues. ((BTW, I lost the Stratux three times on two iFly's while practicing ILS's. But the Blue Light stayed flickering and the web UI was responsive. I just went from iFly .65 to .68. I'll squawk it over at iFly.)

Nokomis449, 2016-03-08:
I flew about 8 hours this weekend with a newly-flashed Stratux image, updated with the latest .sh (as of about last Wednesday) and had no such issues.  Actually, no issues at all with the Stratux.  There were occasional traffic differences between the Stratux and the SkyRadar DX (displayed on 2 seperate devices, Stratux on an Android tablet and SkyRadar on an iFly 740).  Both devices running the latest beta's of iFly.  Traffic differences were minimal enough thay could probably be explained by the different antenna locations.  The Stratux had stock SDR antennas suctiond to the pilot's side window, the SkyRadar elevated in the baggage compartment of the Mooney.  Screenshot of Stratux traffic on iFly:

[image]

AvSquirrel, 2016-03-08:
@Ergonomicmike -- Looks like the FAA's tweaking the TIS-B system again. Your local ground stations are sending radar-derived targets for everyone, even if they're already transmitting ADS-B. The two targets I circled are the same airplane. One's TIS-B, the other's ADS-B.

[image]

Other observations:
- The target with the "VFR" callsign is transmitting a random, anonymous address, rather than the FAA-assigned ICAO code. That's a neat little feature that's available on most UAT Out transmitters if you're squawking 1200. 
- Does UND own your airspace, or what? I thought we were thick with them up here.

skypuppy, 2016-03-08:
> screenshot_2016-03-04-09-23-53

Jeez, look at the quantity of traffic!  Wow.

Ergonomicmike, 2016-03-08:
I was wondering if it might be a FAA problem.

Thanks for looking into it. Will close this now. 

I don't know who UND is. We have a few flight schools in the area (because, like, 360 days of VFR).

closed 2016-03-08 by Ergonomicmike

AvSquirrel, 2016-03-08:
University of North Dakota. Huge flight program in Grand Forks -- that airport sees about 300,000 operations a year.

Ergonomicmike, 2016-03-08:
We teach the Chinese here. I expect that will bite us someday.