cyoung/stratux #313: Error when trying to update to v0,8r1

Ergonomicmike, 2016-03-08:
1. Stratux version: reimaged to vanilla v0.7b1
2. Stratux config:

    SDR
      [ ] single
      [X] dual

    GPS
      [X] yes
      [ ] no
      type: BU-353

    AHRS
      [X] yes
      [ ] no

    power source:
Anker E5
    usb cable:
Anker cable
3. EFB app and version: (e.g., WingX Pro7 8.6.2)
N/A
    EFB platform: (e.g., iOS 9.2)
Android KK
    EFB hardware: (e.g., iPad Mini 2)
Asus ZenPad S 8
4. Description of your issue:

I just tried updating to v0.8r1 via the updater. I get a window that says

"192.168.10.1 says
error"

Tried two different browsers. Same message.

To be fair, I had done an rpi-update to my Stratux. But I get the same error after I reimaged back to v0.7b1

peepsnet, 2016-03-08:
If you updated your PI then the hostapd was probable updated. The Edimax uses a special version of hostapd.

How did you update to v.8? more detail

jpoirier, 2016-03-08:
@peepsnet do you know where the special version of hostapd is coming from?

jpoirier, 2016-03-08:
I've used this one http://www.juergenkeil.de/download/hostapd-2.2.rtl871xdrv.gz and compiled from source (http://www.edimax.com/images/Image/Driver_Utility/Wireless/NIC/EW-7811Un/EW-7811Un_Linux_driver_v1.0.0.5.zip) and they both seem to work but I'd like to know if the one chris is using with the image build has mods? And if so I'd like to see what they are.

peepsnet, 2016-03-08:
http://dl.dropbox.com/u/1663660/hostapd/hostapd.zip
This came from @cyoung 

I extracted it when I updated my pi. Now I keep it in the /root so I can fix my pi every time I update the OS

jpoirier, 2016-03-08:
@peepsnet pulling the binary from a stratux image isn't a problem, I'm just curious where it came from originally

peepsnet, 2016-03-08:
http://willhaley.com/blog/raspberry-pi-hotspot-ew7811un-rtl8188cus/

I'm not sure but this is a starting point!

peepsnet, 2016-03-08:
http://elinux.org/RPi_USB_Wi-Fi_Adapters

jpoirier, 2016-03-08:
If the hostapd binary that chris is using isn't some sort of golden binary then I don't see the point of messing  with unknown binaries when edimax hosts the latest hostapd source on their website (link posted in previous comment); it's easy to compile.

peepsnet, 2016-03-08:
I am just forwarding the info. I do remember reading an article that mentioned the file available in that website. Where did you see the link to the edimax hostapd on edimax.com?

Ergonomicmike, 2016-03-08:
@peepsnet I tried to update the way that's worked until this time. (Download the .sh to my tablet. Go to the Settings page in the UI. Click on 'select system file.' I get the error when I click 'install.')

This issue could be related to #310.

jpoirier, 2016-03-08:
apologies for hijacking this thread

@peepsnet you can dig it out of the following zip file http://www.edimax.com/images/Image/Driver_Utility/Wireless/NIC/EW-7811Un/EW-7811Un_Linux_driver_v1.0.0.5.zip or copy it from here https://github.com/jpoirier/stratux-setup/wpa_supplicant_hostapd  

The stratux-setup/wpa_supplicant_hostapd  folder is from the edimax zip file

Ergonomicmike, 2016-03-08:
No problem about hijacking this thread. I'm ashamed to say that the problem was a corrupt .sh file. (I suppose it would help if the error message were more specific about the error. So make another issue/request for enhancement someday?) Downloaded again and all is well. Will close this.

As for the special hostapd file, I make a copy from the sd card to the card. Isn't that  good enough?

closed 2016-03-08 by Ergonomicmike

jpoirier, 2016-03-08:
@Ergonomicmike that works

Just trying to get some sort of baseline on the hostapd binary; I've had wifi issues on several of the stratux units I put together where the wifi just stops working. 

The hostapd chris uses for the stratux images is timestamped 2012 and I've found newer versions, as well as, fairly recent source code.

The wifi issue I'm seeing is at random times it stops working, anywhere between ~45 minutes to ~2 hours after startup. The newer versions of hostapd don't seem to die once they're running but I do see startup problems. I have found that I'm able to mitigate the startup problem by doing one of the following

- with dongles inserted, disable both 978 and 1090 via the webui, reboot, then wait for about a minute after the wifi comes up before enabling 978/1090 

- with the dongles extracted, enable both 978 and 1090 via the web ui, reboot, then wait for about a minute after the wifi comes up before inserting the dongles

I know the wifi is up when I see the stratux ssid listed

skypuppy, 2016-03-08:
I have noticed hostapd errors during boot times.  I wonder if that is 
related.  What does dmesg tell you about hostapd after wifi drops?

Ergonomicmike, 2016-03-08:
@jpoirier Oh, I didn't know you were having Wi-Fi disconnects. When you say that the Wi-Fi stops working - is that a blue light of death thing?

FWIW, although I've had BLOD in flight, I've never had a start up issue. If you're having problems right at start up, and since pulling SDR's mitigates it, it sounds more like a power issue. Have you tried my bypass hack of wiring pin 2 of the GPIO to a 5v USB pad? If the start up problem goes away, then it must have been power.

As for the date of the hostapd binary - I ran rpi-update, which is supposed to update "firmware." As I understand it, "firmware" on the Pi means to update stuff in the first partition of the SD card. I don't know if that includes hostapd. I would report the date of mine to you. But I ran the "file" command and it doesn't tell me what I expected. If there's a way that I can determine the build of my binary, let me know. (Or you could install and run rpi-update at your end to see if it makes a difference. It updates the kernel to 4.1.9.)

jpoirier, 2016-03-08:
@skypuppy haven't looked yet, still experimenting with different versions
of hostapd in my setup script

@Ergonomicmike Yes, BLODs running the standard stratux image. I get start
up issues using the newer hostapd binaries I'm experimenting with, but
putting a short delay in gen_gdl90 before letting the dongle handling code
configure the dongles at power up seems to solve the problem.

hostapd lives on the main partition in /usr/sbin. When a stratux image is
built using spindle the original hostapd that comes with the kernel is
replaced by a *special* version that chris has (the same one peepsnet
linked to). If you look at the original time stamp (ls -al
/usr/sbin/hostapd) you'll see it's from 2012, assuming it's the original
timestamp. If you do $ readelf -p .comment /usr/local/hostapd  you'll see
it's compiled against Debian 4.3, which is a really old version.

jpoirier, 2016-03-09:
On a side note. I almost exclusively use an email app to reply to conversation that come from github and ehen replying I'd typically not top post, etc., and the @someperson thing didn't make any sense. But now that I've been replying directly via the github page itself I can see why the "@" symbol is used and now I feel like a major jerk for getting on to @bradanlane about not just quoting and bottom posting. I do find it difficult to follow a thread when nothing is quoted and replies are top posted.

Ergonomicmike, 2016-03-09:
@jpoirier I ran your commands. My hostapd is dated Feb 15 23:32 and says Debian 4.6.3. I don't understand how rpi-update could have done that to a file in Stratux's /usr/sbin. Maybe Chris updated hostapd?

And I was going to suggest delaying the start up of the dongles for power reasons. But you're way ahead of me. (Which is a good thing.)

jpoirier, 2016-03-09:
On Tue, Mar 8, 2016 at 6:08 PM, Ergonomicmike <notifications@github.com>
wrote:

> @jpoirier <https://github.com/jpoirier> I ran your commands. My hostapd
> is dated Feb 15 23:32 and says Debian 4.6.3. I don't understand how
> rpi-update could have done that to a file in Stratux's /usr/sbin. Maybe
> Chris updated hostapd?
>
Probably the same/correct binary since it's dumping Debian 4.6.3, the time
stamp was probably touched during a copy/transfer.

> And I was going to suggest delaying the start up of the dongles for power
> reasons. But you're way ahead of me. (Which is a good thing.)
>
Hey, great minds think alike. Okay, so mine is only mediocre. :)

skypuppy, 2016-03-09:
Same for me.  Sometimes it is really hard to follow a thread with no 
history along with the current post.  Sometimes, a bare post makes no 
sense at all.

Ergonomicmike, 2016-03-09:
@jpoirier 

> it's compiled against Debian 4.3, which is a really old version

Just to make sure, mine is 4.6.3. Was the "4.3" a typo on your end? I guess it doesn't matter, because I now see that Debian is up to v 8. ("jessie" vs. "etch.")

Ergonomicmike, 2016-03-09:
@skypuppy I noticed the 3 dots in your latest comment here, which quoted someone. I don't see an easy way to do that from the github webpage. (I see quotes, but that shows a vertical bar and text of the quote.) Please, how does one embedded the 3 dot reference?

Ergonomicmike, 2016-03-09:
@jpoirier If you can send me your hostapd binary (via sendspace or wikisend), I can copy it into my setup to see if I get the BLOD on start up too.

skypuppy, 2016-03-09:
I don't know.  I NEVER see my own posts.  Impossible for me to say.

jpoirier, 2016-03-09:
On Tue, Mar 8, 2016 at 7:41 PM, Ergonomicmike <notifications@github.com>
wrote:

> @jpoirier <https://github.com/jpoirier> If you can send me your hostapd
> binary (via sendspace or wikisend), I can copy it into my setup to see if I
> get the BLOD on start up too.
>
I have them on github and you can just git clone
https://github.com/jpoirier/stratux-setup

A) if you want to try hostapd-2.2.rtl871xdrv.gz  (was compiled in 2015) as
root first do:

    mv /usr/sbin/hostapd /usr/sbin/hostapd.orig
    cd to the stratux-setup folder
    gunzip hostapd-2.2.rtl871xdrv.gz
    mv ./hostapd /usr/sbin/hostapd
    reboot

B) if you want to compile from source:

    mv /usr/sbin/hostapd /usr/sbin/hostapd.orig

    cd stratux-setup/wpa_supplicant_hostapd/hostapd

    make

    mv ./hostapd /usr/sbin/hostapd

    reboot

You can look in stratux-setup to see the commands if you have questions

-joe

jpoirier, 2016-03-09:
*stratux-setup.sh for commands

On Tue, Mar 8, 2016 at 8:26 PM, Joseph Poirier <jdpoirier@gmail.com> wrote:

> On Tue, Mar 8, 2016 at 7:41 PM, Ergonomicmike <notifications@github.com>
> wrote:
>
> I have them on github and you can just git clone
> https://github.com/jpoirier/stratux-setup
>
> A) if you want to try [...]

jpoirier, 2016-03-09:
On Tue, Mar 8, 2016 at 7:21 PM, Ergonomicmike <notifications@github.com>
wrote:

> @skypuppy <https://github.com/skypuppy> I noticed the 3 dots in your
> latest comment here, which quoted someone. I don't see an easy way to do
> that from the github webpage. (I see quotes, but that shows a vertical bar
> and text of the quote.) Please, how does one embedded the 3 dot reference?
>
I think the three dots are placed there (as folded text) by the tool you're
using to read the message. To get the quotes (vertical bar) when replying
via the github page, copy and paste the text you want to quote then prepend
a ">" to the sentence/s or paragraph/s. You only need to put a single ">"
before the first word in a paragraph for the entire paragraph to be quoted,
similar for long lines that are wrapped.

Ergonomicmike, 2016-03-09:
@jpoirier Thanks. It was simpler for a dummy like me to unzipping the .gz file from your earlier link for peepsnet, rename the uncompressed file to hostapd (down from all the -2.2.rtl871xdrv suffix) and put that in the usr/sbin directory. (After renaming the original hostapd to .orig.)

It must be working because it booted. No BLOD on boot. (I've got that jumper wire for a little more voltage at the USB.) I don't know what to look for in a test, but I'll run it to see if I notice any problems. Probably won't get to fly for a week tho for the real test.  :-(

Ergonomicmike, 2016-03-09:
P.S. For what it's worth, it does seem reasonable to use a version of hostapd that uses the latest Edimax driver ([v1.0.0.5, released on 12-12-12](http://us.edimax.com/edimax/download/download/data/edimax/us/download/for_home/wireless_adapters/wireless_adapters_n150/ew-7811un)) compiled on at least the wheezy version of Debian.

jpoirier, 2016-03-09:
On Tue, Mar 8, 2016 at 9:55 PM, Ergonomicmike <notifications@github.com>
wrote:

> @jpoirier <https://github.com/jpoirier> Thanks. It was simpler for a
> dummy like me to unzipping the .gz file from your earlier link for
> peepsnet, rename the uncompressed file to hostapd (down from [...]
Glad it was straightforward

> It must be working because it booted. No BLOD on boot. (I've got that
> jumper wire for a little more voltage at the USB.) I don't know what to
> look for in a test, but I'll run it to see if I notice any problems.
> Probably won't get to fly for a week tho for the real test. :-(
>
Hmm, been using one particular stratux unit to run tests, guess I need to
give it a go on a second unit. If I don't have problems on the second unit
then the first is suspect, but if the second unit has the startup problem I
guess I'll look in to adding the jumper wire.

thanks for testing it out!

jpoirier, 2016-03-09:
On Tue, Mar 8, 2016 at 10:33 PM, Ergonomicmike <notifications@github.com>
wrote:

> P.S. For what it's worth, it does seem reasonable to use a version of
> hostapd that uses the latest Edimax driver (v1.0.0.5, released on 12-12-12 [...]

I hadn't even realized the "latest" code was that old. :/  But, yeah, it
would be nice to a version compiled against a recent kernel with an updated
gcc compiler.

I'd be interested to hear how the compiled version ran for you if you do
get a chance to compile and test it.

Ergonomicmike, 2016-03-09:
@jpoirier Okay, I hacked my way thru compiling hostapd from source, albeit still on the wheezy version on the Pi. (Can only do so much hacking. Got a few Warnings along the way about some things missing or not set.)

To my surprise, readelf is still showinf 4.6.3. So I'm wondering if that's a reflection of when the Edimax driver was released?  In fact, the driver from Germany that was compiled in 2015 also shows Debian 4.6.3.

In any event, no BLOD on boot.

jpoirier, 2016-03-09:
On Wed, Mar 9, 2016 at 12:17 AM, Ergonomicmike <notifications@github.com>
wrote:

> @jpoirier <https://github.com/jpoirier> Okay, I hacked my way thru
> compiling hostapd from source, albeit still on the wheezy version on the
> Pi. (Can only do so much hacking. Got a few Warnings along the way about
> some things missing or not set.)
>
Nice

> To my surprise, readelf is still showinf 4.6.3. So I'm wondering if that's
> a reflection of when the Edimax driver was released? In fact, the driver
> from Germany that was compiled in 2015 also shows Debian 4.6.3.
>
Something to do with the RPi distro's env settings; when compiled on my
Fedora machine I get Red Hat info with a current kernel version dumped

> In any event, no BLOD on boot.
>
Cool.

Ergonomicmike, 2016-03-09:
@jpoirier Also, FWIW (and I presume that you've seen this), the German version of hostapd is about 500 KB larger than the original and my compiled one. (The latter are only 3 bits different.)

jpoirier, 2016-03-09:
On Wed, Mar 9, 2016 at 12:52 AM, Ergonomicmike <notifications@github.com>
wrote:

> @jpoirier <https://github.com/jpoirier> Also, FWIW (and I presume that
> you've seen this), the German version of hostapd is about 500 KB larger
> than the original and my compiled one. (The latter are only 3 bits
> different.)
>
Definitely a plus to be able to build third party binaries from source that
we have in our possession. I think we should talk with Chris about getting
the hostapd source in to the stratux build.

cyoung, 2016-03-09:
Just catching up.

@jpoirier: it is http://www.daveconroy.com/wp3/wp-content/uploads/2013/07/hostapd.zip

cyoung, 2016-03-09:
I haven't changed the hostapd binary since the first release.

@Ergonomicmike - If I understand correctly, you're testing the latest hostapd build that works with Edimax to see if BLOD behavior changes?

Ergonomicmike, 2016-03-09:
@cyoung 

>  If I understand correctly, you're testing the latest hostapd build that works with Edimax to see if BLOD behavior changes?

I don't know that I understand it correctly. But yes, I am testing for the BLOD. 

The last thing I did was 1) extracted jpoirier's hostapd stuff from his github zip file along with the v1.0.0.5 driver from Edimax to a directory on my Pi; 2) did a make command. A lot to things happened with CC's in front of them (with a few warnings). 3) When it was done, I had a new hostapd, which I moved to the /usr/sbin directory.

Before all that, I had installed rpi-update and updated my "firmware" on the first partition. (Kernel to 4.1.19 now.) So if the BLOD stops, it could be three variables: The update to v0.8r1, the compiled hostapd, or firmware/kernel updates.

No flying planned until Monday or Tuesday, so I won't have any acid test until then. (It's been running fine here at home. But it always runs fine at home.)

jpoirier, 2016-03-09:
On Wed, Mar 9, 2016 at 7:04 AM, cyoung <notifications@github.com> wrote:

> I haven't changed the hostapd binary since the first release.
>
> @Ergonomicmike <https://github.com/Ergonomicmike> - If I understand
> correctly, you're testing the latest hostapd build that works with Edimax
> to see if BLOD behavior changes?
>
@cyoung

tl;dr we have the latest edimax hostapd source code (circa 2012) and it's
trivial to build, therefore, it'd be easy to include in the stratux image
build process if there was a desire to do so.

We were testing a newer hostapd binary, as well as, a binary built from the
latest source code. But as it turns out, the latest edimax hostapd code
available is from 2012. So not much gained building from source except the
binary would be compiled with a much newer gcc and libc and against a newer
kernel.

@Ergonomicmike

I think my start-up BLOD problem (with the newer hostapd) may have been due
to the crappy AC power connector; when I run my unit via an anker battery I
don't see the problem.

Links to the edimax hostapd code, they all point to the same code:

joe's copy:

https://github.com/jpoirier/stratux-setup/tree/master/wpa_supplicant_hostapd

edimax repo:

http://www.edimax.com/images/Image/Driver_Utility/Wireless/NIC/EW-7811Un/EW-7811Un_Linux_driver_v1.0.0.5.zip

Realtek repo:

http://152.104.125.41/downloads/downloadsView.aspx?Langid=1&PNid=21&PFid=48&Level=5&Conn=4&ProdID=27.
..