cyoung/stratux #314: IMG files on deltas

rapgood, 2016-03-09:
1. Stratux version: all recent
2. Stratux config:

    SDR
      [ ] single
      [ x] dual

    GPS
      [ ] yes
      [x ] no
      type:

    AHRS
      [ ] yes
      [x ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:
Did I miss something? Latest deltas do not offer .img files. Not all of us necessarily want to create a build environment. Will you resume supplying .img files? Thanks!
If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

peepsnet, 2016-03-09:
They are being worked on!

cyoung, 2016-03-09:
>Not all of us necessarily want to create a build environment.

Look for ".sh" files in the future. Going forward from v0.8r1, you are able to update via the web interface (Settings page). It's much faster.

>Will you resume supplying .img files?

v0.8r1 image will be up today. It's not even officially a "release" yet.

closed 2016-03-09 by cyoung